leifliddy/macbook12-audio-driver #1: build an out-of-tree kernel module

meeuw, 2019-08-24:
You might be interested in my work for building an out-of-tree kernel module:
https://github.com/meeuw/snd_hda_macbookpro/tree/master-v5.1.20

This allows you to build a module using kernel-devel instead of kernel-sources.

I'm planning to create an akmod module for Fedora, I wondered if the sources in thos repository are ahead of the patches from davidjo (do you have other sources?).

leifliddy, 2019-08-26:
The patches in this repository are completely different then davidjo's. Our patches would need to be merged together. But, first I need to clean up patch_cirrus.c. There a ton of cruft leftover from testing that needs to be removed. I'm on vacation at the moment, but I should have time to do it next week. 
I like the idea of merging these repositories together. Hopefully, davidjo would be open to this idea. If I could merge my stuff with his, I could shut this repo down.

meeuw, 2019-08-27:
nice work! I have an macbook pro 13,3 but I haven't tried your or davidjo's work on it.

davidjo posted that he intends to push more updates:
https://github.com/davidjo/snd_hda_macbookpro/issues/3

Could you please share how you got your information for the changes you've made? I actually have some kernel development experience and I'd love to help.

closed 2021-07-31 by leifliddy