leifliddy/macbook12-audio-driver #2: Error with in the patch_cirrus.c

Mydlear, 2019-09-16:
Hi,
When i launched your main script i got :

> macbook12-audio-driver/build/hda-4.19.0/patch_cirrus.c:27:10: fatal error: sound/hda_codec.h: Aucun fichier ou dossier de ce type
 #include <sound/hda_codec.h>

So i deduced that the hda_codec.h was in the "build/hda-4.19.0/" folder and not in "build/hda-4.19.0/sound/", so i replaced 
`#include <sound/hda_codec.h>`
by
`#include "hda_codec.h"`

and it worked.
So the first lines of the patch_cirrus.c need to be reviewed.

Regards,
Thomas.

leifliddy, 2019-09-16:
Did you install the kernel headers on your system? (linux-headers-generic, kernel-devel..etc)

#include <sound/hda_codec.h> 
**this designates a relative path from the kernel header directory.
ie
/usr/src/kernels/5.2.13-200.fc30.x86_64/include/

so the full path where this header file would be located would be something like:
/usr/src/kernels/5.2.13-200.fc30.x86_64/include/sound/hda_codec.h

I suppose I could modify the script to just use the kernel header files included in the linux-x.x.x.tar.xz archive (instead of using the ones on the local system). I'll have to think about it...

Mydlear, 2019-09-16:
Thank you very much.

I'm using Ubuntu 18.10 (a modified version to make all known functional driver work) so this is maybe the problem. 
And actually, I was trying to get the audio working on a mbp 13 2017 with touch bar, not a 12', but I don't know how to compile and install this driver so I was trying to modify your install.sh to make the davidjo version work.

I would be glad to receive any advice on this  since I have no clue to what to do

leifliddy, 2019-09-16:
Shouldn't be that hard. Just move davidjo's files to your "/home/thomas/macbook12-audio-driver/build/hda-4.19.0/" directory (which will overwrite the patch_cirrus.c file) 
then run 'make' to compile the driver.
Let me know if that works for you.

Mydlear, 2019-09-16:
I got this when i run `make` or `sudo make` : 

> make -C /lib/modules/4.19.0-041900-generic/build M= modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
  YACC    scripts/kconfig/zconf.tab.c
/bin/sh: 1: bison: not found
make[3]: *** [scripts/Makefile.lib:196: scripts/kconfig/zconf.tab.c] Error 127
make[2]: *** [Makefile:539: syncconfig] Error 2
make[1]: *** [Makefile:635: include/config/auto.conf] Error 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
make: *** [Makefile:5: all] Error 2

and my audio still not working

leifliddy, 2019-09-16:
Try installing bison and see if that helps.
apt-get install bison

Mydlear, 2019-09-16:
now i got this after installing bison and flex

> make -C /lib/modules/4.19.0-041900-generic/build M= modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
make[3]: Dépendance circulaire silentoldconfig <- syncconfig abandonnée.
  WARNING: "silentoldconfig" has been renamed to "syncconfig"
            and is now an internal implementation detail.
            What you want is probably "oldconfig".
            "silentoldconfig" will be removed after Linux 4.19
scripts/kconfig/conf  --syncconfig Kconfig
make[2]: ***  Aucune règle pour fabriquer la cible « arch/x86/entry/syscalls/syscall_32.tbl », nécessaire pour « arch/x86/include/generated/asm/syscalls_32.h ». Arrêt.
make[1]: *** [arch/x86/Makefile:238: archheaders] Error 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
make: *** [Makefile:5: all] Error 2

leifliddy, 2019-09-16:
I was able build these modules in an ubuntu 18.10 docker container no problem. 
Didn't need bison or flex installed.
How are you running 'make'?

Mydlear, 2019-09-16:
i get this 

> make -C /lib/modules/4.19.0-041900-generic/build M=/home/thomas/macbook12-audio-driver/build/hda-4.19.0 modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
arch/x86/Makefile:150: CONFIG_X86_X32 enabled but no binutils support
rm: impossible de supprimer '/home/thomas/macbook12-audio-driver/build/hda-4.19.0/.tmp_versions/snd-hda-codec-cirrus.mod': Permission non accordée
make[1]: *** [Makefile:1504: crmodverdir] Error 1
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
make: *** [Makefile:5: all] Error 2

leifliddy, 2019-09-16:
Try with sudo

Mydlear, 2019-09-16:
i have this now :

> make -C /lib/modules/4.19.0-041900-generic/build M=/home/thomas/macbook12-audio-driver/build/hda-4.19.0 modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
  Building modules, stage 2.
  MODPOST 1 modules
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »

no more error message

Mydlear, 2019-09-16:
but i had to modify the makefile slightly

leifliddy, 2019-09-16:
Odd, I guess that environment variable wasn't set for some reason
```
env  | grep PWD
PWD=/root/macbook12-audio-driver/build/hda
```

Mydlear, 2019-09-16:
Unfortunately, no. I will do some other try from zero and tell you if it work.
Thanks again for help, I have highly appreciated.

leifliddy, 2019-09-16:
Hmmm....looking at davido's driver file, I think you might need to compile/install the snd-hda-intel module as well. It's worth a shot.

Mydlear, 2019-09-16:
I will try this. It is a separate module right?

leifliddy, 2019-09-16:
yeah, just change the top of the Makefile to this:
```
snd-hda-intel-objs := hda_intel.o
snd-hda-codec-cirrus-objs :=    patch_cirrus.o
CFLAGS_hda_intel.o := -I$(src)
obj-$(CONFIG_SND_HDA_INTEL) += snd-hda-intel.o
obj-$(CONFIG_SND_HDA_CODEC_CIRRUS) += snd-hda-codec-cirrus.o
```

Mydlear, 2019-09-16:
Still not working :(

leifliddy, 2019-09-16:
davidjo specifies the audio specs for the macbook14,3 are '2/4 channel 44.1 kHz S24_LE S32_LE'

I would try playing around with pulse audio configs.
At the minimum, I would specify the default-sample-format and the default-sample-rate.

**/etc/pulse/daemon.conf**
```
resample-method = soxr-vhq
avoid-resampling = false
default-sample-format = s32le
default-sample-rate = 44100
default-sample-channels = 2
default-channel-map = front-left,front-right
```
..then kill pulse audio, or reboot

If that doesn't work try changing these lines 
**/etc/pulse/default.pa**
from
```
### Automatically load driver modules depending on the hardware available
.ifexists module-udev-detect.so
load-module module-udev-detect
.else
### Use the static hardware detection module (for systems that lack udev support)
load-module module-detect
.endif
```
to
```
### Automatically load driver modules depending on the hardware available
#.ifexists module-udev-detect.so
#load-module module-udev-detect
#.else
### Use the static hardware detection module (for systems that lack udev support)
#load-module module-detect
#.endif
load-module module-alsa-sink device=hw:0,0
```
..then reboot
when the system comes back up, try using **alsamixer** to adjust the volume

leifliddy, 2019-09-16:
BTW, I changed the Makefile to use M=$(shell pwd)

Mydlear, 2019-09-17:
I tried to modify the deamon.conf and the default.pa but i've got no result.

leifliddy, 2019-09-17:
Did you ever run 
```
depmod -a
```

Mydlear, 2019-09-17:
Oh ok, well no I didn't and I will test it when I can

Mydlear, 2019-09-17:
It's still not working.
But when i make a ` dmesg | grep snd_hda` i got this among the results :

> [    5.423802] snd_hda_intel: UNKNOWN subsystem id 0x106b3600
[    5.423815] snd_hda_codec_cirrus: probe of hdaudioC0D0 failed with error -1
[    5.432486] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
[    5.432488] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    5.432489] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)

So i guess that the drivers are installed but not entierely so there might be some part of the install that failed maybe ?

leifliddy, 2019-09-17:
can you the output of:
```
aplay -l
dmesg | grep snd_hda_intel
```

Mydlear, 2019-09-17:
sure : 

> **** Liste des Périphériques Matériels PLAYBACK ****
carte 0: PCH [HDA Intel PCH], périphérique 0: Generic Analog [Generic Analog]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 0: PCH [HDA Intel PCH], périphérique 3: HDMI 0 [HDMI 0]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 0: PCH [HDA Intel PCH], périphérique 7: HDMI 1 [HDMI 1]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 0: PCH [HDA Intel PCH], périphérique 8: HDMI 2 [HDMI 2]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 0: PCH [HDA Intel PCH], périphérique 9: HDMI 3 [HDMI 3]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 0: PCH [HDA Intel PCH], périphérique 10: HDMI 4 [HDMI 4]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

and : 

> thomas@thomas-MacBookPro:~$ dmesg | grep snd_hda_intel
[    5.203845] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.450514] snd_hda_intel: Patching for CS8409 explicit 0
[    5.450515] snd_hda_intel: pre cs_8409_parse_auto_config
[    5.450515] snd_hda_intel: cs_8409_parse_auto_config
[    5.452307] snd_hda_intel: end cs_8409_parse_auto_config
[    5.452308] snd_hda_intel: post cs_8409_parse_auto_config
[    5.452308] snd_hda_intel: headphone config hp_jack_present 0
[    5.452309] snd_hda_intel: headphone config line_jack_present 0
[    5.452309] snd_hda_intel: headphone config speaker_muted 0
[    5.452310] snd_hda_intel: headphone config line_out_muted 0
[    5.452310] snd_hda_intel: headphone config auto_mic 0
[    5.452311] snd_hda_intel: headphone config automute_speaker 0
[    5.452311] snd_hda_intel: headphone config automute_lo 0
[    5.452311] snd_hda_intel: headphone config detect_hp 0
[    5.452312] snd_hda_intel: headphone config detect_lo 0
[    5.452312] snd_hda_intel: headphone config keep_vref_in_automute 0
[    5.452312] snd_hda_intel: headphone config line_in_auto_switch 0
[    5.452313] snd_hda_intel: headphone config auto_mute_via_amp 0
[    5.452313] snd_hda_intel: headphone config suppress_auto_mute 0
[    5.452313] snd_hda_intel: headphone config suppress_auto_mic 0
[    5.452314] snd_hda_intel: headphone config hp_mic 0
[    5.452314] snd_hda_intel: headphone config suppress_hp_mic_detect 0
[    5.452315] snd_hda_intel: auto config pins line_outs 2
[    5.452315] snd_hda_intel: auto config pins line_outs 0x24
[    5.452316] snd_hda_intel: auto config pins line_outs 0x25
[    5.452316] snd_hda_intel: auto config pins speaker_outs 2
[    5.452317] snd_hda_intel: auto config pins speaker_outs 0x24
[    5.452317] snd_hda_intel: auto config pins speaker_outs 0x25
[    5.452317] snd_hda_intel: auto config pins hp_outs 1
[    5.452318] snd_hda_intel: auto config pins hp_outs 0x2c
[    5.452318] snd_hda_intel: auto config pins inputs 2
[    5.452319] snd_hda_intel: auto config pins inputs  pin 0x44
[    5.452319] snd_hda_intel: auto config pins inputs type 0
[    5.452319] snd_hda_intel: auto config pins inputs is head set mic 0
[    5.452320] snd_hda_intel: auto config pins inputs is head phn mic 0
[    5.452320] snd_hda_intel: auto config pins inputs is        boost 1
[    5.452321] snd_hda_intel: auto config pins inputs  pin 0x3c
[    5.452321] snd_hda_intel: auto config pins inputs type 0
[    5.452321] snd_hda_intel: auto config pins inputs is head set mic 0
[    5.452322] snd_hda_intel: auto config pins inputs is head phn mic 0
[    5.452322] snd_hda_intel: auto config pins inputs is        boost 0
[    5.452323] snd_hda_intel: auto config inputs num_adc_nids 8
[    5.452323] snd_hda_intel: auto config inputs adc_nids 0x12
[    5.452324] snd_hda_intel: auto config inputs adc_nids 0x13
[    5.452324] snd_hda_intel: auto config inputs adc_nids 0x14
[... 58 lines cut ...]
[    5.454717] snd_hda_intel: auto config inputs num_adc_nids 8
[    5.454717] snd_hda_intel: auto config inputs adc_nids 0x12
[    5.454718] snd_hda_intel: auto config inputs adc_nids 0x13
[    5.454718] snd_hda_intel: auto config inputs adc_nids 0x14
[    5.454719] snd_hda_intel: auto config inputs adc_nids 0x15
[    5.454719] snd_hda_intel: auto config multiout is num_dacs 2
[    5.454719] snd_hda_intel: auto config multiout is    dac_nids 0x02
[    5.454720] snd_hda_intel: auto config multiout is    dac_nids 0x03
[    5.454720] snd_hda_intel: auto config multiout is    dac_nids 0x00
[    5.454721] snd_hda_intel: auto config multiout is    dac_nids 0x00
[    5.454721] snd_hda_intel: auto config multiout is      hp_nid 0x00
[    5.454722] snd_hda_intel: auto config multiout is  hp_out_nid 0x0a
[    5.454722] snd_hda_intel: auto config multiout is  hp_out_nid 0x00
[    5.454722] snd_hda_intel: auto config multiout is  hp_out_nid 0x00
[    5.454723] snd_hda_intel: auto config multiout is  hp_out_nid 0x00
[    5.454723] snd_hda_intel: auto config multiout is xtr_out_nid 0x00
[    5.454724] snd_hda_intel: auto config multiout is xtr_out_nid 0x00
[    5.454724] snd_hda_intel: auto config multiout is xtr_out_nid 0x00
[    5.454724] snd_hda_intel: auto config multiout is xtr_out_nid 0x00
[    5.454725] snd_hda_intel: auto config multiout is dif_out_nid 0x00
[    5.454726] snd_hda_intel: auto config multiout is slv_dig_out           (null)
[    5.454726] snd_hda_intel: UNKNOWN subsystem id 0x106b3600

périphérique = device
sous-périphérique = sub-device

leifliddy, 2019-09-17:
I think it's failing here in patch_cirrus.c
```    
   err = cs_8409_boot_setup(codec);
       if (err < 0)
               goto error;
```
You should have seen this in th dmesg log
```
snd_hda_intel: Post Patching for CS8409
```
Looks like it's due to the subsystem id  0x106b3600 not being recognized.

Mydlear, 2019-09-17:
Yes exactly i have a mbp 14,2. I should have checked my model better.
So... I will try to contact him.

Mydlear, 2019-09-17:
Anyways, thank you for your help. At least, i documented myself a little on how the kernel work.

leifliddy, 2019-09-17:
The MBP14,1 has two microphones, while the MBP14,2 has three microphones. 
I would try changing any references to 0x106b3300 to 0x106b3600 and see if that works. 
The microphones might not work properly, but the speakers might. It's worth a shot in any case. 
```
patch_cirrus.c:       SND_PCI_QUIRK(0x106b, 0x3300, "MacBookPro 13,1", CS8409_MBP131),
patch_cirrus_data84.h:        snd_hda_coef_item(codec, 0, CS8409_VENDOR_NID, 0x005c, 0x0000, 0x00000038, 3300 ); // i2cPagedWrite  coef read 3300
patch_cirrus_mb141_data84.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b3300, 8); // 0x001f2000
patch_cirrus_mb141_data84.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b3300, 22); // 0x001f2000
patch_cirrus_new84.h:// 14,1 0x106b3300
patch_cirrus_new84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
```

leifliddy, 2019-09-17:
I would create an issue on davidjo's github page and request that he add support for the MBP14,2 model.

Mydlear, 2019-09-17:
Okay, i will try to change theses references.

Mydlear, 2019-09-17:
I don't believe it... IT WORK !

Mydlear, 2019-09-17:
Thanks again, i wouldn't be there without you.

leifliddy, 2019-09-17:
No worries. Nice to hear it's working for you!

Mydlear, 2019-09-17:
Should I close this issue?

leifliddy, 2019-09-17:
Sure, go ahead.

closed 2019-09-17 by Mydlear

leifliddy, 2019-09-17:
I would still contact davidjo and request that he add support for the MBP14,2. Tell him you swapped the device id's with the MBP14,1 and that the speakers are working. That way other people will be able to benefit from this. That's what open source development is all about : )

Mydlear, 2019-09-17:
Yes I are right I will post an issue talking about it.