leifliddy/macbook12-audio-driver #3: Thanks for hints

davidjo, 2019-10-15:
Just wanted to leave a message about the Mydlear updates for 14,2. I have finally got headphone plugin/unplug working - however this is very specific to the 8409/cs42l83 setup - but there may be some ideas you can use or pass on. I have cleaned up the code a lot.
Anyway thanks for the hints about the 14,2 which according to Mydlear does seem to work with  my latest updates - so looks as tho its now working for 14,1, 14,2 and 14,3.

leifliddy, 2019-10-17:
Great work so far, that's really impressive!! I would love the get the headphones working properly with my driver, but I simply don't have the depth of knowledge that it would take to implement that. Your kernel development skills are light-years beyond mine. 

Some of the groundwork has already been laid:
*reference*
```
static int headphones_a1534 (struct hda_codec *codec) {
	int retval;
	//printk("snd_hda_intel: headphones_a1534 begin");
        retval = snd_hda_codec_read_check(codec, 0x00, 0, AC_VERB_PARAMETERS, 0x00000000, 0x10134208, 1); // 0x000f0000
...
```

I did managed to get the headphones working (sort of), but there were two main issues:
1. The jack sense / headphone pin sense didn't always work. Worked less than 50% of the time. 
2. Sometimes the volume was REALLY loud, and other times the volume levels were normal.

With that said, I don't use wired headphones. So it's not really a huge deal for me. But, it would be nice to have a complete, functioning driver.

davidjo, 2019-10-18:
I suspect an issue with unsolicited responses - this gave me great problems - it seems in linux the unsolicited responses are done concurrently with other verbs - individual HDA verbs are protected by mutex locks but if a UR requires multiple verbs this can interfere with other verbs being sent.
What is the HDA chip for the a1534 - had difficulties sorting this out from google. Does it have separate amps and/or headset chips?

leifliddy, 2019-10-24:
I've asked David U to comment on this. 

You might find the following post to be useful. 
https://bugzilla.kernel.org/show_bug.cgi?id=110561#c70

closed 2021-07-31 by leifliddy