leifliddy/macbook12-audio-driver #4: Dependency error? #include <sound/hda_codec.h>

cofc-tech, 2019-10-23:
New to this game, but this appears to be exactly what I need to get the speaker working on my Macbook8,1.  Following the instructions, it errors out though:

`root@Mac-MacBook:/macbook12-audio-driver# ./install.cirrus.driver.sh
--2019-10-23 18:11:52--  https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.15.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.57.176, 2a04:4e42:e::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.57.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-10-23 18:11:52 ERROR 404: Not Found.

--2019-10-23 18:11:52--  https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.15.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.57.176, 2a04:4e42:e::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.57.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-4.15/sound/pci/hda/
linux-4.15/sound/pci/hda/Kconfig
linux-4.15/sound/pci/hda/Makefile
linux-4.15/sound/pci/hda/ca0132_regs.h
linux-4.15/sound/pci/hda/dell_wmi_helper.c
linux-4.15/sound/pci/hda/hda_auto_parser.c
linux-4.15/sound/pci/hda/hda_auto_parser.h
linux-4.15/sound/pci/hda/hda_beep.c
linux-4.15/sound/pci/hda/hda_beep.h
linux-4.15/sound/pci/hda/hda_bind.c
linux-4.15/sound/pci/hda/hda_codec.c
linux-4.15/sound/pci/hda/hda_codec.h
linux-4.15/sound/pci/hda/hda_controller.c
linux-4.15/sound/pci/hda/hda_controller.h
linux-4.15/sound/pci/hda/hda_controller_trace.h
linux-4.15/sound/pci/hda/hda_eld.c
linux-4.15/sound/pci/hda/hda_generic.c
linux-4.15/sound/pci/hda/hda_generic.h
linux-4.15/sound/pci/hda/hda_hwdep.c
linux-4.15/sound/pci/hda/hda_intel.c
linux-4.15/sound/pci/hda/hda_intel.h
linux-4.15/sound/pci/hda/hda_intel_trace.h
linux-4.15/sound/pci/hda/hda_jack.c
linux-4.15/sound/pci/hda/hda_jack.h
linux-4.15/sound/pci/hda/hda_local.h
linux-4.15/sound/pci/hda/hda_proc.c
linux-4.15/sound/pci/hda/hda_sysfs.c
linux-4.15/sound/pci/hda/hda_tegra.c
linux-4.15/sound/pci/hda/local.h
linux-4.15/sound/pci/hda/patch_analog.c
linux-4.15/sound/pci/hda/patch_ca0110.c
linux-4.15/sound/pci/hda/patch_ca0132.c
linux-4.15/sound/pci/hda/patch_cirrus.c
linux-4.15/sound/pci/hda/patch_cmedia.c
linux-4.15/sound/pci/hda/patch_conexant.c
linux-4.15/sound/pci/hda/patch_hdmi.c
linux-4.15/sound/pci/hda/patch_realtek.c
linux-4.15/sound/pci/hda/patch_si3054.c
linux-4.15/sound/pci/hda/patch_sigmatel.c
linux-4.15/sound/pci/hda/patch_via.c
linux-4.15/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/4.15.0-66-generic/build M=/macbook12-audio-driver/build/hda-4.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-66-generic'
  CC [M]  /macbook12-audio-driver/build/hda-4.15.0/patch_cirrus.o
/macbook12-audio-driver/build/hda-4.15.0/patch_cirrus.c:14:10: fatal error: sound/hda_codec.h: No such file or directory
 #include <sound/hda_codec.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.build:330: recipe for target '/macbook12-audio-driver/build/hda-4.15.0/patch_cirrus.o' failed
make[2]: *** [/macbook12-audio-driver/build/hda-4.15.0/patch_cirrus.o] Error 1
Makefile:1577: recipe for target '_module_/macbook12-audio-driver/build/hda-4.15.0' failed
make[1]: *** [_module_/macbook12-audio-driver/build/hda-4.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-66-generic'
Makefile:5: recipe for target 'all' failed
make: *** [all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/4.15.0-66-generic/updates/snd-hda-codec-cirrus.ko_speaker
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
Makefile:10: recipe for target 'install' failed
make: *** [install] Error 1

contents of /lib/modules/4.15.0-66-generic/updates
ls: cannot access '/lib/modules/4.15.0-66-generic/updates': No such file or directory`

leifliddy, 2019-10-24:
As I stated in the readme file, the driver has only been tested on kernels 5.0 and later. 
I think that with kernel 4.15, hda_codec.h was included in sound/pci/hda/ itself (vs a header file)

So you should just need to edit 
*patch_cirrus.c*
and replace 
```#include <sound/hda_codec.h>```
with
```#include "hda_codec.h"`

leifliddy, 2019-10-30:
We're you able to try this out? Do you need help with anything?

leifliddy, 2019-11-05:
@cofc-tech closing issue due to lack of response.

closed 2019-11-05 by leifliddy