leifliddy/macbook12-audio-driver #5: newer kernel?!

Ali-Javadi, 2019-10-27:
mv: cannot move 'hda' to 'build/hda-5.3.0/hda': Permission denied
mv: replace 'build/hda-5.3.0/Makefile.orig', overriding mode 0664 (rw-rw-r--)? y
mv: cannot move 'build/hda-5.3.0/Makefile' to 'build/hda-5.3.0/Makefile.orig': Permission denied
mv: replace 'build/hda-5.3.0/patch_cirrus.c.orig', overriding mode 0664 (rw-rw-r--)? y
mv: cannot move 'build/hda-5.3.0/patch_cirrus.c' to 'build/hda-5.3.0/patch_cirrus.c.orig': Permission denied
cp: cannot create regular file 'build/hda-5.3.0/Makefile': Permission denied
cp: cannot create regular file 'build/hda-5.3.0/patch_cirrus.c': Permission denied
cp: cannot create regular file 'build/hda-5.3.0/patch_cirrus_a1534_setup.h': Permission denied
cp: cannot create regular file 'build/hda-5.3.0/patch_cirrus_a1534_pcm.h': Permission denied
make -C /lib/modules/5.3.0-19-generic/build M=/home/ali/macbook12-audio-driver/build/hda-5.3.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-19-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
  CC [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/patch_cirrus.o
/home/ali/macbook12-audio-driver/build/hda-5.3.0/patch_cirrus.c:1262:1: fatal error: opening dependency file /home/ali/macbook12-audio-driver/build/hda-5.3.0/.patch_cirrus.o.d: Permission denied
 1262 | module_hda_codec_driver(cirrus_driver);
      | ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:290: /home/ali/macbook12-audio-driver/build/hda-5.3.0/patch_cirrus.o] Error 1
make[1]: *** [Makefile:1653: _module_/home/ali/macbook12-audio-driver/build/hda-5.3.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-19-generic'
make: *** [Makefile:5: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker
cp: cannot create regular file '/lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker': No such file or directory
make: *** [Makefile:10: install] Error 1

contents of /lib/modules/5.3.0-19-generic/updates
ls: cannot access '/lib/modules/5.3.0-19-generic/updates': No such file or directory

leifliddy, 2019-10-27:
either run ./install.cirrus.driver.sh as root or use sudo 
```
sudo ./install.cirrus.driver.sh
```

Ali-Javadi, 2019-10-27:
> either run ./install.cirrus.driver.sh as root or use sudo
> 
> ```
> sudo ./install.cirrus.driver.sh
> ```

sudo ./install.cirrus.driver.sh
[sudo] password for ali: 
--2019-10-27 22:46:28--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.3.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.129.176, 151.101.65.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.193.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-10-27 22:46:30 ERROR 404: Not Found.

--2019-10-27 22:46:30--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.3.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.129.176, 151.101.65.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.193.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.3/sound/pci/hda/
linux-5.3/sound/pci/hda/Kconfig
linux-5.3/sound/pci/hda/Makefile
linux-5.3/sound/pci/hda/ca0132_regs.h
linux-5.3/sound/pci/hda/hda_auto_parser.c
linux-5.3/sound/pci/hda/hda_auto_parser.h
linux-5.3/sound/pci/hda/hda_beep.c
linux-5.3/sound/pci/hda/hda_beep.h
linux-5.3/sound/pci/hda/hda_bind.c
linux-5.3/sound/pci/hda/hda_codec.c
linux-5.3/sound/pci/hda/hda_controller.c
linux-5.3/sound/pci/hda/hda_controller.h
linux-5.3/sound/pci/hda/hda_controller_trace.h
linux-5.3/sound/pci/hda/hda_eld.c
linux-5.3/sound/pci/hda/hda_generic.c
linux-5.3/sound/pci/hda/hda_generic.h
linux-5.3/sound/pci/hda/hda_hwdep.c
linux-5.3/sound/pci/hda/hda_intel.c
linux-5.3/sound/pci/hda/hda_intel.h
linux-5.3/sound/pci/hda/hda_intel_trace.h
linux-5.3/sound/pci/hda/hda_jack.c
linux-5.3/sound/pci/hda/hda_jack.h
linux-5.3/sound/pci/hda/hda_local.h
linux-5.3/sound/pci/hda/hda_proc.c
linux-5.3/sound/pci/hda/hda_sysfs.c
linux-5.3/sound/pci/hda/hda_tegra.c
linux-5.3/sound/pci/hda/hp_x360_helper.c
linux-5.3/sound/pci/hda/patch_analog.c
linux-5.3/sound/pci/hda/patch_ca0110.c
linux-5.3/sound/pci/hda/patch_ca0132.c
linux-5.3/sound/pci/hda/patch_cirrus.c
linux-5.3/sound/pci/hda/patch_cmedia.c
linux-5.3/sound/pci/hda/patch_conexant.c
linux-5.3/sound/pci/hda/patch_hdmi.c
linux-5.3/sound/pci/hda/patch_realtek.c
linux-5.3/sound/pci/hda/patch_si3054.c
linux-5.3/sound/pci/hda/patch_sigmatel.c
linux-5.3/sound/pci/hda/patch_via.c
linux-5.3/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.3.0-19-generic/build M=/home/ali/macbook12-audio-driver/build/hda-5.3.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-19-generic'
  CC [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/patch_cirrus.o
  LD [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/snd-hda-codec-cirrus.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/ali/macbook12-audio-driver/build/hda-5.3.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/snd-hda-codec-cirrus.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-19-generic'
cp snd-hda-codec-cirrus.ko /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker
cp: cannot create regular file '/lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker': No such file or directory
make: *** [Makefile:10: install] Error 1

contents of /lib/modules/5.3.0-19-generic/updates
ls: cannot access '/lib/modules/5.3.0-19-generic/updates': No such file or directory

leifliddy, 2019-10-27:
odd...that directory should exist. 
Ok, try creating it and let me know if that works. If so, I'll update the script to check for that condition.
```
mkdir /lib/modules/5.3.0-19-generic/updates
```

leifliddy, 2019-10-27:
I just modified */install.cirrus.driver.sh* with the update directory check.

Ali-Javadi, 2019-10-27:
mkdir: cannot create directory ‘/lib/modules/5.3.0-19-generic/updates’: File exists

git clone https://github.com/leifliddy/macbook12-audio-driver.git
Cloning into 'macbook12-audio-driver'...
remote: Enumerating objects: 177, done.
remote: Counting objects: 100% (177/177), done.
remote: Compressing objects: 100% (135/135), done.
remote: Total 177 (delta 105), reused 95 (delta 40), pack-reused 0
Receiving objects: 100% (177/177), 61.97 KiB | 60.00 KiB/s, done.
Resolving deltas: 100% (105/105), done.
ali@ali-MacBook:~/macbook12-audio-driver$ sudo ./install.cirrus.driver.sh
--2019-10-27 23:10:51--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.3.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.129.176, 151.101.193.176, 151.101.1.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.129.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-10-27 23:10:53 ERROR 404: Not Found.

--2019-10-27 23:10:53--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.3.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.65.176, 151.101.1.176, 151.101.193.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.65.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.3/sound/pci/hda/
linux-5.3/sound/pci/hda/Kconfig
linux-5.3/sound/pci/hda/Makefile
linux-5.3/sound/pci/hda/ca0132_regs.h
linux-5.3/sound/pci/hda/hda_auto_parser.c
linux-5.3/sound/pci/hda/hda_auto_parser.h
linux-5.3/sound/pci/hda/hda_beep.c
linux-5.3/sound/pci/hda/hda_beep.h
linux-5.3/sound/pci/hda/hda_bind.c
linux-5.3/sound/pci/hda/hda_codec.c
linux-5.3/sound/pci/hda/hda_controller.c
linux-5.3/sound/pci/hda/hda_controller.h
linux-5.3/sound/pci/hda/hda_controller_trace.h
linux-5.3/sound/pci/hda/hda_eld.c
linux-5.3/sound/pci/hda/hda_generic.c
linux-5.3/sound/pci/hda/hda_generic.h
linux-5.3/sound/pci/hda/hda_hwdep.c
linux-5.3/sound/pci/hda/hda_intel.c
linux-5.3/sound/pci/hda/hda_intel.h
linux-5.3/sound/pci/hda/hda_intel_trace.h
linux-5.3/sound/pci/hda/hda_jack.c
linux-5.3/sound/pci/hda/hda_jack.h
linux-5.3/sound/pci/hda/hda_local.h
linux-5.3/sound/pci/hda/hda_proc.c
linux-5.3/sound/pci/hda/hda_sysfs.c
linux-5.3/sound/pci/hda/hda_tegra.c
linux-5.3/sound/pci/hda/hp_x360_helper.c
linux-5.3/sound/pci/hda/patch_analog.c
linux-5.3/sound/pci/hda/patch_ca0110.c
linux-5.3/sound/pci/hda/patch_ca0132.c
linux-5.3/sound/pci/hda/patch_cirrus.c
linux-5.3/sound/pci/hda/patch_cmedia.c
linux-5.3/sound/pci/hda/patch_conexant.c
linux-5.3/sound/pci/hda/patch_hdmi.c
linux-5.3/sound/pci/hda/patch_realtek.c
linux-5.3/sound/pci/hda/patch_si3054.c
linux-5.3/sound/pci/hda/patch_sigmatel.c
linux-5.3/sound/pci/hda/patch_via.c
linux-5.3/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.3.0-19-generic/build M=/home/ali/macbook12-audio-driver/build/hda-5.3.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-19-generic'
  CC [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/patch_cirrus.o
  LD [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/snd-hda-codec-cirrus.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/ali/macbook12-audio-driver/build/hda-5.3.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/ali/macbook12-audio-driver/build/hda-5.3.0/snd-hda-codec-cirrus.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-19-generic'
cp snd-hda-codec-cirrus.ko /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker
ln -sf /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko
symlinks -c /lib/modules/5.3.0-19-generic/updates 	# relative path links are easier to follow
absolute: /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko -> /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko_speaker
changed:  /lib/modules/5.3.0-19-generic/updates/snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
depmod -a
killall alsactl &> /dev/null
modprobe -r snd_hda_intel
modprobe: FATAL: Module snd_hda_intel is in use.
make: *** [Makefile:15: install] Error 1

contents of /lib/modules/5.3.0-19-generic/updates
total 128
lrwxrwxrwx 1 root root     31 Oct 27 23:11 snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
-rw-r--r-- 1 root root 127968 Oct 27 23:11 snd-hda-codec-cirrus.ko_speaker
alsactl: no process found

leifliddy, 2019-10-27:
That's normal. It can't reload the snd_hda_intel module because it's in use. Just reboot your system.

Ali-Javadi, 2019-10-27:
yes. it works. thank you so much sir.

closed 2019-10-30 by leifliddy