leifliddy/macbook12-audio-driver #6: Error running install.cirrus.driver.sh

marcusenglert, 2019-12-26:
Hi,
just tried to run install.cirrus.driver.sh on my MCBOOK12 2017: 
.........
killall alsactl &> /dev/null
modprobe -r snd_hda_intel
modprobe: FATAL: Module snd_hda_intel is in use.
Makefile:10: recipe for target 'install' failed
make: *** [install] Error 1

contents of /lib/modules/5.0.0-37-generic/updates
insgesamt 332
drwxr-xr-x 2 root root   4096 Dez 25 18:17 dkms
-rw-r--r-- 1 root root 229160 Dez 26 10:28 hci_uart.ko
lrwxrwxrwx 1 root root     31 Dez 26 10:51 snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
-rw-r--r-- 1 root root 104368 Dez 26 10:51 snd-hda-codec-cirrus.ko_speaker
marcus@Marcus-MacBook:~/macbook12-audio-driver$ alsactl: Kein Prozess gefunden

Thanks for your help

Best

Marcus

leifliddy, 2019-12-26:
That's normal. It can't remove the snd_hda_intel module because it's in use. Just reboot your system, and the speakers should work.

marcusenglert, 2019-12-26:
Thanks! Not working though.... even after reboot

leifliddy, 2019-12-26:
The driver *should* work on a macbook10,1. But, I think anyone has confirmed that yet. 
It should be almost identical hardware-wise to a macbook9,1. 

can you post the output of:
```pacmd list-sinks```
```aplay --list-devices```
```amixer```

Also, If you run 
```alsamixer -c0```

Do see one PCM channel and four S/PDIF {1,4} channels?

marcusenglert, 2019-12-26:
Thanks! Here are the outputs:
1 sink(s) available.
  * index: 0
	name: <alsa_output.pci-0000_00_1f.3.analog-surround-40>
	driver: <module-alsa-card.c>
	flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
	state: SUSPENDED
	suspend cause: IDLE 
	priority: 9039
	volume: front-left: 61604 /  94% / -1,61 dB,   front-right: 61604 /  94% / -1,61 dB,   rear-left: 61604 /  94% / -1,61 dB,   rear-right: 61604 /  94% / -1,61 dB
	        balance 0,00
	base volume: 65536 / 100% / 0,00 dB
	volume steps: 65537
	muted: no
	current latency: 0,00 ms
	max request: 0 KiB
	max rewind: 0 KiB
	monitor source: 0
	sample spec: s16le 4ch 44100Hz
	channel map: front-left,front-right,rear-left,rear-right
	             Surround 4.0
	used by: 0
	linked by: 0
	configured latency: 0,00 ms; range is 0,50 .. 185,76 ms
	card: 0 <alsa_card.pci-0000_00_1f.3>
	module: 7
	properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "CS4208 Analog"
		alsa.id = "CS4208 Analog"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0x92820000 irq 32"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1f.3"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "9d71"
		device.product.name = "Sunrise Point-LP HD Audio"
		device.form_factor = "internal"
		device.string = "surround40:0"
		device.buffering.buffer_size = "65536"
		device.buffering.fragment_size = "32768"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-surround-40"
		device.profile.description = "Analog Surround 4.0"
		device.description = "Eingebautes Tongerät Analog Surround 4.0"
		alsa.mixer_name = "Cirrus Logic CS4208"
		alsa.components = "HDA:10134208,106b6600,00100401 HDA:8086280b,80860101,00100000"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	ports:
		analog-output: Analoge Ausgabe (priority 9900, latency offset 0 usec, available: unknown)
			properties:
				
	active port: <analog-output>

**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: PCH [HDA Intel PCH], Gerät 0: CS4208 Analog [CS4208 Analog]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: PCH [HDA Intel PCH], Gerät 3: HDMI 0 [HDMI 0]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: PCH [HDA Intel PCH], Gerät 7: HDMI 1 [HDMI 1]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: PCH [HDA Intel PCH], Gerät 8: HDMI 2 [HDMI 2]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: PCH [HDA Intel PCH], Gerät 9: HDMI 3 [HDMI 3]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: PCH [HDA Intel PCH], Gerät 10: HDMI 4 [HDMI 4]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0

Simple mixer control 'PCM',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 247 [97%] [-1.60dB]
  Front Right: Playback 247 [97%] [-1.60dB]
Simple mixer control 'IEC958',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',1
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',2
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',3
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',4
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]

I see one PCM and five S/PDIF Channels (one without number and 1,2,3,4)

Many thanks!!

Best

Marcus

leifliddy, 2019-12-26:
this what is looks like on my macbook9,1
**pacmd list-sinks**
```
1 sink(s) available.
  * index: 0
        name: <alsa_output.pci-0000_00_1f.3.analog-surround-40>
        driver: <module-alsa-card.c>
        flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
        state: SUSPENDED
        suspend cause: IDLE
        priority: 9039
        volume: front-left: 43299 /  66% / -10.80 dB,   front-right: 41350 /  63% / -12.00 dB,   rear-left: 43299 /  66% / -10.80 dB,   rear-right: 41350 /  63% / -12.00 dB
                balance -0.05
        base volume: 65536 / 100% / 0.00 dB
        volume steps: 65537
        muted: no
        current latency: 0.00 ms
        max request: 0 KiB
        max rewind: 0 KiB
        monitor source: 0
        sample spec: s16le 4ch 44100Hz
        channel map: front-left,front-right,rear-left,rear-right
                     Surround 4.0
        used by: 0
        linked by: 0
        configured latency: 0.00 ms; range is 0.50 .. 2000.00 ms
        card: 0 <alsa_card.pci-0000_00_1f.3>
        module: 6
        properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "CS4208 Analog"
                alsa.id = "CS4208 Analog"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "HDA Intel PCH"
                alsa.long_card_name = "HDA Intel PCH at 0xc1820000 irq 34"
                alsa.driver_name = "snd_hda_intel"
                device.bus_path = "pci-0000:00:1f.3"
                sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card0"
                device.bus = "pci"
                device.vendor.id = "8086"
                device.vendor.name = "Intel Corporation"
                device.product.id = "9d70"
                device.product.name = "Sunrise Point-LP HD Audio"
                device.form_factor = "internal"
                device.string = "surround40:0"
                device.buffering.buffer_size = "705600"
                device.buffering.fragment_size = "352800"
                device.access_mode = "mmap+timer"
                device.profile.name = "analog-surround-40"
                device.profile.description = "Analog Surround 4.0"
                device.description = "Built-in Audio Analog Surround 4.0"
                alsa.mixer_name = "Cirrus Logic CS4208"
                alsa.components = "HDA:10134208,106b6500,00100401 HDA:80862809,80860101,00100000"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card-pci"
        ports:
                analog-output: Analog Output (priority 9900, latency offset 0 usec, available: unknown)
                        properties:

        active port: <analog-output>
```

**amixer **
```
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 43299 [66%] [on]
  Front Right: Playback 41350 [63%] [on]
  Rear Left: Playback 43299 [66%] [on]
  Rear Right: Playback 41350 [63%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Capture 0 - 65536
  Front Left: Capture 65536 [100%] [on]
  Front Right: Capture 65536 [100%] [on]
  Rear Left: Capture 65536 [100%] [on]
  Rear Right: Capture 65536 [100%] [on]
```

**the next step would be posting the output of ```alsa-info.sh```

I'm not sure where the issue lies. There's only very minor differences hardware-wise between the macbook9,1 and macbook10,1.  The vendor node commands used in ```patch_cirrus_a1534_setup.h``` might be different. That's what I was afraid of....but I didn't have a macbook10,1 to test on when this driver was being developed. Unfortunately, it's not going to be an easy process to get the macbook10,1 supported. I really need three things for that to happen. 
1. a macbook10,1 to test on (obviously) 
2. and to get in touch w/ the kernel developer who wrote most of this driver (in the event that I can't figure it the source of the issue)
3. ...and setup a bounty or other form of payment for said kernel developer. (It's not super expensive or anything).

The time needed to get the the macbook10,1 supported would be far less then what it took to get the initial driver working (which was several weeks). I would estimate that it would only take a day or two for the developer to get it working. 

Alternatively, I could put you in touch w/ the developer and see if you guys could work something out. 

In any case, I was thinking about getting a macbook10,1 sometime next year. I could work on the driver when that happens, but that could be a few months away.

leifliddy, 2019-12-26:
I might have figured it out.
```
patch_cirrus_a1534_setup.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b6500, 8); // 0x001f2000
patch_cirrus_a1534_setup.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b6500, 22); // 0x001f2000
```

In ```patch_cirrus_a1534_setup.h``` there are two vendor node commands sent to ```0x106b6500``` however, the device id on the macbook10,1 is ```0x106b6600```

leifliddy, 2019-12-26:
How good are your Linux skills? Can you edit ```patch_cirrus_a1534_setup``` and change the two instances of ```0x106b6500``` to ```0x106b6600 ``` and re-run ```install.cirrus.sh```?

marcusenglert, 2019-12-26:
Many Thanks for your answer!
happy to provide a bounty if it helps. I am not very IT savy so I would probably not be the right partner for a developer...
Pls. see: http://alsa-project.org/db/?f=9a6f36e28529c35b10a8a319b64b63411986e06a

Many Thanks
Marcus

marcusenglert, 2019-12-26:
ahhhh, thanks!
I am a lousy Linux person, but I will try and revert.
Best
Marcus

leifliddy, 2019-12-26:
No worries, I can send you a script to do this if you want.

leifliddy, 2019-12-26:
Can you try this whenever you have time?
```
git clone https://github.com/leifliddy/macbook12-audio-driver.git  
cd macbook12-audio-driver/
cp patch_cirrus/patch_cirrus_a1534_setup.h patch_cirrus/patch_cirrus_a1534_setup.h.orig
sed -i 's/106b6500/106b6600/' patch_cirrus/patch_cirrus_a1534_setup.
./install.cirrus.driver.sh
reboot
```

leifliddy, 2019-12-26:
the alsa-info.sh is pretty much identical between the macbook10,1 and 9,1

```
[root@mac tmp]$ diff -up 10file 9file 
--- 10file	2019-12-26 21:25:52.031397663 +0100
+++ 9file	2019-12-26 21:26:24.220057099 +0100
@@ -6,7 +6,7 @@ Codec: Cirrus Logic CS4208
 Address: 0
 AFG Function Id: 0x1 (unsol 1)
 Vendor Id: 0x10134208
-Subsystem Id: 0x106b6600
+Subsystem Id: 0x106b6500
 Revision Id: 0x100401
 No Modem Function Group found
 Default PCM:
@@ -389,10 +389,10 @@ Node 0x23 [Beep Generator Widget] wcaps
 Node 0x24 [Vendor Defined Widget] wcaps 0xf0e2c1: 16-Channels Digital
   Unsolicited: tag=00, enabled=0
   Processing caps: benign=0, ncoeff=128
-Codec: Intel Kabylake HDMI
+Codec: Intel Skylake HDMI
 Address: 2
 AFG Function Id: 0x1 (unsol 0)
-Vendor Id: 0x8086280b
+Vendor Id: 0x80862809
 Subsystem Id: 0x80860101
 Revision Id: 0x100000
 No Modem Function Group found
@@ -486,4 +486,3 @@ Node 0x07 [Pin Complex] wcaps 0x40778d:
      0x02 0x03 0x04
 Node 0x08 [Vendor Defined Widget] wcaps 0xf00000: Mono
 --endcollapse--
```

marcusenglert, 2019-12-26:
Thank you!
Unfortunately still not working, sorry!
Best
Marcus

leifliddy, 2019-12-27:
Can you please verify that the subsystem id's have changed?
Please post the output from the following grep command
```
cd macbook12-audio-driver/patch_cirrus/
grep -i subsystem patch_cirrus_a1534_setup.h
```

arturasb, 2019-12-27:
Hi.

Your work is inspiring. I have a question regarding the Mac BookPro 14,1 - is it possible to utilize your patch for the 14,1 model ? Can you point what modifications are needed to your files in order to work on 14,1 ?

Regards
ArtūrasB.

marcusenglert, 2019-12-27:
Hi,
here you go:
retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b6600, 8); // 0x001f2000
        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b6600, 22); // 0x001f2000

Thank you!

Best

Marcus

arturasb, 2019-12-27:
> Hi,
> here you go:
> retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b6600, 8); // 0x001f2000
> retval = snd_hda_codec_read_check(codec, [...]

Thank you.
Unfortunately no sound after changes were made and everything was recompiled and installed :( I guess there are 14,1 specific bits and pieces which needs to be addressed somehow.

leifliddy, 2019-12-27:
@arturasb this driver is only for the 12 inch macbooks (2015-2017). 
davidjo's repo provides an audio driver for the MBP 14,1 14,2, and 14,3 models.
https://github.com/davidjo/snd_hda_macbookpro

arturasb, 2019-12-27:
> @arturasb this driver is only for the 12 inch macbooks (2015-2017).
> davidjo's repo provides an audio driver for the MBP 14,1 14,2, and 14,3 models.
> https://github.com/davidjo/snd_hda_macbookpro

Got. Posted my question there :D

leifliddy, 2019-12-28:
so....I now have the tools necessary to create the macbook10,1 driver myself. 
Today, I was able to recreate the macbook9,1 driver from macOS kernel logs.
No need for an outside developer anymore! 
I've also created a macOS bootable usb stick running a custom (macOS) kernel that can be used to generate the logs needed to create the Linux driver. I just need a macbook10,1 to test on.

marcusenglert, 2019-12-28:
wow! that is great news, thank you so much!
Anything I can do? Where are you based? 
Thanks and kind regards 
Marcus

leifliddy, 2019-12-28:
I'm based out of Germany. If you want, I could send you a usb stick that you would boot macOS from. 
From there, you would just run a couple of commands to generate and extract the necessary log files. 

**I'll delete this portion of the message later...
If you'd like to help out send me an email to leif.linux@gmail.

leifliddy, 2020-01-12:
Can you post the output of  ```amixer contents```
```
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=4,min=0,max=65536,step=1
  : values=11225,9257,9257,9257
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=4,min=0,max=65536,step=1
  : values=65536,65536,65536,65536
```

savoca, 2020-01-12:
Hi! I also have a 2017 model, maybe I can add some info.

My install:
```
alex@macbook:~$ lsb_release -d
Description:	Ubuntu 19.10
alex@macbook:~$ uname -r
5.3.0-26-generic
```
 
Subsystem/Vendor IDs from alsa-info:
```
alex@macbook:~$ grep -e "Subsystem Id" -e "Vendor Id:" Desktop/alsa-info.txt 
Vendor Id: 0x10134208
Subsystem Id: 0x106b6600
Vendor Id: 0x8086280b
Subsystem Id: 0x80860101
```

The switch.audio script works without errors.

amixer contents:
```
numid=26,iface=CARD,name='HDMI/DP,pcm=10 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=2,iface=CARD,name='HDMI/DP,pcm=3 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=8,iface=CARD,name='HDMI/DP,pcm=7 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=14,iface=CARD,name='HDMI/DP,pcm=8 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=20,iface=CARD,name='HDMI/DP,pcm=9 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=3,iface=MIXER,name='IEC958 Playback Con Mask'
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=9,iface=MIXER,name='IEC958 Playback Con Mask',index=1
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=15,iface=MIXER,name='IEC958 Playback Con Mask',index=2
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=21,iface=MIXER,name='IEC958 Playback Con Mask',index=3
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=27,iface=MIXER,name='IEC958 Playback Con Mask',index=4
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=4,iface=MIXER,name='IEC958 Playback Pro Mask'
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0x00 AES2=0x00 AES3=0x00]
numid=10,iface=MIXER,name='IEC958 Playback Pro Mask',index=1
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0x00 AES2=0x00 AES3=0x00]
numid=16,iface=MIXER,name='IEC958 Playback Pro Mask',index=2
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0x00 AES2=0x00 AES3=0x00]
numid=22,iface=MIXER,name='IEC958 Playback Pro Mask',index=3
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0x00 AES2=0x00 AES3=0x00]
numid=28,iface=MIXER,name='IEC958 Playback Pro Mask',index=4
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0x00 AES2=0x00 AES3=0x00]
numid=5,iface=MIXER,name='IEC958 Playback Default'
  ; type=IEC958,access=rw------,values=1
[... 84 lines cut ...]
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',1
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',2
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',3
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',4
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
```

Full alsa-info: https://paste.ubuntu.com/p/vPbfM2Tp6Y/
Win10 cirrus info including init verbs: https://paste.ubuntu.com/p/YY9qTnhjsZ/

Thank you for your hard work! :+1:

leifliddy, 2020-01-12:
Thanks for that. Something isn't right with the mixers. If you look at my ```amixer``` output above, I get the same output with just the stock kernel module. With the macbook10,1, there's no master playback switch or volume control created. Do you also get the same output with the stock kernel module, or is it different?

Could you try this:
1.
```echo "options snd_hda_codec_cirrus dyndbg==pmf" > /etc/modprobe.d/cirrus.conf```
**this will enable kernel debug messages for the ```snd_hda_codec_cirrus``` module
2.
then run the install script from this test repo:
https://github.com/leifliddy/macbook10-1-audio-driver-test

3.
reboot

4. then post this output
```dmesg | egrep 'snd|cirrus'```

savoca, 2020-01-12:
amixer contents with stock kernel:
```
numid=40,iface=CARD,name='HDMI/DP,pcm=10 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=16,iface=CARD,name='HDMI/DP,pcm=3 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=22,iface=CARD,name='HDMI/DP,pcm=7 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=28,iface=CARD,name='HDMI/DP,pcm=8 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=34,iface=CARD,name='HDMI/DP,pcm=9 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=12,iface=CARD,name='Headphone Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=10,iface=CARD,name='Internal Mic Phantom Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=11,iface=CARD,name='Mic Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=13,iface=CARD,name='Speaker Phantom Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=9,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=8,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=0,max=127,step=0
  : values=87
  | dBscale-min=-63.50dB,step=0.50dB,mute=0
numid=2,iface=MIXER,name='Line Out Playback Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on
numid=1,iface=MIXER,name='Line Out Playback Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=127,step=0
  : values=127,127
  | dBscale-min=-63.50dB,step=0.50dB,mute=0
numid=7,iface=MIXER,name='Mic Boost Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=3,step=0
  : values=0,0
  | dBscale-min=-20.00dB,step=10.00dB,mute=0
numid=5,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on
numid=4,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=63,step=0
  : values=63,63
  | dBscale-min=-51.00dB,step=1.00dB,mute=0
numid=17,iface=MIXER,name='IEC958 Playback Con Mask'
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=23,iface=MIXER,name='IEC958 Playback Con Mask',index=1
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=29,iface=MIXER,name='IEC958 Playback Con Mask',index=2
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=35,iface=MIXER,name='IEC958 Playback Con Mask',index=3
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=41,iface=MIXER,name='IEC958 Playback Con Mask',index=4
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x0f AES1=0xff AES2=0x00 AES3=0x00]
numid=18,iface=MIXER,name='IEC958 Playback Pro Mask'
[... 244 lines cut ...]
[   38.100476] snd_hda_intel: cs_4208_init_explicit enter
[   38.100478] snd_hda_intel: cs_4208_init snd_hda_gen_init NOT CALLED
[   38.100480] snd_hda_intel: cs_4208_init_explicit end
[   38.100746] snd_hda_intel: playback_pcm_open hook
[   38.100758] snd_hda_codec_cirrus:cs_4208_playback_pcm_open: snd_hda_codec_cirrus hdaudioC0D0: playback_pcm_open nid 0x04 rates 0x00000040 formats 0x0000000000000004
[   38.100760] snd_hda_intel: playing to speakers
[   38.100761] snd_hda_intel: play_a1534 begin
[   38.101083] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 10: 0x00000111 0x00000000
[   38.101537] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 22: 0x00000101 0x00000001
[   38.105375] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 123: 0x00000040 0x00000000
[   38.105703] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 137: 0x00000030 0x00000033
[   38.105765] snd_hda_intel: play_a1534 end
[   38.106124] snd_hda_codec_cirrus:cs_4208_playback_pcm_prepare: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_prepare enter
[   38.118482] snd_hda_intel: command nid cs_4208_playback_pcm_hook BAD HOOK PREPARE init 0 last 1578862626 cur 1578862648
[   38.118621] snd_hda_intel: command nid cs_4208_playback_pcm_hook BAD power check 0x01 2 0
[   38.118623] snd_hda_intel: command nid cs_4208_playback_pcm_hook BAD HOOK PREPARE end
[   38.118632] snd_hda_codec_cirrus:cs_4208_playback_pcm_prepare: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_prepare end
[   48.660389] snd_hda_codec_cirrus:cs_4208_playback_pcm_cleanup: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_cleanup enter
[   48.660396] snd_hda_codec_cirrus:cs_4208_playback_pcm_cleanup: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_cleanup exit
```
Lastly:
```
alex@macbook:~$ amixer sget Master
amixer: Unable to find simple control 'Master',0
```

leifliddy, 2020-01-13:
I found this link today.
https://forum.manjaro.org/t/no-sound-on-macbook-10-1/116882/8

Apparently, this guy is running Manjaro w/ kernel 5.4 on a macbook10,1
And his amixer output looks like this:

```
$ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 43527 [66%] [on]
  Front Right: Playback 43527 [66%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 30419 [46%] [on]
  Front Right: Capture 30419 [46%] [on]
```

leifliddy, 2020-01-13:
Just a quick note about this test project:
https://github.com/leifliddy/macbook10-1-audio-driver-test

I built this driver using the logs generated from a custom AppleHDA kext we installed on marcusenglert's macbook10,1 (using the same process I used to build this driver)
So, the macbook10-1 driver is specifically geared towards the macbook10,1, but there's only very minor differences between these two drivers. The subsystem id's are different, and some of the GPIO masks are different, but that's it. I was able to install the macbook10-1-audio-driver-test driver on my macbook9,1 and it worked. So, these changes weren't functionally significant at all. 
So....back to square one. Once we do get the audio working on a macbook10,1. I'll merge these two projects together. Even though these changes won't be functionally significant, at least the correct subsystem id's and gpio masks will be used.

savoca, 2020-01-14:
I've tried with up to kernel 5.5-rc5, so maybe I'm getting a separate Ubuntu 19.10? I can try 18.04 next, but it doesn't ship with a kernel supporting the touchpad so I use Cubic to replace the kernel:

https://www.linuxuprising.com/2018/07/how-to-customize-ubuntu-or-linux-mint.html

leifliddy, 2020-01-14:
I wouldn't try Ubuntu. marcusenglert is running Ubuntu 18.04 ,and is running into the same issue.
I'm not even sure if this missing 'Master',0 mixer control is the source of the issue or not, but it would be nice to rule it out. 
I'm a long time-Fedora user. Would you be willing to try it out? I could give you clear instructions on how to install Fedora to a usb stick (w/ persistent storage), update the kernel, compile the driver...etc
Should only take 15 min or so to setup everything up.

savoca, 2020-01-14:
I'll give it a shot first! :) Am a long time Ubuntu/Debian user.

leifliddy, 2020-01-14:
Cool, no worries. Let me know how it goes. I sent davijo
https://github.com/davidjo

...logs, alsa-info...etc from marcusenglert's macbook10,1 laptop + my macbook9,1 laptop. 
I'll let you guys know if finds anything of significance.

quiteDEADlee, 2020-03-09:
Hey, you recommended me this way from the Manjaro forums for help with my macbook10,1 (12 inch 2017) that's not playing audio. I ran this and my speakers still aren't working :cry: here's the outputs from my machine: 

ALSA: http://alsa-project.org/db/?f=342443b5e837a1d7c248e303e7a1c482646a6fc9
amixer contents: `numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=58982,58982
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536`
amixer scontents: `Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 58982 [90%] [on]
  Front Right: Playback 58982 [90%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 65536 [100%] [on]
  Front Right: Capture 65536 [100%] [on]`

Any help is much appreciated!

leifliddy, 2020-03-09:
I sent a Fedora 31 live usb stick pre-loaded with the following driver to @marcusenglert 
https://github.com/leifliddy/macbook10-1-audio-driver-test

...and he confirmed that it worked. I'd like him to run a few more tests --like installing this driver on the usb stick (https://github.com/leifliddy/macbook12-audio-driver) and see if that works as well.
**According to Marcus and savoca  neither of these drivers worked on Ubuntu, which I'm not sure why. I think it may be due to an os-level or application-level config issue that's specific to Ubuntu.

So, the only thing I know for sure at this point --is that the speakers work with Fedora 31 w/ the macbook10-1-audio-driver-test driver. 
Can you try installing the macbook10-1-audio-driver-test driver? Just use the same install instructions that you used for this driver, then reboot your system. 
Let me know if that works for you or not.

techguy9, 2020-10-11:
I would really like to see this working on the 10,1 on Fedora and Arch. Ran the install script on Fedora 32 and it currently results in errors on install and no working driver after reboot:
`--2020-10-11 13:17:01--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.8.13.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 199.232.33.176, 2a04:4e42:45::432
Connecting to cdn.kernel.org (cdn.kernel.org)|199.232.33.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.8.13/sound/pci/hda/
linux-5.8.13/sound/pci/hda/Kconfig
linux-5.8.13/sound/pci/hda/Makefile
linux-5.8.13/sound/pci/hda/ca0132_regs.h
linux-5.8.13/sound/pci/hda/hda_auto_parser.c
linux-5.8.13/sound/pci/hda/hda_auto_parser.h
linux-5.8.13/sound/pci/hda/hda_beep.c
linux-5.8.13/sound/pci/hda/hda_beep.h
linux-5.8.13/sound/pci/hda/hda_bind.c
linux-5.8.13/sound/pci/hda/hda_codec.c
linux-5.8.13/sound/pci/hda/hda_controller.c
linux-5.8.13/sound/pci/hda/hda_controller.h
linux-5.8.13/sound/pci/hda/hda_controller_trace.h
linux-5.8.13/sound/pci/hda/hda_eld.c
linux-5.8.13/sound/pci/hda/hda_generic.c
linux-5.8.13/sound/pci/hda/hda_generic.h
linux-5.8.13/sound/pci/hda/hda_hwdep.c
linux-5.8.13/sound/pci/hda/hda_intel.c
linux-5.8.13/sound/pci/hda/hda_intel.h
linux-5.8.13/sound/pci/hda/hda_intel_trace.h
linux-5.8.13/sound/pci/hda/hda_jack.c
linux-5.8.13/sound/pci/hda/hda_jack.h
linux-5.8.13/sound/pci/hda/hda_local.h
linux-5.8.13/sound/pci/hda/hda_proc.c
linux-5.8.13/sound/pci/hda/hda_sysfs.c
linux-5.8.13/sound/pci/hda/hda_tegra.c
linux-5.8.13/sound/pci/hda/hp_x360_helper.c
linux-5.8.13/sound/pci/hda/patch_analog.c
linux-5.8.13/sound/pci/hda/patch_ca0110.c
linux-5.8.13/sound/pci/hda/patch_ca0132.c
linux-5.8.13/sound/pci/hda/patch_cirrus.c
linux-5.8.13/sound/pci/hda/patch_cmedia.c
linux-5.8.13/sound/pci/hda/patch_conexant.c
linux-5.8.13/sound/pci/hda/patch_hdmi.c
linux-5.8.13/sound/pci/hda/patch_realtek.c
linux-5.8.13/sound/pci/hda/patch_si3054.c
linux-5.8.13/sound/pci/hda/patch_sigmatel.c
linux-5.8.13/sound/pci/hda/patch_via.c
linux-5.8.13/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.8.13-200.fc32.x86_64/build M=/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13 modules
make[1]: Entering directory '/usr/src/kernels/5.8.13-200.fc32.x86_64'
  CC [M]  /home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus.o
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus.c:43:18: error: field ‘last_play_time’ has incomplete type
   43 |  struct timespec last_play_time;
      |                  ^~~~~~~~~~~~~~
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus.c:48:18: error: field ‘first_play_time’ has incomplete type
   48 |  struct timespec first_play_time;
      |                  ^~~~~~~~~~~~~~~
In file included from /home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_setup.h:1158,
                 from /home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus.c:533:
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h: In function ‘cs_4208_playback_pcm_hook’:
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h:172:20: error: storage size of ‘curtim’ isn’t known
  172 |    struct timespec curtim;
      |                    ^~~~~~
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h:173:4: error: implicit declaration of function ‘getnstimeofday’ [-Werror=implicit-function-declaration]
  173 |    getnstimeofday(&curtim);
      |    ^~~~~~~~~~~~~~
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h:172:20: warning: unused variable ‘curtim’ [-Wunused-variable]
  172 |    struct timespec curtim;
      |                    ^~~~~~
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h:206:19: error: storage size of ‘curtim’ isn’t known
  206 |   struct timespec curtim;
      |                   ^~~~~~
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h:206:19: warning: unused variable ‘curtim’ [-Wunused-variable]
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h: In function ‘cs_4208_jack_unsol_event’:
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus_a1534_pcm.h:287:10: error: too few arguments to function ‘snd_hda_jack_tbl_get_from_tag’
  287 |  event = snd_hda_jack_tbl_get_from_tag(codec, tag);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus.c:17:
/home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/hda_jack.h:68:1: note: declared here
   68 | snd_hda_jack_tbl_get_from_tag(struct hda_codec *codec,
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13/patch_cirrus.o] Error 1
make[1]: *** [Makefile:1752: /home/sebastian/macbook10-1-audio-driver-test/build/hda-5.8.13] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.8.13-200.fc32.x86_64'
make: *** [Makefile:5: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.8.13-200.fc32.x86_64/updates/snd-hda-codec-cirrus.ko_speaker
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:10: install] Error 1

contents of /lib/modules/5.8.13-200.fc32.x86_64/updates
total 0
`
I'm willing to help in any way to have sound working on it and have it functional across distros.

leifliddy, 2020-10-11:
@techguy9 Haven't updated that ```macbook10-1-audio-driver-test``` repo in a while as it seems like savoca and marcusenglert  have dropped off the map. Ok, I've just updated it. I've gotten rid the debug versions of the driver files I was using before. 
If you want to see the ```codec_dbg``` statements in the kernel log then create the following modprobe file: 
```echo "options snd_hda_codec_cirrus dyndbg==pmf" > /etc/modprobe.d/cirrus.conf```

It would also be helpful if you posted the output of the following commands:

```
amixer
amixer contents
amixer scontents
amixer sget Master
pacmd list-sinks
```

techguy9, 2020-10-31:
@leifliddy Sorry for the delay, here are the outputs of the commands after running the test repo script on Fedora 33 and restarting (with Analog Stereo Output as the set configuration):
```
$ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 65536 [100%] [on]
  Front Right: Playback 65536 [100%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 65536 [100%] [on]
  Front Right: Capture 65536 [100%] [on]
```

```
$ amixer contents
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
```

```
$ amixer scontents
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 65536 [100%] [on]
  Front Right: Playback 65536 [100%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 65536 [100%] [on]
  Front Right: Capture 65536 [100%] [on]
```

```
$ amixer sget Master
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 65536 [100%] [on]
  Front Right: Playback 65536 [100%] [on]
```

```
$ pacmd list-sinks
1 sink(s) available.
  * index: 6
	name: <alsa_output.pci-0000_00_1f.3.analog-stereo>
	driver: <module-alsa-card.c>
	flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
	state: SUSPENDED
[... 294 lines cut ...]
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_open: snd_hda_codec_cirrus hdaudioC0D0: playback_pcm_open nid 0x04 rates 0x00000040 formats 0x0000000000000004
21:01:54 kernel: snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
21:01:54 kernel: snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 137: 0x00000000 0x00000033
21:01:54 kernel: snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 begin
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_open: snd_hda_codec_cirrus hdaudioC0D0: playback_pcm_open nid 0x04 rates 0x00000040 formats 0x0000000000000004
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_cleanup: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_cleanup end
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_prepare: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_prepare end
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_hook: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook end
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_prepare: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_prepare start
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_open: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_open end
21:01:54 kernel: snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
21:01:54 kernel: snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 137: 0x00000030 0x00000033
21:01:54 kernel: snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 begin
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_playback_pcm_open: snd_hda_codec_cirrus hdaudioC0D0: playback_pcm_open nid 0x04 rates 0x00000040 formats 0x0000000000000004
21:01:54 kernel: snd_hda_codec_cirrus:cs_4208_init_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_init_explicit end
21:01:44 kernel: snd_hda_codec_cirrus:cs_4208_build_controls_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_build_controls_explicit end
21:01:44 kernel: snd_hda_codec_cirrus:cs_4208_init_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_init_explicit end
21:01:44 kernel: snd_hda_codec_cirrus:cs_4208_build_pcms_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_build_pcms_explicit end
21:01:44 kernel: snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
21:01:44 kernel: snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
21:01:44 kernel: snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 end
21:01:44 kernel: snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 1102: 0x00000003 0x00000002
21:01:44 kernel: snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 start
21:01:44 kernel: snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
```

leifliddy, 2020-10-31:
Not sure why your dmesg output is in reverse order. 

But, here's what it looks like on the macbook9,1

BTW, I have some additional debug options enabled in
**/etc/modprobe.d/cirrus.conf**
```
options snd_hda_codec_cirrus dyndbg==pmf
options snd_hda_codec dyndbg==pmf
options snd_hda_intel dyndbg==pmf 
```
dmesg output
```
snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 start
snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 24: 0x00000001 0x00000000
snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 451: 0x00000040 0x00000000
snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 1335: 0x00000111 0x00000000
snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 end
snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 begin
snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
snd_hda_codec:apply_fixup: snd_hda_codec_cirrus hdaudioC0D0: CS4208: Apply fix-func for (null)
snd_hda_codec:apply_fixup: snd_hda_codec_cirrus hdaudioC0D0: CS4208: Apply fix-func for (null)
snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
snd_hda_codec_cirrus hdaudioC0D0:    inputs:
snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
snd_hda_codec:apply_fixup: snd_hda_codec_cirrus hdaudioC0D0: CS4208: Apply fix-func for (null)
```

vs your macbook10,1:
```
21:01:44 kernel: snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
21:01:44 kernel: snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 start
21:01:44 kernel: snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 1102: 0x00000003 0x00000002
21:01:44 kernel: snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 end
21:01:44 kernel: snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
21:01:44 kernel: snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
```

**it looks like it's not finding the speaker pinout. 

which is odd considering the savoca's dmesg output on his macbook10,1 was 
```
snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
snd_hda_codec_cirrus hdaudioC0D0:    inputs:
snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
```

It also looks like the mb10,1 failing to read the return value (from seq number 1102), which probably shouldn't be happening. 
```
retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_GPIO_DATA, 0x00000000, 0x00000002, 1102);
```
**The mb9,1 isn't failing to read any ```AC_VERB_GET_GPIO_DATA``` return values

the differences between the mb9,1 and 10,1 in this area are:
```
-	snd_hda_codec_write(codec, codec->core.afg, 0, AC_VERB_SET_GPIO_MASK, 0x00000004); // 0x00171604
-	retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_GPIO_DATA, 0x00000000, 0x00000004, 1102); // 0x001f1500
-	snd_hda_codec_write(codec, codec->core.afg, 0, AC_VERB_SET_GPIO_MASK, 0x00000006); // 0x00171606
-	retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_GPIO_DATA, 0x00000000, 0x00000006, 1105); // 0x001f1500
+	snd_hda_codec_write(codec, codec->core.afg, 0, AC_VERB_SET_GPIO_MASK, 0x00000002); // 0x00171602
+	retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_GPIO_DATA, 0x00000000, 0x00000002, 1102); // 0x001f1500
```

I don't own a macbook10,1 and had rely on someone else to provide me me with the logging data that was necessary to create this function with. So, I'm not able to verify whether this data is correct or not.
Unfortunately, I just don't have much time to dedicate towards this at the moment.

xaiki, 2021-01-08:
already discussed this with @leifliddy by email:
I can confirm that sound works on Macbook10,1 on Debian with kernel 5.10 and the code in this repository.
I can also confirm that it works with this code: https://github.com/leifliddy/macbook10-1-audio-driver-test

(probably this bug should be renamed to Macbook10,1 issues, and closed)

for completness and futher reference, i'm attaching my logs:
### MAINLINE KERNEL (i.e. no patch)
#### dmesg
```
Linux version 5.10.0-1-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-3) 10.2.1 20201224, GNU ld (GNU Binutils for Debian) 2.35.1) #1 SMP Debian 5.10.4-1 (2020-12-31)
Command line: BOOT_IMAGE=/vmlinuz-5.10.0-1-amd64 root=/dev/mapper/kithira--vg-root ro quiet
x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
BIOS-provided physical RAM map:
BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
BIOS-e820: [mem 0x0000000000100000-0x0000000083dc1fff] usable
BIOS-e820: [mem 0x0000000083dc2000-0x0000000083fc3fff] reserved
BIOS-e820: [mem 0x0000000083fc4000-0x0000000084cbafff] usable
BIOS-e820: [mem 0x0000000084cbb000-0x0000000084cbbfff] ACPI NVS
BIOS-e820: [mem 0x0000000084cbc000-0x0000000084cbcfff] reserved
BIOS-e820: [mem 0x0000000084cbd000-0x0000000084f4dfff] usable
BIOS-e820: [mem 0x0000000084f4e000-0x0000000084f4efff] reserved
BIOS-e820: [mem 0x0000000084f4f000-0x000000008ab9efff] usable
BIOS-e820: [mem 0x000000008ab9f000-0x000000008ac0efff] type 20
BIOS-e820: [mem 0x000000008ac0f000-0x000000008af2efff] reserved
BIOS-e820: [mem 0x000000008af2f000-0x000000008af7efff] ACPI NVS
BIOS-e820: [mem 0x000000008af7f000-0x000000008affefff] ACPI data
BIOS-e820: [mem 0x000000008afff000-0x000000008affffff] usable
BIOS-e820: [mem 0x000000008b000000-0x000000008fffffff] reserved
BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
BIOS-e820: [mem 0x00000000ff939000-0x00000000ff968fff] reserved
BIOS-e820: [mem 0x0000000100000000-0x000000046effffff] usable
NX (Execute Disable) protection: active
efi: EFI v2.40 by Apple
efi: ACPI=0x8affe000 ACPI 2.0=0x8affe014 SMBIOS=0x8af00000 SMBIOS 3.0=0x8aefe000 
secureboot: Secure boot could not be determined (mode 0)
SMBIOS 3.0.0 present.
DMI: Apple Inc. MacBook10,1/Mac-EE2EBD4B90B839A8, BIOS 429.60.3.0.0 11/03/2020
tsc: Detected 1400.000 MHz processor
tsc: Detected 1399.946 MHz TSC
e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
e820: remove [mem 0x000a0000-0x000fffff] usable
last_pfn = 0x46f000 max_arch_pfn = 0x400000000
MTRR default type: write-back
MTRR fixed ranges enabled:
  00000-9FFFF write-back
  A0000-FFFFF uncachable
MTRR variable ranges enabled:
  0 base 00C0000000 mask 7FC0000000 uncachable
  1 base 00A0000000 mask 7FE0000000 uncachable
  2 base 0090000000 mask 7FF0000000 uncachable
  3 base 008C000000 mask 7FFC000000 uncachable
  4 base 008B000000 mask 7FFF000000 uncachable
  5 base 4000000000 mask 4000000000 uncachable
  6 disabled
  7 disabled
  8 disabled
  9 disabled
[... 3003 lines cut ...]
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook BAD power check 0x01 2 0
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_PREPARE end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_CLOSE start
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_CLOSE end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_prepare end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_cleanup start
snd_hda_codec_cirrus hdaudioC0D0: hda_codec_cleanup_stream: NID=0x4
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_cleanup end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_open start
snd_hda_codec_cirrus hdaudioC0D0: playback_pcm_open nid 0x04 rates 0x00000040 formats 0x0000000000000004
snd_hda_codec_cirrus hdaudioC0D0: play_a1534 begin
snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 10: 0x00000111 0x00000000
snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 22: 0x00000101 0x00000001
snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 123: 0x00000040 0x00000000
snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_open end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_prepare start
snd_hda_codec_cirrus hdaudioC0D0: hda_codec_setup_stream: NID=0x4, stream=0x1, channel=0, format=0x4013
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_PREPARE start
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook BAD HOOK PREPARE init 0 last 1610090349 cur 1610090351
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook BAD power check 0x01 2 0
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_PREPARE end
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_CLOSE start
snd_hda_codec_cirrus hdaudioC0D0: cs_4208_playback_pcm_hook HDA_GEN_PCM_ACT_

xaiki, 2021-01-08:
please ignore the PNP usb sound device, the dongle i'm using to boot debian has a soundcard

xaiki, 2021-01-08:
as you can see there aren't big differences in the alsa side between the 10-1 and this driver:
```diff
┌[~/src/Mac/logs]
└[kithira] diff -Nru main.driver.log 10-1.driver.log                    11:41:22
--- main.driver.log	2021-01-08 04:28:13.884694862 -0300
+++ 10-1.driver.log	2021-01-08 04:25:52.017348898 -0300
@@ -48,7 +48,7 @@
   Rear Left: Playback 32826 [50%] [on]
   Rear Right: Playback 32826 [50%] [on]
 2 sink(s) available.
-    index: 0
+    index: 1
 	name: <alsa_output.usb-0c76_USB_PnP_Audio_Device-00.analog-stereo>
 	driver: <module-alsa-card.c>
 	flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
@@ -63,15 +63,15 @@
 	current latency: 0.00 ms
 	max request: 0 KiB
 	max rewind: 0 KiB
-	monitor source: 0
+	monitor source: 1
 	sample spec: s16le 2ch 48000Hz
 	channel map: front-left,front-right
 	             Stereo
 	used by: 0
 	linked by: 0
 	configured latency: 0.00 ms; range is 0.50 .. 2000.00 ms
-	card: 0 <alsa_card.usb-0c76_USB_PnP_Audio_Device-00>
-	module: 6
+	card: 1 <alsa_card.usb-0c76_USB_PnP_Audio_Device-00>
+	module: 22
 	properties:
 		alsa.resolution_bits = "16"
 		device.api = "alsa"
@@ -103,6 +103,8 @@
 		device.profile.name = "analog-stereo"
 		device.profile.description = "Analog Stereo"
 		device.description = "USB PnP Audio Device Analog Stereo"
+		alsa.mixer_name = "USB Mixer"
+		alsa.components = "USB0c76:161f"
 		module-udev-detect.discovered = "1"
 		device.icon_name = "audio-card-usb"
 	ports:
@@ -110,7 +112,7 @@
 			properties:
 				device.icon_name = "audio-speakers"
 	active port: <analog-output-speaker>
-  * index: 1
+  * index: 3
 	name: <alsa_output.pci-0000_00_1f.3.analog-surround-40>
 	driver: <module-alsa-card.c>
 	flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
@@ -125,15 +127,15 @@
 	current latency: 0.00 ms
 	max request: 0 KiB
 	max rewind: 0 KiB
-	monitor source: 2
+	monitor source: 4
 	sample spec: s16le 4ch 44100Hz
 	channel map: front-left,front-right,rear-left,rear-right
 	             Surround 4.0
 	used by: 0
 	linked by: 0
 	configured latency: 0.00 ms; range is 0.50 .. 2000.00 ms
-	card: 1 <alsa_card.pci-0000_00_1f.3>
-	module: 7
+	card: 0 <alsa_card.pci-0000_00_1f.3>
+	module: 6
 	properties:
 		alsa.resolution_bits = "16"
 		device.api = "alsa"
@@ -164,6 +166,8 @@
 		device.profile.name = "analog-surround-40"
 		device.profile.description = "Analog Surround 4.0"
 		device.description = "Built-in Audio Analog Surround 4.0"
+		alsa.mixer_name = "Cirrus Logic CS4208"
+		alsa.components = "HDA:10134208,106b6600,00100401 HDA:8086280b,80860101,00100000"
 		module-udev-detect.discovered = "1"
 		device.icon_name = "audio-card-pci"
 	ports:
```

xaiki, 2021-01-08:
on the kernel side of thing I'm seeing interesting similarities and differences, for instance, the calls to `hda_codec_setup_stream` have a NID of 0x04 on the working module and 0x06 on the mainline one.

tulcas, 2021-01-14:
The microphone does not work on Fedora 33, i tried to change te codec and not worl the same
❯ cat /proc/asound/card*/codec* | grep Codec

Codec: Intel Haswell HDMI
Codec: Cirrus Logic CS4208

❯ cat /etc/asound.conf
#
# Place your global alsa-lib configuration here...
#
options snd-hda-intel model=mbp11

leifliddy, 2021-01-14:
The microphone doesn't work with this driver. Should probably update the README with that info.

closed 2021-07-31 by leifliddy