leifliddy/macbook12-audio-driver #7: Single channel only on Ubuntu 18.04 Macbook Pro 14,3

WeiPhil, 2020-03-29:
Hi, 
First of all thank you for that awesome driver! I could successfully install on my kernel 5.3.0-42 on Ubuntu 18.04 LTS. It works partially as the sound seems to be coming only from one channel. When looking at the alsamixer I have only one PCM bar. And when going to the sound options and testing the sound only "front left" does work and the 3 other channel have no feedback.
Any idea where this could come from?

Best,
Philippe

leifliddy, 2020-03-31:
This repo is for the macbook 12 inch models. 
For the MBP 14,3 use this repo:
https://github.com/leifliddy/snd_hda_macbookpro

I've you've already run the ```install.cirrrus.sh``` script from this repo, then you'll need to 
remove this line:

**/etc/pulse/daemon.conf**
```
default-sample-channels = 4  #remove or comment out this line 
```

Let me know if that works for you.

WeiPhil, 2020-04-05:
Hi, Thank you for your answer!
I actually installed the driver from snd_hda_macbookpro, sorry for the mix up. 
I did the edit you mentioned and I now get only two channels in the settings, however still only the left speaker seems to work. I also noticed that when playing any music or sound it cuts out parts and staggers.
Any command outputs that could help me debug those problems?

leifliddy, 2020-04-05:
That seems to be a common issue with that driver:
https://github.com/Dunedan/mbp-2016-linux/issues/136
https://github.com/davidjo/snd_hda_macbookpro/issues/6

Davidjo is actual developer for that driver (I just wrote an installer for it), it might a be good idea to create an an issue (or post to an existing issue) on his repo. Hopefully, he'll release an update later this year.

closed 2020-04-22 by leifliddy