leifliddy/macbook12-audio-driver #8: 2015 12" MacBook8,1 - Ubuntu Debugging

john-morton, 2020-05-06:
@leifliddy - unfortunately, I have no sound after performing the compilation and installation on my 2015 12" MacBook 8,1 with a freshly installed Ubuntu 20.04 operating system (kernel 5.4).

I have uploaded my ALSA information here, hoping it'll help the debugging process:

http://alsa-project.org/db/?f=af42c4fe0967e9a5f6aa00061486eea31db14c16

I notice that changes have been made within Settings\Sound now, and I see the following:

Output\Output Device: Analog Output - Built-in Audio (single item)
Output\Configuration\Analog: Surround 4.0 Output (no audio on any of the four output tests)
Output\Configuration\Analog: Stereo Output (no audio on either of the two output tests)
Output\Balance: Left-Right

Input\Input Device: [Nothing]
Input\Configuration: [Nothing]
Input\Volume: [Slider]

I'm more than happy to assist where possible, please just let me know.

Just in case you might need it, here is the compilation/installation output:

```
root@MacBook:/home/sound/macbook12-audio-driver# ./install.cirrus.driver.sh 
--2020-05-05 00:33:50--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.4.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.165.176, 2a04:4e42:27::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.165.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-05-05 00:33:50 ERROR 404: Not Found.

--2020-05-05 00:33:50--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.4.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.165.176, 2a04:4e42:27::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.165.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 109441440 (104M) [application/x-xz]
Saving to: ‘build/linux-5.4.tar.xz’

linux-5.4.tar.xz                100%[=====================================================>] 104.37M  12.4MB/s    in 8.4s    

2020-05-05 00:34:07 (12.4 MB/s) - ‘build/linux-5.4.tar.xz’ saved [109441440/109441440]

linux-5.4/sound/pci/hda/
linux-5.4/sound/pci/hda/Kconfig
linux-5.4/sound/pci/hda/Makefile
linux-5.4/sound/pci/hda/ca0132_regs.h
linux-5.4/sound/pci/hda/hda_auto_parser.c
linux-5.4/sound/pci/hda/hda_auto_parser.h
linux-5.4/sound/pci/hda/hda_beep.c
linux-5.4/sound/pci/hda/hda_beep.h
linux-5.4/sound/pci/hda/hda_bind.c
linux-5.4/sound/pci/hda/hda_codec.c
linux-5.4/sound/pci/hda/hda_controller.c
linux-5.4/sound/pci/hda/hda_controller.h
linux-5.4/sound/pci/hda/hda_controller_trace.h
linux-5.4/sound/pci/hda/hda_eld.c
linux-5.4/sound/pci/hda/hda_generic.c
linux-5.4/sound/pci/hda/hda_generic.h
linux-5.4/sound/pci/hda/hda_hwdep.c
linux-5.4/sound/pci/hda/hda_intel.c
linux-5.4/sound/pci/hda/hda_intel.h
linux-5.4/sound/pci/hda/hda_intel_trace.h
linux-5.4/sound/pci/hda/hda_jack.c
linux-5.4/sound/pci/hda/hda_jack.h
linux-5.4/sound/pci/hda/hda_local.h
linux-5.4/sound/pci/hda/hda_proc.c
linux-5.4/sound/pci/hda/hda_sysfs.c
linux-5.4/sound/pci/hda/hda_tegra.c
linux-5.4/sound/pci/hda/hp_x360_helper.c
linux-5.4/sound/pci/hda/patch_analog.c
linux-5.4/sound/pci/hda/patch_ca0110.c
linux-5.4/sound/pci/hda/patch_ca0132.c
linux-5.4/sound/pci/hda/patch_cirrus.c
linux-5.4/sound/pci/hda/patch_cmedia.c
linux-5.4/sound/pci/hda/patch_conexant.c
linux-5.4/sound/pci/hda/patch_hdmi.c
linux-5.4/sound/pci/hda/patch_realtek.c
linux-5.4/sound/pci/hda/patch_si3054.c
linux-5.4/sound/pci/hda/patch_sigmatel.c
linux-5.4/sound/pci/hda/patch_via.c
linux-5.4/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.4.0-28-generic/build M=/home/sound/macbook12-audio-driver/build/hda-5.4.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  CC [M]  /home/sound/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus.o
  LD [M]  /home/sound/macbook12-audio-driver/build/hda-5.4.0/snd-hda-codec-cirrus.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/sound/macbook12-audio-driver/build/hda-5.4.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/sound/macbook12-audio-driver/build/hda-5.4.0/snd-hda-codec-cirrus.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-28-generic/updates/snd-hda-codec-cirrus.ko_speaker
ln -sf /lib/modules/5.4.0-28-generic/updates/snd-hda-codec-cirrus.ko_speaker /lib/modules/5.4.0-28-generic/updates/snd-hda-codec-cirrus.ko
symlinks -c /lib/modules/5.4.0-28-generic/updates 	# relative path links are easier to follow
absolute: /lib/modules/5.4.0-28-generic/updates/snd-hda-codec-cirrus.ko -> /lib/modules/5.4.0-28-generic/updates/snd-hda-codec-cirrus.ko_speaker
changed:  /lib/modules/5.4.0-28-generic/updates/snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
depmod -a
#uncomment these lines if /etc/pulse/default.pa has been modified with the changes listed in pulse_audio_configs/default.pa
#killall alsactl &> /dev/null
#modprobe -r snd_hda_intel
#modprobe -r snd_hda_codec_cirrus
#modprobe snd_hda_codec_cirrus
#modprobe snd_hda_intel
#sleep 2
#killall pulseaudio &> /dev/null
#sleep 2

contents of /lib/modules/5.4.0-28-generic/updates
total 128
lrwxrwxrwx 1 root root     31 May  5 00:34 snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
-rw-r--r-- 1 root root 129400 May  5 00:34 snd-hda-codec-cirrus.ko_speaker

modifying /etc/pulse/daemon.conf
default-sample-channels = 4
root@MacBook:/home/sound/macbook12-audio-driver#
```

leifliddy, 2020-05-09:
Did the audio work before? I don't know that anyone's actually tested this driver on a 2015 model.

leifliddy, 2020-05-15:
I can help you get the audio working, but I'll need you to collect some data from your system.

john-morton, 2020-05-16:
I've never had any audio from this 2015 12" MacBook 8,1 - other than via MacOS or Windows.

I'd be only too happy to assist the debugging efforts. How can I help / what do you need from it?

leifliddy, 2020-05-17:
The easiest way to do this would be for me to send you a bootable usb stick running a modified version of macos --that you would just need to boot from and then run a few commands to collect necessary information needed to create the driver with. First I need to see how much I can compress the cloned disk image by. I'll send you an update on this later.

leifliddy, 2020-05-17:
I was able to compress the dd image down to 19 GB. I split the image up into 17 1.1GB files. Planning on uploading these files to mega.nz. Let me know if this works for you or if you have a better/alternate solution. You would need to get a hold of a 128GB usb stick to write the image to (since that's what I used as my source drive). 

Feel free to email me at leif.liddy@gmail.com

leifliddy, 2020-05-17:
Actually, I could just upload the image files to my web server. 
Email me on the address I sent you earlier and I'll set you up an account.

kedarbramhe, 2020-05-29:
Hi @leifliddy 
I have the same issue on macbook 12" 8,1 .
I have some experience working in C and kernel.
Please let me know if I can be any help .
Thanks

leifliddy, 2020-05-30:
Thanks, I don't have a ton of time to dedicate towards this for next couple of weeks. 
Ok, here's where we're at so far. 

This driver was based the work done by davidjo
https://github.com/davidjo/snd_hda_macbookpro

Using his method of extracting the codec replay commands, I was able to create the 
```setup_a1534``` (commands that occur during the boot process) and 
```play_a1534``` (commands that occur while playing audio) 
functions which are found in ```patch_cirrus_a1534_setup.h```

An outside kernel developer was pretty much responsible for creating everything else: 
creating ```patch_cirrus_a1534_pcm.h``` and modifying ```patch_cirrus.c```

So, I've created a ```macbook8-1-test``` repo
https://github.com/leifliddy/macbook8-1-audio-driver-test

the only difference between this repo and the main repo are the ```setup_a1534``` and ```play_a1534``` functions in ```patch_cirrus_a1534_setup.h```

**I've also created an alternate file: ```patch_cirrus_a1534_setup.h.alt``` that you can also test (in lieu of patch_cirrus_a1534_setup.h). It contains a slighty different variation on the setup_a1534 and play_a1534 functions.

This was because the codec command logs that John sent me didn't match up exactly with the logs that I gathered from the macbook9,1 and macbook10,1 models. It seemed to contain more data then was necessary. 

```patch_cirrus_a1534_setup.h```       is the pared down version
```patch_cirrus_a1534_setup.h.alt```  is the exact representation of John's macbook8,1 logs

I might be of limited use in troubleshooting this on a kernel-level. This driver is far from perfect and has not kept pace with davidjo's recent developments. 

To begin troubleshooting, I would add the following modprobe option:
```
echo "options snd_hda_codec_cirrus dyndbg==pmf" > /etc/modprobe.d/cirrus.conf
```

Then I would just create debug statements in the patch files, and try and sort out why it's not working.

Also, there seems an issue that affects Ubuntu but not Fedora (or Arch linux), and I'm not sure why. 

If you build the macbook8,1-test driver on an Ubuntu system 
A master control won't be created
```
root@MacBook:~# amixer sget Master
amixer: Unable to find simple control 'Master',0
```
**the above outpout is from John's system, although he probably shouldn't have ran that command as root, but as a normal user

It will however create this control on a Fedora system
```
[leif.liddy@mac ~]$ amixer sget Master
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 24163 [37%] [on]
  Front Right: Playback 24163 [37%] [on]
  Rear Left: Playback 24163 [37%] [on]
  Rear Right: Playback 24163 [37%] [on]
```

Let me know if need any additional help. I can try contacting the original kernel developer who worked on this project if need be.

kedarbramhe, 2020-06-19:
@leifliddy Hi there , how can I view the codec command logs ? 
and I guess patch_cirrus_a1534_setup.h.mod is the file you are referring to as patch_cirrus_a1534_setup.h.alt

leifliddy, 2020-11-18:
@kedarbramhe sorry, I must have missed this message. I worked with John Morton over the course of a few weeks in an attempt to get the audio working, but to no avail. If you'd like to try your hand at it, I can give you all the information + tools that you'll need. Let me know if you're still interested.

closed 2021-07-31 by leifliddy