leifliddy/macbook12-audio-driver #10: How do I remove cirrus driver

md1023, 2021-08-17:
Thanks for getting my macbook12 speakers to work, but now I want to go back. Could you tell how to rollback driver installation by `install.cirrus.driver.sh`?

leifliddy, 2021-08-17:
Just remove the module from the updates directory
```
rm /lib/modules/$(uname -r)/updates/snd-hda-codec-cirrus.ko
depmod -a
reboot
```

Or you could just run a script like this to switch between the stock kernel driver and this driver 
```
#!/bin/bash

hda_dir="/lib/modules/$(uname -r)/kernel/sound/pci/hda"
update_dir="/lib/modules/$(uname -r)/updates"

cirrus_headphones=$(find $hda_dir -type f | grep snd-hda-codec-cirrus.ko)  	#module from stock kernel (headphones)  #will find .ko or .ko.xz module
cirrus_speaker="$update_dir/snd-hda-codec-cirrus.ko_speaker"  			#speaker module
cirrus_active="$update_dir/snd-hda-codec-cirrus.ko"  				#active module (symlink)

speaker_module_active=$(readlink -f $cirrus_active | grep '_speaker$')

if [[ $speaker_module_active ]]; then
   echo 'switching to stock kernel driver'
   ln -sf $cirrus_headphones $cirrus_active
else
   echo 'switching to modified speaker driver'
   ln -sf $cirrus_speaker $cirrus_active
fi

depmod -a
```

Just run it once and reboot. Do the same if/when you want to switch back.

md1023, 2021-08-19:
Thank you for quick response!

That's not really a purge script, rather augmented [switch.audio.driver.sh](https://github.com/leifliddy/macbook12-audio-driver/blob/master/support_scripts/switch.audio.driver.sh) script, however it still had some effect.

In fact, I searched why my built-in microphone stopped working and found that I had to set duplex configuration.
[image]
[image]

My TRRS jack from headphones isn't detected however - that's what I originally wanted and still do.

I've changed several kernels since initial driver installation, `ko`-files laid where deleted kernel was no more. When I reinstalled driver for actual kernel - features came back.

closed 2021-08-19 by md1023