leifliddy/macbook12-audio-driver #12: Doesn't work in kernel 5.13 (Ubuntu 21.10) for MBP13,1

yerke, 2021-11-07:
Hello.
Thank you very much for working on this. I just tried it on kernel 5.13 (Ubuntu 21.10) for MBP13,1, and it didn't work unfortunately.
It seems to install correctly, but then I don't see any audio devices after rebooting.
Would it be possible to fix it? Thanks a lot!

leifliddy, 2021-11-07:
This repo only supports the macbook 12-inch 2016 and 2017 models. 
The MBP13,1 has a different audio codec, which I believe is a CS8409 (vs the CS4208 that 12-inch models use)
Davidjo's repo should support the MBP13,1:
https://github.com/davidjo/snd_hda_macbookpro

closed 2021-11-07 by leifliddy

siyan4, 2022-02-08:
May you make this into the main branch?

File: install.*.sh
Modify the 3rd line:
kernel_version=$(uname -r | cut -d '-' -f1) #ie 5.2.8
to:
kernel_version=$(uname -v | cut -d ' ' -f4 | cut -d '-' -f1) #ie 5.2.8

leifliddy, 2022-02-08:
On my system, ```uname -v``` does not show that information

```
[leif.liddy@black ~]$ uname -r
5.16.5-200.fc35.x86_64

[leif.liddy@black ~]$ uname -v
#1 SMP PREEMPT Tue Feb 1 21:37:11 UTC 2022
```