leifliddy/macbook12-audio-driver #13: Mint 20 kernel 5.4

ironhands79, 2022-01-05:
My apologies, as it's clearly stated this won't work on my kernel, but was wondering if there are plans to expand the support? I'm running a MacBook9,1 and haven't seen anything else nearly as complete of a solution for the sound and BT.  Tried installing both anyway but not change, received some compilation errors as well but I assume that's due to my version.  Appreciate the effort you've put into this project!

leifliddy, 2022-01-05:
Anything after kernel 5.0 is supported --so kernel 5.4. is supported. 
Can you post the output of the compilation errors you're seeing. 

Did you install these packages?
```apt install wget symlinks make gcc linux-headers-generic``` (symlinks is optional)

ironhands79, 2022-01-05:
yes, I've installed the package above, and am running the script with sudo.

I will apologize in advance, I'm a linux novice

`make[1]: Entering directory '/usr/src/linux-headers-5.4.0-92-generic'
  CC [M]  /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus.o
In file included from ./include/linux/printk.h:348,
                 from ./include/linux/kernel.h:15,
                 from ./include/asm-generic/bug.h:19,
                 from ./arch/x86/include/asm/bug.h:83,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/slab.h:15,
                 from /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus.c:9:
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_pcm.h: In function ‘cs_4208_playback_pcm_hook’:
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_pcm.h:211:20: warning: format ‘%lld’ expects argument of type ‘long long int’, but argument 5 has type ‘__kernel_time_t’ {aka ‘long int’} [-Wformat=]
  211 |   codec_dbg(codec, "cs_4208_playback_pcm_hook BAD HOOK PREPARE init %d last %lld cur %lld", spec->play_init,spec->last_play_time.tv_sec,curtim.tv_sec);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/dynamic_debug.h:125:15: note: in definition of macro ‘__dynamic_func_call’
  125 |   func(&id, ##__VA_ARGS__);  \
      |               ^~~~~~~~~~~
./include/linux/dynamic_debug.h:157:2: note: in expansion of macro ‘_dynamic_func_call’
  157 |  _dynamic_func_call(fmt,__dynamic_dev_dbg,   \
      |  ^~~~~~~~~~~~~~~~~~
./include/linux/device.h:1811:2: note: in expansion of macro ‘dynamic_dev_dbg’
 1811 |  dynamic_dev_dbg(dev, dev_fmt(fmt), ##__VA_ARGS__)
      |  ^~~~~~~~~~~~~~~
./include/linux/device.h:1811:23: note: in expansion of macro ‘dev_fmt’
 1811 |  dynamic_dev_dbg(dev, dev_fmt(fmt), ##__VA_ARGS__)
      |                       ^~~~~~~
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/hda_local.h:727:2: note: in expansion of macro ‘dev_dbg’
  727 |  dev_dbg(hda_codec_dev(codec), fmt, ##args)
      |  ^~~~~~~
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_pcm.h:211:3: note: in expansion of macro ‘codec_dbg’
  211 |   codec_dbg(codec, "cs_4208_playback_pcm_hook BAD HOOK PREPARE init %d last %lld cur %lld", spec->play_init,spec->last_play_time.tv_sec,curtim.tv_sec);
      |   ^~~~~~~~~
In file included from /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_setup.h:1679,
                 from /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus.c:528:
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_pcm.h:211:80: note: format string is defined here
  211 |   codec_dbg(codec, "cs_4208_playback_pcm_hook BAD HOOK PREPARE init %d last %lld cur %lld", spec->play_init,spec->last_play_time.tv_sec,curtim.tv_sec);
      |                                                                             ~~~^
      |                                                                                |
      |                                                                                long long int
      |                                                                             %ld
In file included from ./include/linux/printk.h:348,
                 from ./include/linux/kernel.h:15,
                 from ./include/asm-generic/bug.h:19,
                 from ./arch/x86/include/asm/bug.h:83,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/slab.h:15,
                 from /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus.c:9:
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_pcm.h:211:20: warning: format ‘%lld’ expects argument of type ‘long long int’, but argument 6 has type ‘__kernel_time_t’ {aka ‘long int’} [-Wformat=]
  211 |   codec_dbg(codec, "cs_4208_playback_pcm_hook BAD HOOK PREPARE init %d last %lld cur %lld", spec->play_init,spec->last_play_time.tv_sec,curtim.tv_sec);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/dynamic_debug.h:125:15: note: in definition of macro ‘__dynamic_func_call’
  125 |   func(&id, ##__VA_ARGS__);  \
      |               ^~~~~~~~~~~
./include/linux/dynamic_debug.h:157:2: note: in expansion of macro ‘_dynamic_func_call’
  157 |  _dynamic_func_call(fmt,__dynamic_dev_dbg,   \
      |  ^~~~~~~~~~~~~~~~~~
./include/linux/device.h:1811:2: note: in expansion of macro ‘dynamic_dev_dbg’
 1811 |  dynamic_dev_dbg(dev, dev_fmt(fmt), ##__VA_ARGS__)
      |  ^~~~~~~~~~~~~~~
./include/linux/device.h:1811:23: note: in expansion of macro ‘dev_fmt’
 1811 |  dynamic_dev_dbg(dev, dev_fmt(fmt), ##__VA_ARGS__)
      |                       ^~~~~~~
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/hda_local.h:727:2: note: in expansion of macro ‘dev_dbg’
[... 5 lines cut ...]
In file included from /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_setup.h:1679,
                 from /home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus.c:528:
/home/ferrus/macbook12-audio-driver/build/hda-5.4.0/patch_cirrus_a1534_pcm.h:211:89: note: format string is defined here
[... cut ...]
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1675: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-92-generic'
make: *** [Makefile:5: all] Error 2
make: command: Command not found
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-92-generic/updates/snd-hda-codec-cirrus.ko_speaker
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:12: install] Error 1

contents of /lib/modules/5.4.0-92-generic/updates
total 244
`

leifliddy, 2022-01-05:
I've created a mint 20 VM and have confirmed the issue. It looks like the issue is due to Mint creating some sort of "Frankenstein" kernel where they've possibly applied patches from newer kernels. So if the kernel that Mint 20.2 was running was actually a ```5.4.0-x``` then it should have absolutely no issue compiling a module using that kernel source. Honestly, I have no desire to attempt to sort out exactly what Mint has done with their kernel.

I just installed kernel 5.13.0 on Mint 20.2
```
root@mint:~# uname -a
Linux mint 5.13.0-23-generic #23~20.04.2-Ubuntu SMP Fri Dec 10 12:06:47 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
and the module compiles and installs no problem

```
root@mint:~/macbook12-audio-driver# ./install.cirrus.driver.sh 
--2022-01-05 21:07:23--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.13.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42:600::432, 2a04:4e42:400::432, 2a04:4e42:200::432, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42:600::432|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 119297284 (114M) [application/x-xz]
Saving to: ‘build/linux-5.13.tar.xz’

linux-5.13.tar.xz             100%[===============================================>] 113,77M  24,4MB/s    in 4,7s    

2022-01-05 21:07:28 (24,1 MB/s) - ‘build/linux-5.13.tar.xz’ saved [119297284/119297284]

linux-5.13/sound/pci/hda/
linux-5.13/sound/pci/hda/Kconfig
linux-5.13/sound/pci/hda/Makefile
linux-5.13/sound/pci/hda/ca0132_regs.h
linux-5.13/sound/pci/hda/hda_auto_parser.c
linux-5.13/sound/pci/hda/hda_auto_parser.h
linux-5.13/sound/pci/hda/hda_beep.c
linux-5.13/sound/pci/hda/hda_beep.h
linux-5.13/sound/pci/hda/hda_bind.c
linux-5.13/sound/pci/hda/hda_codec.c
linux-5.13/sound/pci/hda/hda_controller.c
linux-5.13/sound/pci/hda/hda_controller.h
linux-5.13/sound/pci/hda/hda_controller_trace.h
linux-5.13/sound/pci/hda/hda_eld.c
linux-5.13/sound/pci/hda/hda_generic.c
linux-5.13/sound/pci/hda/hda_generic.h
linux-5.13/sound/pci/hda/hda_hwdep.c
linux-5.13/sound/pci/hda/hda_intel.c
linux-5.13/sound/pci/hda/hda_intel.h
linux-5.13/sound/pci/hda/hda_intel_trace.h
linux-5.13/sound/pci/hda/hda_jack.c
linux-5.13/sound/pci/hda/hda_jack.h
linux-5.13/sound/pci/hda/hda_local.h
linux-5.13/sound/pci/hda/hda_proc.c
linux-5.13/sound/pci/hda/hda_sysfs.c
linux-5.13/sound/pci/hda/hda_tegra.c
linux-5.13/sound/pci/hda/hp_x360_helper.c
linux-5.13/sound/pci/hda/ideapad_s740_helper.c
linux-5.13/sound/pci/hda/patch_analog.c
linux-5.13/sound/pci/hda/patch_ca0110.c
linux-5.13/sound/pci/hda/patch_ca0132.c
linux-5.13/sound/pci/hda/patch_cirrus.c
linux-5.13/sound/pci/hda/patch_cmedia.c
linux-5.13/sound/pci/hda/patch_conexant.c
linux-5.13/sound/pci/hda/patch_hdmi.c
linux-5.13/sound/pci/hda/patch_realtek.c
linux-5.13/sound/pci/hda/patch_si3054.c
linux-5.13/sound/pci/hda/patch_sigmatel.c
linux-5.13/sound/pci/hda/patch_via.c
linux-5.13/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.13.0-23-generic/build M=/root/macbook12-audio-driver/build/hda-5.13.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-23-generic'
  CC [M]  /root/macbook12-audio-driver/build/hda-5.13.0/patch_cirrus.o
  LD [M]  /root/macbook12-audio-driver/build/hda-5.13.0/snd-hda-codec-cirrus.o
  MODPOST /root/macbook12-audio-driver/build/hda-5.13.0/Module.symvers
  CC [M]  /root/macbook12-audio-driver/build/hda-5.13.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /root/macbook12-audio-driver/build/hda-5.13.0/snd-hda-codec-cirrus.ko
  BTF [M] /root/macbook12-audio-driver/build/hda-5.13.0/snd-hda-codec-cirrus.ko
Skipping BTF generation for /root/macbook12-audio-driver/build/hda-5.13.0/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-23-generic'
strip --strip-debug snd-hda-codec-cirrus.ko
make: command: Command not found
cp snd-hda-codec-cirrus.ko /lib/modules/5.13.0-23-generic/updates/snd-hda-codec-cirrus.ko_speaker
ln -sf /lib/modules/5.13.0-23-generic/updates/snd-hda-codec-cirrus.ko_speaker /lib/modules/5.13.0-23-generic/updates/snd-hda-codec-cirrus.ko
depmod -a

contents of /lib/modules/5.13.0-23-generic/updates
total 144
lrwxrwxrwx 1 root root     70 Jan  5 21:07 snd-hda-codec-cirrus.ko -> /lib/modules/5.13.0-23-generic/updates/snd-hda-codec-cirrus.ko_speaker
-rw-r--r-- 1 root root 140904 Jan  5 21:07 snd-hda-codec-cirrus.ko_speaker
```

leifliddy, 2022-01-05:
https://www.makeuseof.com/upgrade-kernel-linux-mint/

ironhands79, 2022-01-05:
case closed, flawless install, thanks so much for your time

closed 2022-01-05 by ironhands79