leifliddy/macbook12-audio-driver #16: switch.audio.driver.sh not working

TaylorHere, 2022-04-18:
switching to headphones
modprobe: FATAL: Module snd_hda_intel is in use.
modprobe: FATAL: Module snd_hda_codec_cirrus is in use.

leifliddy, 2022-04-18:
Unless you've implemented the changes here to ```default.pa``` (provided you're using ```pulse audio```):
https://raw.githubusercontent.com/leifliddy/macbook12-audio-driver/master/pulse_audio_configs/default.pa

....you'll have to reboot after running that script. 

I would just remove the ```reload_driver``` function. ie 

```
#!/bin/bash

hda_dir="/lib/modules/$(uname -r)/kernel/sound/pci/hda"
update_dir="/lib/modules/$(uname -r)/updates"

cirrus_headphones=$(find $hda_dir -type f | grep snd-hda-codec-cirrus.ko)  	#module from stock kernel (headphones)  #will find .ko or .ko.xz module
cirrus_speaker="$update_dir/snd-hda-codec-cirrus.ko_speaker"  			#speaker module
cirrus_active="$update_dir/snd-hda-codec-cirrus.ko"                             #active module (symlink)

speaker_module_active=$(readlink -f $cirrus_active | grep '_speaker$')

if [[ $speaker_module_active ]]; then
   echo 'switching to headphones'
   ln -sf $cirrus_headphones $cirrus_active
else
   echo 'switching to speakers'
   ln -sf $cirrus_speaker $cirrus_active
fi
```
So you'll need to run that script and then reboot the system for the changes to take effect. 
Without the changes to ```default.pa``` the ```snd_hda_codec_cirrus``` driver can't be unloaded without rebooting.

TaylorHere, 2022-04-19:
thx for your response.

closed 2022-04-19 by TaylorHere