leifliddy/macbook12-audio-driver #18: no sound after sleep/resume (KDE)

Madman600, 2022-06-13:
Sound works until I sleep and resume the macbook (April 2016 12" - Version 9,1)

leifliddy, 2022-06-27:
Suspend and resume is working for you with the internal nvme ssd?

Madman600, 2022-07-03:
Yes.

leifliddy, 2022-07-04:
Really!? AFAIK, this issue has never been solved. 
REF:
https://github.com/Dunedan/mbp-2016-linux/issues/133
https://github.com/cb22/macbook12-spi-driver/pull/2
https://github.com/cb22/macbook12-spi-driver/pull/30

Can you please provide the details on how you managed to get suspend/resume working while booted to the nvme drive? 
How are you suspending and resuming the laptop (via the gui, closing the lid, pressing the power button, command line...etc)
Are you using ```grub``` to boot the system?

Can you please post the output of the following commands?
```
cat /sys/class/dmi/id/product_name
cat /sys/class/dmi/id/bios_version
uname -a
grep PRETTY_NAME /etc/os-release
lsblk
```

leifliddy, 2022-07-04:
I'm running kernel ```5.18.9``` and if I suspend via ```systemctl suspend``` and then resume the system -- the filesystems on the nvme drive become read-only and can no longer be written to. This is the same issue we were running into before. As far as I can tell, nothing has changed.

Madman600, 2022-07-04:
I'm booting using efi. 

MacBook10,1
447.80.3.0.0
Linux frank 5.18.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 09 Jun 2022 16:14:10 +0000 x86_64 GNU/Linux
PRETTY_NAME="Arch Linux"
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
nvme0n1     259:0    0 465.9G  0 disk 
├─nvme0n1p1 259:1    0   300M  0 part /boot
└─nvme0n1p2 259:2    0 465.6G  0 part /
nvme0n2     259:3    0     8K  0 disk

leifliddy, 2022-07-05:
So you actually have a MacBook10,1 which is the 2017 model. That shouldn't make a difference though.
\>I'm booting using efi.
Can you be more specific? If you run ```blkid /dev/nvme0n1p1```  what is value of ```TYPE```?
Are you using ```systemd-boot``` as the bootloader?

Madman600, 2022-07-05:
> I used efibootmgr to add UEFI entry to boot directly. I used this fix to
> get result working: https://github.com/Dunedan/mbp-2016-linux/issues/37.

*resume

Madman600, 2022-07-05:
https://github.com/Dunedan/mbp-2016-linux/issues/37#issuecomment-540202102

leifliddy, 2022-07-05:
I've tried that method before. It doesn't work on the Macbook9,1

DumbGameMaker, 2022-07-07:
tried adding `acpiphp.disable=1` to grub?

leifliddy, 2022-07-08:
It's been years since I last revisited this topic. 
So ```acpiphp.disable=1``` doesn't work, but this does
```
echo 0 > /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/nvme/nvme0/device/d3cold_allowed
```
That's pretty cool. So it looks like the audio does not work after resuming from suspend. 
Unfortunately, it's been ages since I've worked on this driver and I simply don't have the desire to develop it any further at the moment.

ronnygroenhaug, 2022-07-25:
Hello there, I am **not** an experienced Linux user, but have managed to get this far on my Early 2016 MacBook - Ubuntu 22.04 installed and (thanks to this github page) I also have sound now, THANK YOU!! I have been looking and trying all sort of things, but this is the one that works for me. As I understand from your comment above, you will no longer work on this project.
As the inexperienced user that I am, what does the echo-command do, and what do you mean by that "its working" ? Can I have this command run on startup, and have sound? 
As of now I head into Terminal and run your magic commands+reboot when I need sound (as it dissapears after sleep and reboot).
Sorry for my poor knowledge.

DumbGameMaker, 2022-07-26:
> The echo command fixes sleep/wake nvme locking by (i think) disabling something in the controller (maybe like a sleep mode?)Sleep mode still will break audio. After a restart audio should work again though.Sent from my iPhone

Also you *should* not have to run the command before a reboot for sound to workSent from my iPhone

leifliddy, 2022-07-26:
\>As I understand from your comment above, you will no longer work on this project.
I'm not a kernel developer and had a ton of help writing this driver. 
I did spend several hours attempting to get resume/suspend working. It's just a really difficult problem (for me anyway), even just reloading the ```snd_hda_codec_cirrus``` module causes it to stop working. 

\>what does the echo-command do
So that echo command prevents the internal nvme drive from going into a low-powered state. Which is the state it normally goes into when suspending a system. So when you run that echo command and then suspend the system, the nvme drive will remain powered on. 

\>As of now I head into Terminal and run your magic commands+reboot when I need sound (as it dissapears after sleep and reboot).

So when you run those command, it's simply create a new sound module here:
```/lib/modules/$(uname -r)/updates/snd-hda-codec-cirrus.ko```

when you suspend and resume the system, yes the sound will no longer work. But simply rebooting the system shouldn't affect anything. Try powering down the system and powering it back on (as opposed to just rebooting it) and see if that works. 

Also anytime you update your kernel, you'll need to run run those magic commands again (on the next reboot) -- as that module needs to be built for every new kernel that's installed.

x5444, 2022-08-07:
Suspend-to-RAM breaks audio on my 2017 MacBook 10,1 the same way as reported.

However, I found that Suspend-to-Idle keeps the driver functional! I don't yet know what exactly that means for power consumption but for me it's a better solution than having to reboot on every lid-open. Overall I'm happy with this. I hope this isn't stating the obvious for you all.

Practically, that means replacing the systemd-suspend.service with a call to something like `echo 0 > /sys/power/state`, since sadly systemd-logind doesn't seem to support Suspend-to-Idle as a target by itself.

I also noticed some weird behavior when closing the lid _while_ audio is playing but I can work around that. Finally, you still seem to need the NVMe fix (disabling it's sleep state) or it'll crash on resume from idle.

Information on the different sleep states can be found in [the Kernel doc](https://www.kernel.org/doc/html/v5.19/admin-guide/pm/sleep-states.html).

Madman600, 2022-10-11:
I used efibootmgr to add UEFI entry to boot directly. I used this fix to
get result working: https://github.com/Dunedan/mbp-2016-linux/issues/37.

Madman600, 2022-10-11:
Ok but how do I make the sound work on resume with your code?

chrispyfur, 2026-01-14:
Hi all. some of these issue might need combining. The only way I can get the speakers working after sleep or a reboot is with a PRAM reset (hold Command, Option P R at boot until you hear the chime) Suspend works fine though under Nix 25.11, here's summary of my config.

MacBook 12" (MacBook9,1 / MacBook10,1) Suspend Fix

Deep suspend (S3) crashes on the MacBook 12". The system either fails to suspend or fails to wake, often requiring a hard power-off.

  Solution

  Two fixes are required for working suspend:

  1. Use s2idle instead of deep suspend (S3)

  Add kernel parameter:
  mem_sleep_default=s2idle

  S3 (deep) suspend seems broken on this hardware - the system crashes during suspend or fails to wake. s2idle (modern standby) works reliably.

  2. Disable NVMe D3cold

  The Apple NVMe controller (device 0000:01:00.0) doesn't wake properly from D3cold power state. Disable it at boot:

  echo 0 > /sys/bus/pci/devices/0000:01:00.0/d3cold_allowed

  Or as a systemd service:
  [Unit]
  Description=Disable D3cold on NVMe for stable suspend

  [Service]
  Type=oneshot
  ExecStart=/bin/bash -c 'echo 0 > /sys/bus/pci/devices/0000:01:00.0/d3cold_allowed'
  RemainAfterExit=true

  [Install]
  WantedBy=multi-user.target

  Optional: Bluetooth resume fix

  The BCM4350 Bluetooth chip often fails to reinitialize after resume. A systemd service to reload the module helps:

  [Unit]
  Description=Reset Bluetooth after resume
  After=suspend.target hibernate.target hybrid-sleep.target

  [Service]
  Type=oneshot
  ExecStart=/bin/bash -c 'sleep 1 && modprobe -r btbcm hci_uart && sleep 1 && modprobe hci_uart'

  [Install]
  WantedBy=suspend.target hibernate.target hybrid-sleep.target

  Tested on

  - MacBook10,1 (2017)
  - NixOS 25.11, kernel 6.1 LTS