leifliddy/macbook12-audio-driver #20: Update to 22.04

eprompc, 2023-01-30:
Hello. Thanks for your work.
Could you update  your kernel to Ubuntu 22.04 ?? 
Doesn't work :(
Thanks

leifliddy, 2023-01-30:
I don't understand what you're asking. You want me to update my kernel??

leifliddy, 2023-01-30:
Sorry, I don't have time for these of issues...
I have no idea what model macbook you have, what kernel you're running, whether the driver is compiling or not...etc. 
`Doesn't work` tells me absolutely nothing.

closed 2023-01-30 by leifliddy

eprompc, 2023-01-30:
I am very sorry, I honestly thought that this message was no longer going to be read due to the age of the code. 

I apologize, I have a Macbook 12" retina 9.1 with Ubuntu 22.04 working perfectly with Suspend and Bluetooth. I just need to get the audio working through the built-in speakers, but your patch doesn't work for me on Ubuntu 22.04. I was wondering if you could help me with an update to your code. I'm not a Linux expert and I don't know what options I could modify. Thank you very much for your reply and sorry for my English. 
Best regards.

eprompc, 2023-01-30:
Nothing works with your script. Neither speakers nor headset, nor running the script to switch between one and the other. I need to uninstall pulseaudio and re-install it again to get the headphones working again. The installation works correctly, but no sound card appears in the configuration. So all I have to do is re-install pulseaudio and again the headphones sound again, but I can't get the speakers to work. Thank you very much for your help.

leifliddy, 2023-01-30:
>I honestly thought that this message was no longer going to be read due to the age of the code.

I haven't update this project in a while, but everything should still work.    
Speaker output is working on my macbook9,1 running Fedora 37  w/ kernel 6.1.8

 >I need to uninstall pulseaudio and re-install it again to get the headphones working again. 

I'm not sure why that would have any effect.   
Did you remove `lib/modules/$(uname -r)/updates/snd-hda-codec-cirrus.ko` or is that still there?
You could try removing everything after these lines in  `install.cirrus.driver.sh` and see if that has any effect.
```
echo -e "\ncontents of $update_dir"
ls -lA $update_dir
```

After you ran `install.cirrus.driver.sh` did you reboot your system so that the new `snd-hda-codec-cirrus.ko` driver could take affect? 

Can you please try the following?
Perform a `git pull` to pull in a couple updates  I just pushed.
Then run  `install.cirrus.driver.sh` and post the output of:

`ls -la /lib/modules/$(uname -r)/updates/`
and
`cat /sys/devices/virtual/dmi/id/board_version`

eprompc, 2023-01-31:
Primer intento:

> Did you remove `lib/modules/$(uname -r)/updates/snd-hda-codec-cirrus.ko` or is that still there? You could try removing everything after these lines in `install.cirrus.driver.sh` and see if that has [...]

Tested. Reboot and no sound.

eprompc, 2023-01-31:
Second try:

> Can you please try the following? Perform a `git pull` to pull in a couple updates I just pushed. Then run `install.cirrus.driver.sh` and post the output of:
> 
> `ls -la /lib/modules/$(uname -r)/updates/` and `cat /sys/devices/virtual/dmi/id/board_version`

Git pull, update 3 files:
Run  /install.cirrus.driver.sh
Reboot and no sound.

> Output of:   ls -la /lib/modules/$(uname -r)/updates/

`total 452
drwxr-xr-x 2 root root   4096 ene 31 08:35 .
drwxr-xr-x 6 root root   4096 ene 31 08:35 ..
-rw-r--r-- 1 root root 301392 ene 30 23:40 hci_uart.ko
lrwxrwxrwx 1 root root     31 ene 31 08:35 snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
-rw-r--r-- 1 root root 143056 ene 31 08:35 snd-hda-codec-cirrus.ko_speaker
`

> Output of:   cat /sys/devices/virtual/dmi/id/board_version

`MacBook9,1`

Thanks for you time and help.

eprompc, 2023-01-31:
`╰─ sudo ./install.cirrus.driver.sh                                                     ─╯

--2023-01-31 09:01:11--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.0.tar.xz
Resolviendo cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.1.176, 151.101.65.176, ...
Conectando con cdn.kernel.org (cdn.kernel.org)[151.101.193.176]:443... conectado.
Petición HTTP enviada, esperando respuesta... 404 Not Found
2023-01-31 09:01:12 ERROR 404: Not Found.

--2023-01-31 09:01:12--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.tar.xz
Resolviendo cdn.kernel.org (cdn.kernel.org)... 151.101.1.176, 151.101.193.176, 151.101.65.176, ...
Conectando con cdn.kernel.org (cdn.kernel.org)[151.101.1.176]:443... conectado.
Petición HTTP enviada, esperando respuesta... 416 Range Not Satisfiable

    El archivo ya se ha obtenido por completo; no hay nada que hacer.

linux-5.15/sound/pci/hda/
linux-5.15/sound/pci/hda/Kconfig
linux-5.15/sound/pci/hda/Makefile
linux-5.15/sound/pci/hda/ca0132_regs.h
linux-5.15/sound/pci/hda/hda_auto_parser.c
linux-5.15/sound/pci/hda/hda_auto_parser.h
linux-5.15/sound/pci/hda/hda_beep.c
linux-5.15/sound/pci/hda/hda_beep.h
linux-5.15/sound/pci/hda/hda_bind.c
linux-5.15/sound/pci/hda/hda_codec.c
linux-5.15/sound/pci/hda/hda_controller.c
linux-5.15/sound/pci/hda/hda_controller.h
linux-5.15/sound/pci/hda/hda_controller_trace.h
linux-5.15/sound/pci/hda/hda_eld.c
linux-5.15/sound/pci/hda/hda_generic.c
linux-5.15/sound/pci/hda/hda_generic.h
linux-5.15/sound/pci/hda/hda_hwdep.c
linux-5.15/sound/pci/hda/hda_intel.c
linux-5.15/sound/pci/hda/hda_intel.h
linux-5.15/sound/pci/hda/hda_intel_trace.h
linux-5.15/sound/pci/hda/hda_jack.c
linux-5.15/sound/pci/hda/hda_jack.h
linux-5.15/sound/pci/hda/hda_local.h
linux-5.15/sound/pci/hda/hda_proc.c
linux-5.15/sound/pci/hda/hda_sysfs.c
linux-5.15/sound/pci/hda/hda_tegra.c
linux-5.15/sound/pci/hda/hp_x360_helper.c
linux-5.15/sound/pci/hda/ideapad_s740_helper.c
linux-5.15/sound/pci/hda/patch_analog.c
linux-5.15/sound/pci/hda/patch_ca0110.c
linux-5.15/sound/pci/hda/patch_ca0132.c
linux-5.15/sound/pci/hda/patch_cirrus.c
linux-5.15/sound/pci/hda/patch_cmedia.c
linux-5.15/sound/pci/hda/patch_conexant.c
linux-5.15/sound/pci/hda/patch_cs8409-tables.c
linux-5.15/sound/pci/hda/patch_cs8409.c
linux-5.15/sound/pci/hda/patch_cs8409.h
linux-5.15/sound/pci/hda/patch_hdmi.c
linux-5.15/sound/pci/hda/patch_realtek.c
linux-5.15/sound/pci/hda/patch_si3054.c
linux-5.15/sound/pci/hda/patch_sigmatel.c
linux-5.15/sound/pci/hda/patch_via.c
linux-5.15/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.15.0-58-generic/build M=/home/eprom/macbook12-audio-driver/build/hda-5.15.0 modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.15.0-58-generic'
  CC [M]  /home/eprom/macbook12-audio-driver/build/hda-5.15.0/patch_cirrus.o
  LD [M]  /home/eprom/macbook12-audio-driver/build/hda-5.15.0/snd-hda-codec-cirrus.o
  MODPOST /home/eprom/macbook12-audio-driver/build/hda-5.15.0/Module.symvers
  CC [M]  /home/eprom/macbook12-audio-driver/build/hda-5.15.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/eprom/macbook12-audio-driver/build/hda-5.15.0/snd-hda-codec-cirrus.ko
  BTF [M] /home/eprom/macbook12-audio-driver/build/hda-5.15.0/snd-hda-codec-cirrus.ko
Skipping BTF generation for /home/eprom/macbook12-audio-driver/build/hda-5.15.0/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: se sale del directorio '/usr/src/linux-headers-5.15.0-58-generic'
strip --strip-debug snd-hda-codec-cirrus.ko
cp snd-hda-codec-cirrus.ko /lib/modules/5.15.0-58-generic/updates/snd-hda-codec-cirrus.ko_speaker
ln -srf /lib/modules/5.15.0-58-generic/updates/snd-hda-codec-cirrus.ko_speaker /lib/modules/5.15.0-58-generic/updates/snd-hda-codec-cirrus.ko
depmod -a

contents of /lib/modules/5.15.0-58-generic/updates
total 444
-rw-r--r-- 1 root root 301392 ene 30 23:40 hci_uart.ko
lrwxrwxrwx 1 root root     31 ene 31 09:01 snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker
-rw-r--r-- 1 root root 143056 ene 31 09:01 snd-hda-codec-cirrus.ko_speaker
`

I see what could be a mistake:

`Skipping BTF generation for /home/eprom/macbook12-audio-driver/build/hda-5.15.0/snd-hda-codec-cirrus.ko due to unavailability of vmlinux`

Thanks

leifliddy, 2023-01-31:
Ok, assuming you've already rebooted. 
If you want to revert the system to using the original `snd-hda-codec-cirrus.ko` module
You can simply do either
```
rm /lib/modules/$(uname -r)/updates/snd-hda-codec-cirrus.ko*
depmod -a
reboot
```

Or run that switch.audio.driver.sh script

```
./support_scripts/switch.audio.driver.sh
reboot
```
This should repoint the symlink like this 
```
ls -lA /lib/modules/$(uname -r)/updates/
lrwxrwxrwx 1 root root     47 Jan 31 10:43 snd-hda-codec-cirrus.ko -> ../kernel/sound/pci/hda/snd-hda-codec-cirrus.ko
-rw-r--r-- 1 root root 142992 Jan 31 10:42 snd-hda-codec-cirrus.ko_speaker
```

I really don't have much time to dedicate towards testing out every distro and setup out there. 
So you're going to have to troubleshoot this on your own.  
I mean I could help you if you're running Fedora -- because that's what I'm familiar with.  
Also, don't worry about that `BTF` msg, it's harmless. 

Here are some things you can do.

1. look at past issues like this one:
https://github.com/leifliddy/macbook12-audio-driver/issues/14

2. run amixer, aplay..etc
You should be seeing something like this -- if it's working.

```
[leif.liddy@mac ~]$ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 30890 [47%] [on]
  Front Right: Playback 30187 [46%] [on]
  Rear Left: Playback 30890 [47%] [on]
  Rear Right: Playback 30187 [46%] [on]
```

```
[leif.liddy@mac ~]$ aplay --list-devices
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: CS4208 Analog [CS4208 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

if you're running pulseaudio, you can also try running
`pacmd list-sinks`

3. `dmesg` (or `journalctl -b`) output

```
[root@mac ~]# dmesg -T | grep snd
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
```

if you want to see more verbose output run the following and reboot

`echo "options snd_hda_codec_cirrus dyndbg==pmf" > /etc/modprobe.d/cirrus.conf`

```
[root@mac ~]# dmesg -T | grep snd
[Tue Jan 31 11:52:22 2023] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 start
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 24: 0x00000001 0x00000000
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 451: 0x00000040 0x00000000
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:snd_hda_codec_read_check: snd_hda_codec_cirrus hdaudioC0D0: command nid BAD read check return value at 1335: 0x00000111 0x00000000
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:setup_a1534: snd_hda_codec_cirrus hdaudioC0D0: setup_a1534 end
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 begin
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:play_a1534: snd_hda_codec_cirrus hdaudioC0D0: play_a1534 end
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:cs_4208_build_pcms_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_build_pcms_explicit start
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:cs_4208_build_pcms_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_build_pcms_explicit end
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:cs_4208_init_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_init_explicit start
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:cs_4208_init_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_init_explicit end
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:cs_4208_build_controls_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_build_controls_explicit start
[Tue Jan 31 11:52:22 2023] snd_hda_codec_cirrus:cs_4208_build_controls_explicit: snd_hda_codec_cirrus hdaudioC0D0: cs_4208_build_controls_explicit end
```

Good luck and I hope you figure this out!

leifliddy, 2023-01-31:
You can also run `alsa-info.sh`  and compare that to the output from my system here:
http://alsa-project.org/db/?f=12c71f05de25ae2f522341739cbd2fdc27387342

eprompc, 2023-02-01:
> I really don't have much time to dedicate towards testing out every distro and setup out there.
> So you're going to have to troubleshoot this on your own.
> I mean I could help you if you're running Fedora -- because that's what I'm familiar with.
> Also, don't worry about that `BTF` msg, it's harmless.

Hi, I understand you perfectly, no problem. I will try to do some tests, and if I see that I don't succeed I will continue using my headphones as I have been doing so far.

The thing is that everything seems to be working fine, I have checked amixer, pulseaudio and pavucontrol and everything seems to be ok. What I discovered today is that when using the device change script I get this error:
`
╰─ sudo ./switch.audio.driver.sh                                                       
switching to headphones
modprobe: FATAL: Module snd_hda_intel is in use.
modprobe: FATAL: Module snd_hda_codec_cirrus is in use.

╰─ sudo ./switch.audio.driver.sh                                                      
switching to speakers
modprobe: FATAL: Module snd_hda_intel is in use.
modprobe: FATAL: Module snd_hda_codec_cirrus is in use.
`
I will continue testing, and if I find the error I will let you know so I can put the solution, because there will be users who use Ubuntu before Fedora. Honestly I have never tried Fedora, but the use of Ubuntu is simply because it is the one we use at work.

Thank you very much for taking some of your time. Best regards

leifliddy, 2023-02-01:
Those errors are harmless as you can't reload the `snd_hda_codec_cirrus` driver (without making config changes to pulseaudio anyway)

To prevent those errors -- you can just modify that script like this.  

```bash
#!/bin/bash

hda_dir="/lib/modules/$(uname -r)/kernel/sound/pci/hda"
update_dir="/lib/modules/$(uname -r)/updates"

cirrus_headphones=$(find $hda_dir -type f | grep snd-hda-codec-cirrus.ko)  	#module from stock kernel (headphones)  #will find .ko or .ko.xz module
cirrus_speaker="$update_dir/snd-hda-codec-cirrus.ko_speaker"  			#speaker module
cirrus_active="$update_dir/snd-hda-codec-cirrus.ko"  				#active module (symlink)

speaker_module_active=$(readlink -f $cirrus_active | grep '_speaker$')

if [[ $speaker_module_active ]]; then
   echo 'switching to headphones'
   ln -srf $cirrus_headphones $cirrus_active
   reload_driver
else
   echo 'switching to speakers'
   ln -srf $cirrus_speaker $cirrus_active
   reload_driver
fi
```

Remember that you need to `reboot` after you run that script for the change to take effect. 

The issue that you're experiencing is not a kernel-related problem. Perhaps Ubuntu has implemented something config-wise (alsa, pulseaudio, blacklisted modules...???) that is causing the speaker sound not to work. 
 
If you look though the issue history there are plenty of people who've managed to get the sound working on either ubuntu or linux mint.

eprompc, 2023-02-05:
Hi again, I've been testing these days and there is something that must be silly and escapes me. 

As I understood you we have the same model of laptop, Macbook 9,1 early 2016. And as I have seen your script works with Ubuntu 18.04. I have tried ubuntu 22.04 - ubuntu 19.04 - Ubuntu 18.04, and Mint 18 and 22. On none of them it works. 

I have even booted the system with MacOS thinking that the speakers were broken, but on MacOS they work perfectly. 

I have seen people in the comments that even got the sound on Ubuntu 22.04. I have reread the installation process 20 times, and repeated the process thoroughly until I got tired, but I really do not know what can be happening because no error is reflected. 

What despair!!!

Continued ...

leifliddy, 2023-02-05:
Did you perform any of the steps that I listed here?
https://github.com/leifliddy/macbook12-audio-driver/issues/20#issuecomment-1410184837

So far you haven't posted the output of any of those commands, or the dmesg log, or the output of `alsa-info.sh`
I'm not sure exactly how I'm meant to help you if you don't post anything. 

Simply stating "it doesn't work" doesn't really provide me with any useful information.

eprompc, 2023-02-05:
You're absolutely right, but I honestly don't see anything unusual.

`╰─ aplay --list-devices                                                                                                             ─╯
**** Lista de PLAYBACK dispositivos hardware ****
tarjeta 0: PCH [HDA Intel PCH], dispositivo 0: CS4208 Analog [CS4208 Analog]
  Subdispositivos: 0/1
  Subdispositivo #0: subdevice #0
tarjeta 0: PCH [HDA Intel PCH], dispositivo 3: HDMI 0 [HDMI 0]
  Subdispositivos: 1/1
  Subdispositivo #0: subdevice #0
tarjeta 0: PCH [HDA Intel PCH], dispositivo 7: HDMI 1 [HDMI 1]
  Subdispositivos: 1/1
  Subdispositivo #0: subdevice #0
tarjeta 0: PCH [HDA Intel PCH], dispositivo 8: HDMI 2 [HDMI 2]
  Subdispositivos: 1/1
  Subdispositivo #0: subdevice #0
tarjeta 0: PCH [HDA Intel PCH], dispositivo 9: HDMI 3 [HDMI 3]
  Subdispositivos: 1/1
  Subdispositivo #0: subdevice #0
tarjeta 0: PCH [HDA Intel PCH], dispositivo 10: HDMI 4 [HDMI 4]
  Subdispositivos: 1/1
  Subdispositivo #0: subdevice #0

`

`╰─ pacmd list-sinks                                                                                                                 ─╯

1 sink(s) available.
  * index: 35
	name: <alsa_output.pci-0000_00_1f.3.analog-stereo>
	driver: <module-alsa-card.c>
	flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
	state: RUNNING
	suspend cause: (none)
	priority: 9039
	volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
	        balance 0.00
	base volume: 65536 / 100% / 0.00 dB
	volume steps: 65537
	muted: no
	current latency: 23.76 ms
	max request: 4 KiB
	max rewind: 4 KiB
	monitor source: 35
	sample spec: s16le 2ch 44100Hz
	channel map: front-left,front-right
	             Estéreo
	used by: 1
	linked by: 1
	configured latency: 23.22 ms; range is 0.50 .. 2000.00 ms
	card: 0 <alsa_card.pci-0000_00_1f.3>
	module: 22
	properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "CS4208 Analog"
		alsa.id = "CS4208 Analog"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0x92820000 irq 37"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1f.3"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "9d70"
		device.product.name = "Sunrise Point-LP HD Audio"
		device.form_factor = "internal"
		device.string = "front:0"
		device.buffering.buffer_size = "352800"
		device.buffering.fragment_size = "176400"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-stereo"
		device.profile.description = "Estéreo analógico"
		device.description = "Audio Interno Estéreo analógico"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	ports:
		analog-output: Analog Output (priority 9900, latency offset 0 usec, available: unknown)
			properties:
				
	active port: <analog-output>
`

eprompc, 2023-02-05:
╰─ sudo dmesg -T | grep snd                                                                                                         ─╯
[sudo] contraseña para eprom: 
[dom feb  5 18:57:32 2023] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
[dom feb  5 18:57:33 2023] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18

eprompc, 2023-02-05:
These two messages are after executing your script and restarting

Thanks!!!!

eprompc, 2023-02-05:
╰─ sudo echo "options snd_hda_codec_cirrus dyndbg==pmf" > /etc/modprobe.d/cirrus.conf                                               ─╯
zsh: permission denied: /etc/modprobe.d/cirrus.conf

leifliddy, 2023-02-05:
```**** Lista de PLAYBACK dispositivos hardware ****
tarjeta 0: PCH [HDA Intel PCH], dispositivo 0: CS4208 Analog [CS4208 Analog]
Subdispositivos: 0/1
```

That should say `Subdispositivos: 1/1` not `Subdispositivos: 0/1`

eprompc, 2023-02-05:
And what do you think I can do to change it?

leifliddy, 2023-02-05:
Also, you have these devices on your system that I don't see on mine. 
```
tarjeta 0: PCH [HDA Intel PCH], dispositivo 9: HDMI 3 [HDMI 3]
Subdispositivos: 1/1
Subdispositivo #0: subdevice #0
tarjeta 0: PCH [HDA Intel PCH], dispositivo 10: HDMI 4 [HDMI 4]
Subdispositivos: 1/1
Subdispositivo #0: subdevice #0
```

>And what do you think I can do to change it?

That's what you're going to have to figure out. 
I would examine your alsa and pulseaudio configs on your system. 
What is Ubuntu doing different in that regard?

eprompc, 2023-02-05:
OK, I'll do some research, thanks for finding out where I should look into it.

leifliddy, 2023-02-05:
Also look at this script from a related-project
https://github.com/davidjo/snd_hda_macbookpro/blob/master/install.cirrus.driver.sh

```
# NOTE for Ubuntu we need to use the distribution kernel sources as they seem
# to be significantly modified from the mainline kernel sources generally with backports from later kernels
# (so far the actual debian kernels seem to be close to mainline kernels)
```
So this project uses the Ubuntu distribution kernel source (vs pulling the source from kernel.org)
You could also try doing that.

eprompc, 2023-02-06:
> So this project uses the Ubuntu distribution kernel source (vs pulling the source from kernel.org)
> You could also try doing that.

But I don't understand what change I would have to make in the script to achieve this.
I will check the alsa and pulseaudio configuration files.

eprompc, 2023-02-06:
This is my alsa config:

╰─ cat /etc/modprobe.d/alsa-base.conf                                                                                               ─╯
# autoloader aliases
install sound-slot-0 /sbin/modprobe snd-card-0
install sound-slot-1 /sbin/modprobe snd-card-1
install sound-slot-2 /sbin/modprobe snd-card-2
install sound-slot-3 /sbin/modprobe snd-card-3
install sound-slot-4 /sbin/modprobe snd-card-4
install sound-slot-5 /sbin/modprobe snd-card-5
install sound-slot-6 /sbin/modprobe snd-card-6
install sound-slot-7 /sbin/modprobe snd-card-7

# Cause optional modules to be loaded above generic modules
install snd /sbin/modprobe --ignore-install snd $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-ioctl32 ; /sbin/modprobe --quiet --use-blacklist snd-seq ; }
#
# Workaround at bug #499695 (reverted in Ubuntu see LP #319505)
install snd-pcm /sbin/modprobe --ignore-install snd-pcm $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-pcm-oss ; : ; }
install snd-mixer /sbin/modprobe --ignore-install snd-mixer $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-mixer-oss ; : ; }
install snd-seq /sbin/modprobe --ignore-install snd-seq $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-seq-midi ; /sbin/modprobe --quiet --use-blacklist snd-seq-oss ; : ; }
#
install snd-rawmidi /sbin/modprobe --ignore-install snd-rawmidi $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-seq-midi ; : ; }
# Cause optional modules to be loaded above sound card driver modules
install snd-emu10k1 /sbin/modprobe --ignore-install snd-emu10k1 $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-emu10k1-synth ; }
install snd-via82xx /sbin/modprobe --ignore-install snd-via82xx $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist snd-seq ; }

# Load saa7134-alsa instead of saa7134 (which gets dragged in by it anyway)
install saa7134 /sbin/modprobe --ignore-install saa7134 $CMDLINE_OPTS && { /sbin/modprobe --quiet --use-blacklist saa7134-alsa ; : ; }
# Prevent abnormal drivers from grabbing index 0
options bt87x index=-2
options cx88_alsa index=-2
options saa7134-alsa index=-2
options snd-atiixp-modem index=-2
options snd-intel8x0m index=-2
options snd-via82xx-modem index=-2
options snd-usb-audio index=-2
options snd-usb-caiaq index=-2
options snd-usb-ua101 index=-2
options snd-usb-us122l index=-2
options snd-usb-usx2y index=-2
# Ubuntu #62691, enable MPU for snd-cmipci
options snd-cmipci mpu_port=0x330 fm_port=0x388
# Keep snd-pcsp from being loaded as first soundcard
options snd-pcsp index=-2
# Keep snd-usb-audio from beeing loaded as first soundcard
options snd-usb-audio index=-2

leifliddy, 2023-02-06:
>But I don't understand what change I would have to make in the script to achieve this.

You could just do something like this 

```bash
[ $(grep '^NAME=' /etc/os-release | grep Ubuntu) ]
isubuntu=$(grep '^NAME=' /etc/os-release | grep Ubuntu)

if [ $isubuntu ]; then

	# NOTE for Ubuntu we need to use the distribution kernel sources as they seem
	# to be significantly modified from the mainline kernel sources generally with backports from later kernels
	# (so far the actual debian kernels seem to be close to mainline kernels)

	# NOTE this will likely NOT work for Ubuntu hwe kernels which are even more highly
        #           modified with extensive backports from later kernel versions
        #           (and in any case there is no linux-source-... package for hwe kernels)

	if [ ! -e /usr/src/linux-source-$kernel_version.tar.bz2 ]; then

		echo "Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-$kernel_version.tar.bz2"
		echo "assuming the linux kernel source package is not installed"
		echo "please install the linux kernel source package:"
		echo "sudo apt install linux-source-$kernel_version"
		echo "NOTE - This does not work for HWE kernels"

		exit 1
        fi

	tar --strip-components=3 -xvf /usr/src/linux-source-$kernel_version.tar.bz2 --directory=build/ linux-source-$kernel_version/sound/pci/hda

else
        # attempt to download linux-x.x.x.tar.xz kernel
        wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir

        if [[ $? -ne 0 ]]; then
        # if first attempt fails, attempt to download linux-x.x.tar.xz kernel
        kernel_version=$major_version.$minor_version
        wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
        fi

        [[ $? -ne 0 ]] && echo "kernel could not be downloaded...exiting" && exit

        tar --strip-components=3 -xvf $build_dir/linux-$kernel_version.tar.xz linux-$kernel_version/sound/pci/hda --directory=build/

fi
```

eprompc, 2023-02-06:
> > But I don't understand what change I would have to make in the script to achieve this.
> 
> You could just do something like this
> 
> ```shell
> [ $(grep '^NAME=' /etc/os-release | grep Ubuntu) ]
> isubuntu=$(grep '^NAME=' /etc/os-release | grep Ubuntu)
> 
> if [ $isubuntu ]; then
> 
> 	# NOTE for Ubuntu we need to use the distribution kernel sources as they seem
> 	# to be significantly modified from the mainline kernel sources generally with backports from later kernels
> 	# (so far the actual debian kernels seem to be close to mainline kernels)
> 
> 	# NOTE this will likely NOT work for Ubuntu hwe kernels which are even more highly
>         #           modified with extensive backports from later kernel versions
>         #           (and in any case there is no linux-source-... package for hwe kernels)
> 
> 
> 	if [ ! -e /usr/src/linux-source-$kernel_version.tar.bz2 ]; then
> 
> 		echo "Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-$kernel_version.tar.bz2"
> 		echo "assuming the linux kernel source package is not installed"
> 		echo "please install the linux kernel source package:"
> 		echo "sudo apt install linux-source-$kernel_version"
> 		echo "NOTE - This does not work for HWE kernels"
> 
> 		exit 1
>         fi
> 
> 	tar --strip-components=3 -xvf /usr/src/linux-source-$kernel_version.tar.bz2 --directory=build/ linux-source-$kernel_version/sound/pci/hda
> 
> else
>         # attempt to download linux-x.x.x.tar.xz kernel
>         wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
> 
>         if [[ $? -ne 0 ]]; then
>         # if first attempt fails, attempt to download linux-x.x.tar.xz kernel
>         kernel_version=$major_version.$minor_version
>         wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
>         fi
> 
>         [[ $? -ne 0 ]] && echo "kernel could not be downloaded...exiting" && exit
> 
>         tar --strip-components=3 -xvf $build_dir/linux-$kernel_version.tar.xz linux-$kernel_version/sound/pci/hda --directory=build/
> 
> fi
> ```
Thanks really, but I'm sorry, but I don't know where in the installation script I have to add that code. I have tried at the beginning and at the end of the script and it gives me errors. Thanks

leifliddy, 2023-02-06:
Just look at the script closely and really examine it. 

The current method to download and extract the kernel source involves these steps

```bash
# attempt to download linux-x.x.x.tar.xz kernel
wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir

if [[ $? -ne 0 ]]; then
# if first attempt fails, attempt to download linux-x.x.tar.xz kernel
kernel_version=$major_version.$minor_version
wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
fi

[[ $? -ne 0 ]] && echo "kernel could not be downloaded...exiting" && exit

tar --strip-components=3 -xvf $build_dir/linux-$kernel_version.tar.xz linux-$kernel_version/sound/pci/hda --directory=build/
```
So that's what needs to be replaced.

eprompc, 2023-02-07:
Perfecto, entendía que más o menos era así, pero me arroja errores en directorios

╰─ sudo ./install.cirrus.driver.sh                                                                                                  ─╯
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
mv: unable to `stat' on `hda': File or directory does not exist mv: unable to `stat' on `build/hda-5.15.0/Makefile': File or directory does not exist mv: unable to `stat' on `build/hda-5.15.0/patch_cirrus.c': File or directory does not exist
cp: target 'build/hda-5.15.0/' is not a directory ./install.cirrus.driver.sh: line 87: cd: build/hda-5.15.0: No file or directory make: *** No target was specified and no makefile was found.  High. make: *** No rule to build target 'install'.  High.

contents of /lib/modules/5.15.0-58-generic/updates 
total 436
-rw-r--r-- 1 root root 301392 feb  3 12:36 hci_uart.ko
lrwxrwxrwx 1 root root     47 feb  6 22:29 snd-hda-codec-cirrus.ko -> ../kernel/sound/pci/hda/snd-hda-codec-cirrus.ko
-rw-r--r-- 1 root root 143056 feb  6 16:56 snd-hda-codec-cirrus.ko_speaker

leifliddy, 2023-02-07:
fair point, I'll sort that out tonight.

eprompc, 2023-02-07:
Thank you very much! hopefully this will solve it!  😁

leifliddy, 2023-02-08:
try this one 
```bash
#!/bin/bash

kernel_version=$(uname -r | cut -d '-' -f1)  #ie 5.2.7
major_version=$(echo $kernel_version | cut -d '.' -f1)
minor_version=$(echo $kernel_version | cut -d '.' -f2)
major_minor=${major_version}${minor_version}

build_dir='build'
update_dir="/lib/modules/$(uname -r)/updates"
patch_dir='patch_cirrus'
hda_dir="$build_dir/hda-$kernel_version"

[[ ! -d $update_dir ]] && mkdir $update_dir
[[ ! -d $build_dir ]] && mkdir $build_dir
[[ -d $hda_dir ]] && rm -rf $hda_dir

isubuntu=$(grep '^NAME=' /etc/os-release | grep Ubuntu)

if [ $isubuntu ]; then

    # NOTE for Ubuntu we need to use the distribution kernel sources as they seem
    # to be significantly modified from the mainline kernel sources generally with backports from later kernels
    # (so far the actual debian kernels seem to be close to mainline kernels)

    # NOTE this will likely NOT work for Ubuntu hwe kernels which are even more highly
    # modified with extensive backports from later kernel versions

    if [ ! -e /usr/src/linux-source-$kernel_version.tar.bz2 ]; then
        echo "Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-$kernel_version.tar.bz2"
        echo "it apears the linux kernel source package is not installed"
        echo "install the linux kernel source package with:"
        echo "sudo apt install linux-source-$kernel_version"
        echo "NOTE - This does not work for HWE kernels"
        exit 1
    fi

    tar --strip-components=3 -xvf /usr/src/linux-source-$kernel_version.tar.bz2 linux-source-$kernel_version/sound/pci/hda --directory=build/

else
    # attempt to download linux-x.x.x.tar.xz kernel
    wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir

    if [[ $? -ne 0 ]]; then
        # if first attempt fails, attempt to download linux-x.x.tar.xz kernel
        kernel_version=$major_version.$minor_version
        wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
    fi

    [[ $? -ne 0 ]] && echo "kernel could not be downloaded...exiting" && exit

    tar --strip-components=3 -xvf $build_dir/linux-$kernel_version.tar.xz linux-$kernel_version/sound/pci/hda --directory=build/

fi

mv hda $hda_dir
mv $hda_dir/Makefile $hda_dir/Makefile.orig
mv $hda_dir/patch_cirrus.c $hda_dir/patch_cirrus.c.orig
cp $patch_dir/Makefile $patch_dir/patch_cirrus.c $patch_dir/patch_cirrus_a1534_setup.h $patch_dir/patch_cirrus_a1534_pcm.h $hda_dir/

# if kernel version is < 5.6 then change
# timespec64 to timespec
# ktime_get_real_ts64 to getnstimeofday
if [ $major_minor -lt 56 ]; then
   sed -i 's/timespec64/timespec/' $hda_dir/patch_cirrus.c
   sed -i 's/timespec64/timespec/' $hda_dir/patch_cirrus_a1534_pcm.h
   sed -i 's/ktime_get_real_ts64/getnstimeofday/' $hda_dir/patch_cirrus_a1534_pcm.h
fi

cd $hda_dir
make
make install

echo -e "\ncontents of $update_dir"
ls -lA $update_dir

# modify /etc/pulse/daemon.conf
# ensure four channels are enabled: 'default-sample-channels = 4'

pulse_daemon_conf='/etc/pulse/daemon.conf'
[[ ! -f $pulse_daemon_conf ]] && exit

default_sample_channels=$(grep '^default-sample-channels.*$' $pulse_daemon_conf)

if [ -n "$default_sample_channels" ]; then
   if [[ "$default_sample_channels" != 'default-sample-channels = 4' ]]; then
	  echo -e "\nmodifying /etc/pulse/daemon.conf\ndefault-sample-channels = 4"
	  sed -i 's/^default-sample-channels.*$/default-sample-channels = 4/' $pulse_daemon_conf
	  killall pulseaudio &> /dev/null
   fi
else
	  echo -e "\nmodifying /etc/pulse/daemon.conf\ndefault-sample-channels = 4"
	  echo 'default-sample-channels = 4' >> $pulse_daemon_conf
	  killall pulseaudio &> /dev/null
fi
```

eprompc, 2023-02-08:
> try this one
> 
> ```shell
> #!/bin/bash
> 
> kernel_version=$(uname -r | cut -d '-' -f1)  #ie 5.2.7
> major_version=$(echo $kernel_version | cut -d '.' -f1)
> minor_version=$(echo $kernel_version | cut -d '.' -f2)
> major_minor=${major_version}${minor_version}
> 
> build_dir='build'
> update_dir="/lib/modules/$(uname -r)/updates"
> patch_dir='patch_cirrus'
> hda_dir="$build_dir/hda-$kernel_version"
> 
> [[ ! -d $update_dir ]] && mkdir $update_dir
> [[ ! -d $build_dir ]] && mkdir $build_dir
> [[ -d $hda_dir ]] && rm -rf $hda_dir
> 
> isubuntu=$(grep '^NAME=' /etc/os-release | grep Ubuntu)
> 
> if [ -n $isubuntu ]; then
> 
>     # NOTE for Ubuntu we need to use the distribution kernel sources as they seem
>     # to be significantly modified from the mainline kernel sources generally with backports from later kernels
>     # (so far the actual debian kernels seem to be close to mainline kernels)
> 
>     # NOTE this will likely NOT work for Ubuntu hwe kernels which are even more highly
>     # modified with extensive backports from later kernel versions
> 
>     if [ ! -e /usr/src/linux-source-$kernel_version.tar.bz2 ]; then
>         echo "Ubuntu linux kernel source not found: /usr/src/linux-source-$kernel_version.tar.bz2"
>         echo "install the kernel source package with:"
>         echo "sudo apt install linux-source-$kernel_version"
>         echo "NOTE - This does not work for HWE kernels"
>         exit 1
>     fi
> 
>     tar --strip-components=3 -xvf /usr/src/linux-source-$kernel_version.tar.bz2 linux-source-$kernel_version/sound/pci/hda --directory=build/
> 
> else
>     # attempt to download linux-x.x.x.tar.xz kernel
>     wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
> 
>     if [[ $? -ne 0 ]]; then
>         # if first attempt fails, attempt to download linux-x.x.tar.xz kernel
>         kernel_version=$major_version.$minor_version
>         wget -c https://cdn.kernel.org/pub/linux/kernel/v$major_version.x/linux-$kernel_version.tar.xz -P $build_dir
>     fi
> 
>     [[ $? -ne 0 ]] && echo "kernel could not be downloaded...exiting" && exit
> 
>     tar --strip-components=3 -xvf $build_dir/linux-$kernel_version.tar.xz linux-$kernel_version/sound/pci/hda --directory=build/
> 
> fi
> 
> mv hda $hda_dir
> mv $hda_dir/Makefile $hda_dir/Makefile.orig
> mv $hda_dir/patch_cirrus.c $hda_dir/patch_cirrus.c.orig
> cp $patch_dir/Makefile $patch_dir/patch_cirrus.c $patch_dir/patch_cirrus_a1534_setup.h $patch_dir/patch_cirrus_a1534_pcm.h $hda_dir/
> 
> # if kernel version is < 5.6 then change
> # timespec64 to timespec
> # ktime_get_real_ts64 to getnstimeofday
> if [ $major_minor -lt 56 ]; then
>    sed -i 's/timespec64/timespec/' $hda_dir/patch_cirrus.c
>    sed -i 's/timespec64/timespec/' $hda_dir/patch_cirrus_a1534_pcm.h
>    sed -i 's/ktime_get_real_ts64/getnstimeofday/' $hda_dir/patch_cirrus_a1534_pcm.h
> fi
> 
> cd $hda_dir
> make
> make install
> 
> echo -e "\ncontents of $update_dir"
> ls -lA $update_dir
> 
> # modify /etc/pulse/daemon.conf
> # ensure four channels are enabled: 'default-sample-channels = 4'
> 
> pulse_daemon_conf='/etc/pulse/daemon.conf'
> [[ ! -f $pulse_daemon_conf ]] && exit
> 
> default_sample_channels=$(grep '^default-sample-channels.*$' $pulse_daemon_conf)
> 
> if [ -n "$default_sample_channels" ]; then
>    if [[ "$default_sample_channels" != 'default-sample-channels = 4' ]]; then
> 	  echo -e "\nmodifying /etc/pulse/daemon.conf\ndefault-sample-channels = 4"
> 	  sed -i 's/^default-sample-channels.*$/default-sample-channels = 4/' $pulse_daemon_conf
> 	  killall pulseaudio &> /dev/null
>    fi
> else
> 	  echo -e "\nmodifying /etc/pulse/daemon.conf\ndefault-sample-channels = 4"
> 	  echo 'default-sample-channels = 4' >> $pulse_daemon_conf
> 	  killall pulseaudio &> /dev/null
> fi
> ```

First of all, thank you very much for your time to help me. Your script works perfectly without errors, but it seems that this is not the solution to my problem. I have created in macbook12-audio-driver folder an executable file 'test.sh' with your code, I run it with sudo ./test.sh and it works without problem, but when I reboot I still have no audio output from the speakers. 

I have been tonight analyzing the /var/log/syslog searching for the word 'audio' and I have found a string that says 'audio is autoconfigured with speaker_outs=0 and another one that says headphones_outs=1'. I'm going to try to find where that setting might be and try to change it. I can't think of anything else, because I have searched the internet for the error : ' card 0: PCH [HDA Intel PCH], device 0: CS4208 Analog [CS4208 Analog] Subdevices: 0/1' but I haven't found anything that can guide me to modify the error.

Anyway, I'm sorry for wasting your time, but I'll keep looking in case someone else might find themselves in the same situation as me.

Thank you.

eprompc, 2023-02-08:
Otra cosa que he apreciado es que cuando instalo el driver, desaparecen todos los dispositivos de entrada. Cuando vuelvo al driver original entonces si tengo sonido de entrada por el micrófono.

eprompc, 2023-02-08:
sudo dmesg | egrep -i "snd|firmware"                                  

[    0.166754] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
[    4.098637] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBook9,1.bin failed with error -2
[    4.100279] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBook9,1.txt failed with error -2
[    4.100471] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2
[    4.391924] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4350/5 wl0: Nov 26 2015 03:48:57 version 7.35.180.133 (r602372) FWID 01-c45b39d6
[    4.976232] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/skl_dmc_ver1_27.bin (v1.27)
[    5.123550] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[    5.235463] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    6.809420] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
[    6.809429] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    6.809433] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
[    6.809436] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[    6.809439] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[    6.809441] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
[    6.809444] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18

eprompc, 2023-02-08:
I have also tried using hdajackretask and when I want to select internal speaker it tells me that it is not connected.

leifliddy, 2023-02-08:
>Another thing I have noticed is that when I install the driver, all input devices disappear. When I go back to the original driver then I do have input sound from the microphone.

Exactly the original driver has microphone and headphone jack support. 

Oh well it was worth a shot. Unfortunately I have no idea what Ubuntu has done to their build config-wise -- and I have no desire to install Ubuntu to try and figure it out (although it's probably something simple)
What you could do try switching from pulse audio to pipewire.  
Or maybe try installing Ubuntu 22.10. 
If all else fails, I would try installing a non-Ubuntu-based distro like Debian, Fedora, or Arch.
Anyways, I hope that helps.