leifliddy/macbook12-audio-driver #21: Linux Kernel 6 Compiles and Installs but No Sound

eggsnblakey, 2023-02-10:
Thanks for making this available. Just a note that the driver appears to correctly compile and install using Linux kernel 6 (current POP_OS 22.04 build as of 10 Feb 2022), but I'm not getting any sound on a 2015 12" MacBook. I'll keep fiddling with it.

eggsnblakey, 2023-02-10:
Haha that would explain it.  Thanks

leifliddy, 2023-02-10:
It's not for lack of trying either. I spent weeks trying to get it to work. There's just something different with that model.

amirian, 2023-07-01:
Would you post any changes about Macbook8,1 here? I tested with Ubuntu 22.04.2 LTS, not working. What's the difference with Macbook9,1?

leifliddy, 2023-07-01:
I don't own a macbook8,1 any longer and can't recall exactly what the differences were -- but they were pretty minor.

megabitsenmzq, 2023-12-09:
This driver also does not work on my 2017 Pro. The old driver snd_hda_macbookpro worked before. I removed the old one and then installed this one. Both dkms and non-dkms methods don't work. The OS shows there is a dummy output, but that's it.

Do you have any ideas 🤔?

System Info:
Linux version 6.1.0-13-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29)

leifliddy, 2023-12-09:
@megabitsenmzq The project only supports the macbook9,1 and 10,1 models -- which are the macbook (2016 - 2017) 12-inch models. 

Only davidjo's project supports the macbook pro models
https://github.com/davidjo/snd_hda_macbookpro
It also supports dkms -- which is something that I wrote for him
https://github.com/davidjo/snd_hda_macbookpro/commit/3c19d07b2068810d86d2d92567d3b42d23715860

But, it looks like it's not documented on the README page. 
Anyways....try his driver out. I'm not sure why you removed `snd_hda_macbookpro` if it was working before. It's not an old driver -- it's the driver for the macbook pro models.

megabitsenmzq, 2023-12-09:
> @megabitsenmzq The project only supports the macbook9,1 and 10,1 models -- which are the macbook (2016 - 2017) 12-inch models.
> 
> Only davidjo's project supports the macbook pro models https://github.com/davidjo/snd_hda_macbookpro It also supports dkms -- which is something that I wrote for him [davidjo/snd_hda_macbookpro@3c19d07](https://github.com/davidjo/snd_hda_macbookpro/commit/3c19d07b2068810d86d2d92567d3b42d23715860)
> 
> But, it looks like it's not documented on the README page. Anyways....try his driver out. I'm not sure why you removed `snd_hda_macbookpro` if it was working before. It's not an old driver -- it's the driver for the macbook pro models.

Got it! Thank you for your reply.