leifliddy/macbook12-audio-driver #24: Note: the Macbook8,1 (2015) is not supported

MaxnoSPB, 2023-10-14:
can you fix it?

SteveOverSea, 2023-11-07:
Have you tried installing the driver anyway? I have a Macbook 8.1. and it works for me

leifliddy, 2023-11-07:
No, but I worked with John Morton over the course of a few weeks -- but he could never seem to get it working. 
 https://github.com/leifliddy/macbook12-audio-driver/issues/8
Multiple people have stated in pervious issues that is didn't work for them.
Could you please run the `alsa-info.sh` script (normally part of the `alsa-utils` package) and post a link to the output?

skyostil, 2024-04-15:
Here's `alsa-info` after installing this driver on a macbook8,1: [alsa-info-macbook8,1.txt](https://gist.github.com/skyostil/6fde8071b869d82496bd1875a0071b22)

Sound isn't working in this case, as in no audio is coming from the speakers regardless of the volume setting.

leifliddy, 2024-04-15:
Have you read the README page. That model isn't supported.

```
Work in progress audio driver for the 12" MacBook that's largely based on davidjo's [snd_hda_macbookpro](https://github.com/davidjo/snd_hda_macbookpro)

macbooks supported: MacBook9,1 (2016) and Macbook10,1 (2017)
Note: the Macbook8,1 (2015) is not supported
```

closed 2024-04-15 by leifliddy

SteveOverSea, 2024-04-26:
I just ran the alsa-info command too and then realized I am actually having a MacBook 9.1 not 8.1., sorry for the confusion