leifliddy/macbook12-audio-driver #25: issue implementing Audio Driver

2bmoJake, 2024-01-06:
Heyy I've installed your cs4208 audio driver and bluetooth driver. But only bluetooth driver works, the audio output still wont work from speakers. But i dont know if i need to enter a command for it to start running?

Or if i shouldnt have alsamixer, pulseaudio or pipewire running at the same time?

These are the steps i did to install 
1. `git clone https://github.com/leifliddy/macbook12-audio-driver.git
cd macbook12-audio-driver/` 
2. `sudo ./install.cirrus.driver.sh -i`
3. `reboot`

To check my install after rebooting -
I wrote `ls` cmd and the audio driver did pop up in there. 
I then wrote `dkms status` and the driver popped up there too as - _macbook12-audio/0.1, 5.15.0-91-generic, x86_64: installed_

But audio from the speakers still didnt work :/  So i went used cmd `inxi -SMA` to check if anything seemed off but no.   **Then** I checked alsamixer and there was now only one output, PCM,  and also there was the same old S/PDIF0,1,2,3 and 4. And S/PDIF being on MM.

I tried this first with 6.2.0-39 kernel, no result but then i timeshifted to before installing the 6.2 kernel. So i tried installing audio driver again using the same method on the 5.15.0-91 kernel. But still no sound.

Im sure of my computer being a Macbook 10,1 with the cs4208 audio chip in it. Running Mint 21.2. Alsamixer, Pulseaudio and Pipewire installed and running.   And leifliddy's Bluetooth Driver is working on the computer.

Thanks in advance :))

Friend 
2b

leifliddy, 2024-01-08:
Can you post the output of `lshw | grep -i 'product: Macbook'`

2bmoJake, 2024-01-08:
product: MacBook10,1

leifliddy, 2024-01-08:
Can you post the output of 
`find /usr/lib | grep snd-hda-codec-cirrus`
and
`uname -a`

leifliddy, 2024-01-08:
It would help if you also posted the output of `alsa-info.sh` somewhere.
Linux Mint uses a lot of older libraries and applications. It wouldn't surprise if it was Mint-related somehow. I would try installing Fedora on a usb drive and see if works with that -- so that we can isolate the issue a bit more.

2bmoJake, 2024-01-09:
`upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.5.1
!!################################

!!Script ran on: Tue Jan  9 12:02:44 UTC 2024

!!Linux Distribution
!!------------------

Linux Mint 21.2 Victoria \n \l DISTRIB_ID=LinuxMint DISTRIB_DESCRIPTION="Linux Mint 21.2 Victoria" NAME="Linux Mint" ID=linuxmint ID_LIKE="ubuntu debian" PRETTY_NAME="Linux Mint 21.2" HOME_URL="https://www.linuxmint.com/" SUPPORT_URL="https://forums.linuxmint.com/" BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/" PRIVACY_POLICY_URL="https://www.linuxmint.com/" UBUNTU_CODENAME=jammy

!!DMI Information
!!---------------

Manufacturer:      Apple Inc.
Product Name:      MacBook10,1
Product Version:   1.0
Firmware Version:  522.0.0.0.0
System SKU:        
Board Vendor:      Apple Inc.
Board Name:        Mac-EE2EBD4B90B839A8

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI0001:00/status 	 15
/sys/bus/acpi/devices/ACPI0008:00/status 	 15
/sys/bus/acpi/devices/APP0001:00/status 	 11
/sys/bus/acpi/devices/APP0002:00/status 	 11
/sys/bus/acpi/devices/APP000D:00/status 	 15
/sys/bus/acpi/devices/APPLE-I2C-HPM0:00/status 	 15
/sys/bus/acpi/devices/APPLE-UART-SSDC:00/status 	 15
/sys/bus/acpi/devices/BCM2E7C:00/status 	 15
/sys/bus/acpi/devices/PNP0C0E:00/status 	 11
/sys/bus/acpi/devices/PNP0C0F:00/status 	 11
/sys/bus/acpi/devices/PNP0C0F:01/status 	 11
/sys/bus/acpi/devices/PNP0C0F:02/status 	 11
/sys/bus/acpi/devices/PNP0C0F:03/status 	 11
/sys/bus/acpi/devices/PNP0C0F:04/status 	 11
/sys/bus/acpi/devices/PNP0C0F:05/status 	 11
/sys/bus/acpi/devices/PNP0C0F:06/status 	 11
/sys/bus/acpi/devices/PNP0C0F:07/status 	 11
/sys/bus/acpi/devices/device:10/status 	 15
/sys/bus/acpi/devices/device:11/status 	 15
/sys/bus/acpi/devices/device:12/status 	 15
/sys/bus/acpi/devices/device:18/status 	 15
/sys/bus/acpi/devices/device:19/status 	 15
/sys/bus/acpi/devices/device:1a/status 	 15
/sys/bus/acpi/devices/device:1b/status 	 15

!!Kernel Information
!!------------------

Kernel release:    5.15.0-91-generic
Operating System:  GNU/Linux
Architecture:      x86_64
Processor:         x86_64
SMP Enabled:       Yes

!!ALSA Version
!!------------

Driver version:     k5.15.0-91-generic
Library version:    
Utilities version:  1.2.6

!!Loaded ALSA modules
!!-------------------

[... 1317 lines cut ...]
[    0.107235] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
--
[    6.583515] intel_rapl_common: Found RAPL domain dram
[    6.627005] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    6.726358] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
[    6.726366] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    6.726370] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
[    6.726373] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[    6.726375] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[    6.726378] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
[    6.726380] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18
[    6.752835] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input7
[    6.755072] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input8
[    6.755412] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input9
[    6.755812] input: HDA Intel PCH HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input10
[    6.756106] input: HDA Intel PCH HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input11
[    6.927791] znvpair: module license 'CDDL' taints kernel.

!!Packages installed
!!--------------------

ii  alsa-topology-conf                         1.2.5.1-2                                  all          ALSA topology configuration files
ii  alsa-ucm-conf                              1.2.6.3-1ubuntu1.9                         all          ALSA Use Case Manager configuration files
ii  alsa-utils                                 1.2.6-1ubuntu1                             amd64        Utilities for configuring and using ALSA
`

2bmoJake, 2024-01-09:
[alsa.txt](https://github.com/leifliddy/macbook12-audio-driver/files/13873728/alsa.txt)

2bmoJake, 2024-01-09:
okay i will try fedora, i tried lmde but it did not work

2bmoJake, 2024-01-09:
ive been discussing the issue here too btw if any info already mentioned would help [https://forums.linuxmint.com/viewtopic.php?p=2416142#p2416142]

leifliddy, 2024-01-09:
The desktop itself doesn't matter. I Fedora cinnamon myself. See if it works with Fedora -- because that would really help to understand where this issue lies. There's either something off with my driver compatibility with the macbook10,1 -- or there's something that's off config-wise with Linux Mint.

2bmoJake, 2024-01-10:
i tried fedora cinnamon and the driver still did not work

2bmoJake, 2024-01-10:
if i would be able to buy a macbook pro, do you have any recommendations on what is the best of the intel mac pros? is there one where i wont run in to these sound issues? preferably one that wont require a audio driver either

vitacell, 2024-01-10:
> if i would be able to buy a macbook pro, do you have any recommendations on what is the best of the intel mac pros? is there one where i wont run in to these sound issues? preferably one that wont require a audio driver either

I bought MacbookAir8,2
https://everymac.com/systems/apple/macbook-air/specs/macbook-air-core-i5-1.6-13-retina-display-true-tone-2019-specs.html
And sounds very good out of the box. Not tested the internal microphone, but my Apple Earpods work great, even the headset mic. That Macintosh just requires a GNU/Linux distro with T2 Linux support. 
https://t2linux.org/
My Macbook Pro late 2013 works great also with any distro.
I am struggling with my iMac16,2 with Cirrus CS4208. Looks like everything work, but missing sound from internal speakers.

2bmoJake, 2024-01-10:
> > if i would be able to buy a macbook pro, do you have any recommendations on what is the best of the intel mac pros? is there one where i wont run in to these sound issues? preferably one that wont require a audio driver either
> 
> I bought MacbookAir8,2 https://everymac.com/systems/apple/macbook-air/specs/macbook-air-core-i5-1.6-13-retina-display-true-tone-2019-specs.html And sounds very good out of the box. Not tested the internal microphone, but my Apple Earpods work great, even the headset mic, and volume adjusting from headset buttons. That Macintosh just requires a GNU/Linux distro with T2 Linux support. https://t2linux.org/ My Macbook Pro late 2013 works great also with any distro. I am struggling with my iMac16,2 with Cirrus CS4208. Looks like everything work, but missing sound from internal speakers.

damn yeah something is off with these cs4208 audio chips man

vitacell, 2024-01-10:
> damn yeah something is off with these cs4208 audio chips man

Well, it's offtopic, but I think that CS4208 is fully working with everyting. It's just sound server, ALSA, pulseaudio, pipewire, don't know how to use all those speakers.  I have everything working, but internal speakers.
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3527

2bmoJake, 2024-01-10:
oh  okay

vitacell, 2024-01-10:
> oh okay yeah i dont know if its the chip itself or what but something is not configured for this chip

Yes, could be some driver problem, or just sound server doesn't know how to use those surrond speakers.

2bmoJake, 2024-01-12:
surround speakers? 
I use analog  output option, should i use the surround option or  why did you  say that?

vitacell, 2024-01-12:
> surround speakers? I use analog output option, should i use the surround option or why did you say that?

At least in iMac 2015 4k, it's just something like 1speaker + 1subwoofer inside one unit. So, you have total of 4 speaker. 
It's something like that: https://www.fixmymac.my/imac-speaker-replacement/
But I think that this systems is some iMacs only.

gc070653, 2024-01-30:
I'm also facing the same problem with the internal speakers of my Mackbook 12 Retina Early 2016 macbook9,1. I tried almost all the distros but so far I've been able to use the facetimeHD camera and the earphones but no sound from speakers even if the bar graph of the equalizer show the sound as it is working . Also following the suggestion I found [here](https://forums.linuxmint.com/viewtopic.php?t=411012) no success. Do you know if anybody succeeds with the audio?
Using HDA-Analizer I discover that in Codec 0 the node ox02 AUD_OUT is set with Mute=true and using hdajackretask the internal speaker entry is empty. Can this help to fix the missing audio?

leifliddy, 2024-01-30:
@gc070653 if the driver was installed, you wouldn't have any audio coming out the headphone jack -- as the headphone functionality is not implemented in this driver. The speakers have been working with this driver for the last 5 years on both my macbook9,1 and macbook10,1
I'm running Fedora, but shouldn't matter as this is a kernel module. 

Can you run the following 
ensure dkms functionality is uninstalled so we can troubleshoot the issue more easily
```
./install.cirrus.driver.sh -u
```

Now run 
```
./install.cirrus.driver.sh
```
and post the full output of that command somewhere (I need to see if the install completed successfully or not -- and where it installed the module to)

Then reboot your system and see if the speakers are working -- if not, post the following:
1. `uname -a`
2. `find /usr/lib | grep snd-hda-codec-cirrus`
3. `journalctl -b | grep snd_hda`

gc070653, 2024-01-31:
**sudo ./install.cirrus.driver.sh**

--2024-01-31 11:09:26--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.1.176, 151.101.65.176, 151.101.129.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.1.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-01-31 11:09:26 ERROR 404: Not Found.

--2024-01-31 11:09:26--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.129.176, 151.101.1.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.193.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.15/sound/pci/hda/
linux-5.15/sound/pci/hda/Kconfig
linux-5.15/sound/pci/hda/Makefile
linux-5.15/sound/pci/hda/ca0132_regs.h
linux-5.15/sound/pci/hda/hda_auto_parser.c
linux-5.15/sound/pci/hda/hda_auto_parser.h
linux-5.15/sound/pci/hda/hda_beep.c
linux-5.15/sound/pci/hda/hda_beep.h
linux-5.15/sound/pci/hda/hda_bind.c
linux-5.15/sound/pci/hda/hda_codec.c
linux-5.15/sound/pci/hda/hda_controller.c
linux-5.15/sound/pci/hda/hda_controller.h
linux-5.15/sound/pci/hda/hda_controller_trace.h
linux-5.15/sound/pci/hda/hda_eld.c
linux-5.15/sound/pci/hda/hda_generic.c
linux-5.15/sound/pci/hda/hda_generic.h
linux-5.15/sound/pci/hda/hda_hwdep.c
linux-5.15/sound/pci/hda/hda_intel.c
linux-5.15/sound/pci/hda/hda_intel.h
linux-5.15/sound/pci/hda/hda_intel_trace.h
linux-5.15/sound/pci/hda/hda_jack.c
linux-5.15/sound/pci/hda/hda_jack.h
linux-5.15/sound/pci/hda/hda_local.h
linux-5.15/sound/pci/hda/hda_proc.c
linux-5.15/sound/pci/hda/hda_sysfs.c
linux-5.15/sound/pci/hda/hda_tegra.c
linux-5.15/sound/pci/hda/hp_x360_helper.c
linux-5.15/sound/pci/hda/ideapad_s740_helper.c
linux-5.15/sound/pci/hda/patch_analog.c
linux-5.15/sound/pci/hda/patch_ca0110.c
linux-5.15/sound/pci/hda/patch_ca0132.c
linux-5.15/sound/pci/hda/patch_cirrus.c
linux-5.15/sound/pci/hda/patch_cmedia.c
linux-5.15/sound/pci/hda/patch_conexant.c
linux-5.15/sound/pci/hda/patch_cs8409-tables.c
linux-5.15/sound/pci/hda/patch_cs8409.c
linux-5.15/sound/pci/hda/patch_cs8409.h
linux-5.15/sound/pci/hda/patch_hdmi.c
linux-5.15/sound/pci/hda/patch_realtek.c
linux-5.15/sound/pci/hda/patch_si3054.c
linux-5.15/sound/pci/hda/patch_sigmatel.c
linux-5.15/sound/pci/hda/patch_via.c
linux-5.15/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.15.0-92-generic/build M=/home/sheila/macbook12-audio-driver/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-92-generic'
  CC [M]  /home/sheila/macbook12-audio-driver/build/hda/patch_cirrus.o
  LD [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.o
  MODPOST /home/sheila/macbook12-audio-driver/build/hda/Module.symvers
  CC [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
  BTF [M] /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
Skipping BTF generation for /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-92-generic'
cp /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/5.15.0-92-generic/updates
depmod -a

contents of /lib/modules/5.15.0-92-generic/updates
total 1360
-rw-r--r-- 1 root root 1392336 Jan 31 11:09 snd-hda-codec-cirrus.ko

**sheila@sheila-MacBook:~$ uname -a**

Linux sheila-MacBook 5.15.0-92-generic #102-Ubuntu SMP Wed Jan 10 09:33:48 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

**sheila@sheila-MacBook:~$ find /usr/lib | grep snd-hda-codec-cirrus**

/usr/lib/modules/5.15.0-92-generic/kernel/sound/pci/hda/snd-hda-codec-cirrus.ko
/usr/lib/modules/5.15.0-92-generic/updates/snd-hda-codec-cirrus.ko
/usr/lib/modules/5.15.0-91-generic/kernel/sound/pci/hda/snd-hda-codec-cirrus.ko

**sheila@sheila-MacBook:~$ journalctl -b | grep snd_hda**

Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_intel 0000:00:1f.3: Applying patch firmware 'hda-jack-retask.fw'
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0:    inputs:
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
Jan 31 11:12:57 sheila-MacBook kernel: snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18

Don't know if it makes sense but if I check the PIN 0x1d in hdajackretask the entry in empty

Thank you for your support
Luca

gc070653, 2024-01-31:
I reinstalled your driver on a fresh install of Sparky Linux with a newer kernel but still no sound.
I tried also with Fedora but I have problems to install it on harddisk.
Here the outputs:
**sheila@sheila-macbook91:~/macbook12-audio-driver$ sudo ./install.cirrus.driver.sh**
--2024-01-31 16:58:37--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.6.13.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.1.176, 151.101.65.176, 151.101.129.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.1.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-6.6.13/sound/pci/hda/
linux-6.6.13/sound/pci/hda/Kconfig
linux-6.6.13/sound/pci/hda/Makefile
linux-6.6.13/sound/pci/hda/ca0132_regs.h
linux-6.6.13/sound/pci/hda/cs35l41_hda.c
linux-6.6.13/sound/pci/hda/cs35l41_hda.h
linux-6.6.13/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.6.13/sound/pci/hda/cs35l41_hda_property.c
linux-6.6.13/sound/pci/hda/cs35l41_hda_property.h
linux-6.6.13/sound/pci/hda/cs35l41_hda_spi.c
linux-6.6.13/sound/pci/hda/cs35l56_hda.c
linux-6.6.13/sound/pci/hda/cs35l56_hda.h
linux-6.6.13/sound/pci/hda/cs35l56_hda_i2c.c
linux-6.6.13/sound/pci/hda/cs35l56_hda_spi.c
linux-6.6.13/sound/pci/hda/hda_auto_parser.c
linux-6.6.13/sound/pci/hda/hda_auto_parser.h
linux-6.6.13/sound/pci/hda/hda_beep.c
linux-6.6.13/sound/pci/hda/hda_beep.h
linux-6.6.13/sound/pci/hda/hda_bind.c
linux-6.6.13/sound/pci/hda/hda_codec.c
linux-6.6.13/sound/pci/hda/hda_component.h
linux-6.6.13/sound/pci/hda/hda_controller.c
linux-6.6.13/sound/pci/hda/hda_controller.h
linux-6.6.13/sound/pci/hda/hda_controller_trace.h
linux-6.6.13/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.6.13/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.6.13/sound/pci/hda/hda_eld.c
linux-6.6.13/sound/pci/hda/hda_generic.c
linux-6.6.13/sound/pci/hda/hda_generic.h
linux-6.6.13/sound/pci/hda/hda_hwdep.c
linux-6.6.13/sound/pci/hda/hda_intel.c
linux-6.6.13/sound/pci/hda/hda_intel.h
linux-6.6.13/sound/pci/hda/hda_intel_trace.h
linux-6.6.13/sound/pci/hda/hda_jack.c
linux-6.6.13/sound/pci/hda/hda_jack.h
linux-6.6.13/sound/pci/hda/hda_local.h
linux-6.6.13/sound/pci/hda/hda_proc.c
linux-6.6.13/sound/pci/hda/hda_sysfs.c
linux-6.6.13/sound/pci/hda/hda_tegra.c
linux-6.6.13/sound/pci/hda/hp_x360_helper.c
linux-6.6.13/sound/pci/hda/ideapad_s740_helper.c
linux-6.6.13/sound/pci/hda/patch_analog.c
linux-6.6.13/sound/pci/hda/patch_ca0110.c
linux-6.6.13/sound/pci/hda/patch_ca0132.c
linux-6.6.13/sound/pci/hda/patch_cirrus.c
linux-6.6.13/sound/pci/hda/patch_cmedia.c
linux-6.6.13/sound/pci/hda/patch_conexant.c
linux-6.6.13/sound/pci/hda/patch_cs8409-tables.c
linux-6.6.13/sound/pci/hda/patch_cs8409.c
linux-6.6.13/sound/pci/hda/patch_cs8409.h
linux-6.6.13/sound/pci/hda/patch_hdmi.c
linux-6.6.13/sound/pci/hda/patch_realtek.c
linux-6.6.13/sound/pci/hda/patch_si3054.c
linux-6.6.13/sound/pci/hda/patch_sigmatel.c
linux-6.6.13/sound/pci/hda/patch_via.c
linux-6.6.13/sound/pci/hda/tas2781_hda_i2c.c
linux-6.6.13/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/6.6.13-amd64/build M=/home/sheila/macbook12-audio-driver/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.13-amd64'
  CC [M]  /home/sheila/macbook12-audio-driver/build/hda/patch_cirrus.o
  LD [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.o
  MODPOST /home/sheila/macbook12-audio-driver/build/hda/Module.symvers
  CC [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
  BTF [M] /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
Skipping BTF generation for /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.6.13-amd64'
cp /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.6.13-amd64/updates
depmod -a

contents of /lib/modules/6.6.13-amd64/updates
total 1640
drwxr-xr-x 2 root root    4096 Jan 31 16:38 dkms
-rw-r--r-- 1 root root 1675176 Jan 31 16:58 snd-hda-codec-cirrus.ko

**root@sheila-macbook91:/home/sheila/macbook12-audio-driver# uname -a**
Linux sheila-macbook91 6.6.13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.13-1 (2024-01-20) x86_64 GNU/Linux

 **root@sheila-macbook91:/home/sheila/macbook12-audio-driver# find /usr/lib | grep snd-hda-codec-cirrus**
/usr/lib/modules/6.6.11-amd64/kernel/sound/pci/hda/snd-hda-codec-cirrus.ko.xz
/usr/lib/modules/6.6.13-amd64/kernel/sound/pci/hda/snd-hda-codec-cirrus.ko.xz
/usr/lib/modules/6.6.13-amd64/updates/snd-hda-codec-cirrus.ko
/usr/lib/modules/6.6.13-amd64/updates/dkms/snd-hda-codec-cirrus.ko.xz

 **root@sheila-macbook91:/home/sheila/macbook12-audio-driver# journalctl -b | grep snd_hda**
Jan 31 16:52:59 sheila-macbook91 kernel: snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0:    inputs:
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
Jan 31 16:53:00 sheila-macbook91 kernel: snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18

gc070653, 2024-01-31:
I've been able to install Fedora 39 and I installed your driver but still no sound.
Here the logs

sheila@fedora:~/macbook12-audio-driver$ sudo ./install.cirrus.driver.sh
--2024-01-31 21:03:51--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.6.13.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.1.176, 151.101.65.176, 151.101.129.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.1.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-6.6.13/sound/pci/hda/
linux-6.6.13/sound/pci/hda/Kconfig
linux-6.6.13/sound/pci/hda/Makefile
linux-6.6.13/sound/pci/hda/ca0132_regs.h
linux-6.6.13/sound/pci/hda/cs35l41_hda.c
linux-6.6.13/sound/pci/hda/cs35l41_hda.h
linux-6.6.13/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.6.13/sound/pci/hda/cs35l41_hda_property.c
linux-6.6.13/sound/pci/hda/cs35l41_hda_property.h
linux-6.6.13/sound/pci/hda/cs35l41_hda_spi.c
linux-6.6.13/sound/pci/hda/cs35l56_hda.c
linux-6.6.13/sound/pci/hda/cs35l56_hda.h
linux-6.6.13/sound/pci/hda/cs35l56_hda_i2c.c
linux-6.6.13/sound/pci/hda/cs35l56_hda_spi.c
linux-6.6.13/sound/pci/hda/hda_auto_parser.c
linux-6.6.13/sound/pci/hda/hda_auto_parser.h
linux-6.6.13/sound/pci/hda/hda_beep.c
linux-6.6.13/sound/pci/hda/hda_beep.h
linux-6.6.13/sound/pci/hda/hda_bind.c
linux-6.6.13/sound/pci/hda/hda_codec.c
linux-6.6.13/sound/pci/hda/hda_component.h
linux-6.6.13/sound/pci/hda/hda_controller.c
linux-6.6.13/sound/pci/hda/hda_controller.h
linux-6.6.13/sound/pci/hda/hda_controller_trace.h
linux-6.6.13/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.6.13/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.6.13/sound/pci/hda/hda_eld.c
linux-6.6.13/sound/pci/hda/hda_generic.c
linux-6.6.13/sound/pci/hda/hda_generic.h
linux-6.6.13/sound/pci/hda/hda_hwdep.c
linux-6.6.13/sound/pci/hda/hda_intel.c
linux-6.6.13/sound/pci/hda/hda_intel.h
linux-6.6.13/sound/pci/hda/hda_intel_trace.h
linux-6.6.13/sound/pci/hda/hda_jack.c
linux-6.6.13/sound/pci/hda/hda_jack.h
linux-6.6.13/sound/pci/hda/hda_local.h
linux-6.6.13/sound/pci/hda/hda_proc.c
linux-6.6.13/sound/pci/hda/hda_sysfs.c
linux-6.6.13/sound/pci/hda/hda_tegra.c
linux-6.6.13/sound/pci/hda/hp_x360_helper.c
linux-6.6.13/sound/pci/hda/ideapad_s740_helper.c
linux-6.6.13/sound/pci/hda/patch_analog.c
linux-6.6.13/sound/pci/hda/patch_ca0110.c
linux-6.6.13/sound/pci/hda/patch_ca0132.c
linux-6.6.13/sound/pci/hda/patch_cirrus.c
linux-6.6.13/sound/pci/hda/patch_cmedia.c
linux-6.6.13/sound/pci/hda/patch_conexant.c
linux-6.6.13/sound/pci/hda/patch_cs8409-tables.c
linux-6.6.13/sound/pci/hda/patch_cs8409.c
linux-6.6.13/sound/pci/hda/patch_cs8409.h
linux-6.6.13/sound/pci/hda/patch_hdmi.c
linux-6.6.13/sound/pci/hda/patch_realtek.c
linux-6.6.13/sound/pci/hda/patch_si3054.c
linux-6.6.13/sound/pci/hda/patch_sigmatel.c
linux-6.6.13/sound/pci/hda/patch_via.c
linux-6.6.13/sound/pci/hda/tas2781_hda_i2c.c
linux-6.6.13/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/6.6.13-200.fc39.x86_64/build M=/home/sheila/macbook12-audio-driver/build/hda modules
make[1]: Entering directory '/usr/src/kernels/6.6.13-200.fc39.x86_64'
  CC [M]  /home/sheila/macbook12-audio-driver/build/hda/patch_cirrus.o
  LD [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.o
  MODPOST /home/sheila/macbook12-audio-driver/build/hda/Module.symvers
  CC [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
  BTF [M] /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
Skipping BTF generation for /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.6.13-200.fc39.x86_64'
cp /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.6.13-200.fc39.x86_64/updates
depmod -a

contents of /lib/modules/6.6.13-200.fc39.x86_64/updates
total 1640
-rw-r--r--. 1 root root 1676904 Jan 31 21:04 snd-hda-codec-cirrus.ko

sheila@fedora:~/macbook12-audio-driver$ uname -a
Linux fedora 6.6.13-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Jan 20 18:03:28 UTC 2024 x86_64 GNU/Linux

sheila@fedora:~$ find /usr/lib | grep snd-hda-codec-cirrus
/usr/lib/modules/6.5.6-300.fc39.x86_64/kernel/sound/pci/hda/snd-hda-codec-cirrus.ko.xz
/usr/lib/modules/6.6.13-200.fc39.x86_64/kernel/sound/pci/hda/snd-hda-codec-cirrus.ko.xz
/usr/lib/modules/6.6.13-200.fc39.x86_64/updates/snd-hda-codec-cirrus.ko

sheila@fedora:~$ journalctl -b | grep snd_hda
Jan 31 21:02:27 fedora kernel: snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS4208: line_outs=1 (0x1d/0x0/0x0/0x0/0x0) type:speaker
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x10/0x0/0x0/0x0/0x0)
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0:    inputs:
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x19
Jan 31 21:02:27 fedora kernel: snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x18

leifliddy, 2024-02-02:
@gc070653 

hmmm....something is off with your Fedora setup
Why is it installing the module under `/lib/modules/6.6.13-amd64`

Also your kernel headers are stored under `/usr/src/linux-headers-6.6.13-amd64`!?

Did you install or build a custom kernel?
In any case, that's a stock Fedora....

Actually, could you post the output of the following cmds?
```
cat /etc/redhat-release
uname -a
rpm -q kernel-core
rpm -qf /usr/src/linux-headers-6.6.13-amd64
rpm -qf /lib/modules/6.6.13-amd64
```

for comparison here's what your output looks like.

```
make -C /lib/modules/6.6.13-amd64/build M=/home/sheila/macbook12-audio-driver/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.13-amd64'
...
BTF [M] /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
Skipping BTF generation for /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.6.13-amd64'
cp /home/sheila/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.6.13-amd64/updates
depmod -a

contents of /lib/modules/6.6.13-amd64/updates
total 1640
drwxr-xr-x 2 root root 4096 Jan 31 16:38 dkms
-rw-r--r-- 1 root root 1675176 Jan 31 16:58 snd-hda-codec-cirrus.ko
```

And here's mine (the kernel directories are different and I'd like to know why)
```
make -C /lib/modules/6.6.13-200.fc39.x86_64/build M=/root/git.projects/macbook12-audio-driver/build/hda modules
make[1]: Entering directory '/usr/src/kernels/6.6.13-200.fc39.x86_64'
...
  BTF [M] /root/git.projects/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko
Skipping BTF generation for /root/git.projects/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.6.13-200.fc39.x86_64'
cp /root/git.projects/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.6.13-200.fc39.x86_64/updates
depmod -a

contents of /lib/modules/6.6.13-200.fc39.x86_64/updates
total 1640
-rw-r--r--. 1 root root 1676928 Feb  2 04:23 snd-hda-codec-cirrus.ko
```

leifliddy, 2024-02-02:
@gc070653 I don't get it -- your Fedora output looks exactly like mine. 
The only difference I see is that the size of my module is slightly different -- but  that could be completely normal 
```
contents of /lib/modules/6.6.13-200.fc39.x86_64/updates
total 1640
-rw-r--r--. 1 root root 1676928 Feb  2 04:23 snd-hda-codec-cirrus.ko
```
Do you know if the speakers work with macOS?

Does the startup chime sound when you boot the system?
Here's how you can mute and unmute the startup chime from Linux.
If it's already muted, try unmuting it and see if works. 

```
# unmute startup chime
printf "\x07\x00\x00\x00\x00" > /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82

# mute startup chime
printf "\x07\x00\x00\x00\x80" > /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82
```

Could you post the output of the following cmds from Fedora?

```
amixer contents
amixer scontents
amixer sget Master
aplay --list-devices
```

Here's what the output looks like on my system
```
[leif.liddy@mac ~]$ amixer contents
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=4,min=0,max=65536,step=1
  : values=36036,36036,35156,35156
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=4,min=0,max=65536,step=1
  : values=36036,36036,35156,35156

[leif.liddy@mac ~]$ amixer scontents
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 36036 [55%] [on]
  Front Right: Playback 36036 [55%] [on]
  Rear Left: Playback 35156 [54%] [on]
  Rear Right: Playback 35156 [54%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Capture 0 - 65536
  Front Left: Capture 36036 [55%] [on]
  Front Right: Capture 36036 [55%] [on]
  Rear Left: Capture 35156 [54%] [on]
  Rear Right: Capture 35156 [54%] [on]

[leif.liddy@mac ~]$ aplay --list-devices
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: CS4208 Analog [CS4208 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

[leif.liddy@mac ~]$ amixer sget Master
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 36036 [55%] [on]
  Front Right: Playback 36036 [55%] [on]
  Rear Left: Playback 35156 [54%] [on]
  Rear Right: Playback 35156 [54%] [on]
```

gc070653, 2024-02-03:
I discovered what was wrong on my macbook.
Following your suggestion, I tried to unmute the chime sound but the operation was not permitted in Linux also with sudo.
At that point I rebooted in Macos and I enabled the sound at startup option in that environment.
Rebooting in Linux the speakers were working.
So while in Macos the sound at startup does not affect the speakers if you listen music, in Linux this option, set in Macos, completely disable the speakers.
Thank  you so much for your support
Ciao
Luca