leifliddy/macbook12-audio-driver #27: Macbook9,1 12 Retina Early 2016

gc070653, 2024-01-18:
So far I've not been able to make the internal speakers working  on any distro I tried, Manjaro Ubuntu Linux Mint, even after installing your patch.
After your patch the system is showing correctly the Analog Output-Built-in Audio and if I play some song I see the sound bar moving but no sound is coming out. 
Earphones are not working any longer as expected.
I also tried to add 'options snd_hda_intel model=mbp11' into the alsa-base.conf but nothing changed.

Checking your patch_cirrus.c, I did not found any reference to macbook9,1, can be this the reason why the driver is not working?
I reference to this:

/*
 * CS4208 support:
 * Its layout is no longer compatible with CS4206/CS4207
 */
enum {
	CS4208_MAC_AUTO,
	CS4208_MBA6,
	CS4208_MBP11,
	CS4208_MACMINI,
	CS4208_GPIO0,
};

static const struct hda_model_fixup cs4208_models[] = {
	{ .id = CS4208_GPIO0, .name = "gpio0" },
	{ .id = CS4208_MBA6, .name = "mba6" },
	{ .id = CS4208_MBP11, .name = "mbp11" },
	{ .id = CS4208_MACMINI, .name = "macmini" },
	{}
};

static const struct snd_pci_quirk cs4208_fixup_tbl[] = {
	SND_PCI_QUIRK_VENDOR(0x106b, "Apple", CS4208_MAC_AUTO),
	{} /* terminator */
};

/* codec SSID matching */
static const struct snd_pci_quirk cs4208_mac_fixup_tbl[] = {
	SND_PCI_QUIRK(0x106b, 0x5e00, "MacBookPro 11,2", CS4208_MBP11),
	SND_PCI_QUIRK(0x106b, 0x6c00, "MacMini 7,1", CS4208_MACMINI),
	SND_PCI_QUIRK(0x106b, 0x7100, "MacBookAir 6,1", CS4208_MBA6),
	SND_PCI_QUIRK(0x106b, 0x7200, "MacBookAir 6,2", CS4208_MBA6),
	SND_PCI_QUIRK(0x106b, 0x7b00, "MacBookPro 12,1", CS4208_MBP11),
	{} /* terminator */
};
 
Alsactl init is showing : 

Found hardware: "HDA-Intel" "Cirrus Logic CS4208" 
"HDA:10134208,106b6500,00100401 HDA:80862809,80860101,001000000" "0x8086" "0x7270"
Hardware is initialized using a generic method

Also the reference to 0x106b 0x6500 is missing

Sorry if I've not been very clear and thank you in advance
Ciao
Luca

gc070653, 2024-01-25:
Using HDA-Analizer I discover that in Codec 0 the node ox02 AUD_OUT is set with Mute=true and using hdajackretask the internal speaker entry is empty. Can this help to fix the missing audio?

Alth3c0w, 2024-08-11:
For anyone having this issue, enabling "Play Startup Sound" in macos fixes the audio issue. For some reason, having this setting on mutes the speakers, but disabling it allowed the speakers to work in Ubuntu on the 2016 Macbook for me.

patx, 2024-09-17:
@Alth3c0w interesting. I will test this out ASAP. Thanks

aj-pd, 2024-09-22:
Same issue here. Tried enabling the startup sound and it didn't help. Any other things to try? Thanks

gc070653, 2024-09-22:
In my MacBook I have Linux mint and iOS installed.To make the sound working I enable the start up sound in iOS and after the sound driver installation in mint the sound is working

Yahoo Mail: Search, Organize, Conquer

aj-pd, 2024-09-22:
Exactly what I did, and it didn't work. Spent a few hours searching online for other solutions, but couldn't find anything that would get it working

gc070653, 2024-09-22:
When you boot the MacBook do you listen the sound?

Yahoo Mail: Search, Organize, Conquer

aj-pd, 2024-09-22:
@gc070653  Yes there is a startup chime. Although it's kind of lacking base, it sounds too tinny, don't think I've seen this with other macs. But it does chime every startup, yes. I have fedora kde on it, but I've tried ubuntu too, it was the same.

gc070653, 2024-09-22:
I have installed LMDE 6 Cinnamon and it is fully working: sound, bluetooth and camera
As you can see from the attached picture the sound config at the end of the proceduee provides only Analog Stereo Output instead of Speakers-Built-In Audio and Internal Microphone.

aj-pd, 2024-09-22:
@gc070653 
There are no attached screenshots, probably got lost in translation somewhere. Maybe I'll try to install mint when I have time for this, but I doubt it will work. And I really prefer kde more than any of the mint DEs, but I guess it's worth trying if that will make this laptop at least somewhat useful

aj-pd, 2024-09-22:
@gc070653 
Update: after another reboot it started working for some reason. But the sound is kind of tinny, as I mentioned before, as if the base was completely cut off. Very different from the sound on the Mac OS. But at least there is sound now :+1:

Alth3c0w, 2024-09-22:
To fix the tinny sound, you can first change the audio device to use "Analog Surround 4.0 Output" in the Sound Configuration in Settings, and then use an equalizer. I use PulseEffects which has allowed me to get the audio at least close to MacOS. Here's the json config file for PE that I use if anyone wants it.

[bars2.json](https://github.com/user-attachments/files/17091342/bars2.json)

tobiasgrossmann, 2024-11-01:
> For anyone having this issue, enabling "Play Startup Sound" in macos fixes the audio issue. For some reason, **having this setting disabled mutes the speakers**, but disabling it allowed the speakers to work in Ubuntu on the 2016 Macbook for me.

For clarification: You mean: "**Enable** Startup Sound" in Mac OS will enable the Speakers?  Thanks

leonardpitzu, 2024-12-14:
No need to do anything in macOS, I do not even have it installed any more. Just do a PRAM/NVRAM reset and the issue is solved. Once you hear the chime while booting up you're good to go once the driver is installed.

mistraid121, 2025-03-07:
Hi,
I also have sound under linux only if startup sound (chime) is enabled. Is there any solution to have it working under linux with muted chime? Or is there a way to reduce startup chime to a minimum level?
thanks

maxwowpow, 2025-06-25:
works on arch. wonder where apple put this annoying sound, maybe even in the nvram.
NB arch guys: check your /boot/efi/EFI/BOOT has the BOOTX64.efi (your arch grub), because after NVRAM nuke "other" boot entries are wiped from NVRAM.

Silhouette6, 2025-06-29:
another stupid solution is if you wanna enable speaker, just poweroff and use a charge cable to activate the mac-buildin audio driver (a "beep" will show up with charging the computer when poweroff, which means the audio driver already loaded.)
And I am also wonder if there is any solution to reduce the startup chime to a minimun level, i think that will be the best solution (load the mac-buildin audio driver every startup but avoid the annoying chime)...

Silhouette6, 2025-06-29:
looking forward to your replies. >w<

Silhouette6, 2025-07-18:
> This is what I wrote to the efivars to mute the chime: (with still sound under linux)
> 
> * 7c436110-ab2a-4bbb-a880-fe41995c9f82-SystemAudioVolumeDB: 0xd7
> * 7c436110-ab2a-4bbb-a880-fe41995c9f82-SystemAudioVolume: 0x2d

could u tell me how to alter these efivars? I start up with command+R enter the recover mode to use apple's terminal and run these commands as root:
# nvram SystemAudioVolume=0xd7
# nvram SystemAudioVolumeDB=0x2d
finish and restart, the chime doesn't mute, then i check the vars SystemAudioVolume&SytemAudioVolumeDB and find them doesn't changed to 0x7d and 0x2d.
maybe it didn't work in mac's terminal? must I alter these vars in linux using efivars? how to do that?

mistraid121, 2025-07-18:
Yes, do it from linux (this is a linux audio driver). Something like:
`chattr -i /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82`
`printf "\x07\x00\x00\x00\xd7" > /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82`
`chattr +i /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82`
I have not tested it on a dual boot with macos. Macos can probably modify these values...

maxwowpow, 2025-09-25:
during my experiments I lost the audio with the driver installed (chime active, no sound in arch).

chime can be disabled with efivars (arch):

```sh
sudo su

sudo chattr -i /sys/firmware/efi/efivars/SystemAudioVolumeDB-7c436110-ab2a-4bbb-a880-fe41995c9f82
rm /sys/firmware/efi/efivars/SystemAudioVolumeDB-7c436110-ab2a-4bbb-a880-fe41995c9f82
printf "\x07\x00\x00\x00\xd7" > ~/SystemAudioVolumeDB-7c436110-ab2a-4bbb-a880-fe41995c9f82
sudo cp ~/SystemAudioVolumeDB-7c436110-ab2a-4bbb-a880-fe41995c9f82 /sys/firmware/efi/efivars/
sudo chattr +i /sys/firmware/efi/efivars/SystemAudioVolumeDB-7c436110-ab2a-4bbb-a880-fe41995c9f82

sudo chattr -i /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82
rm /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82
printf "\x07\x00\x00\x00\x2d" > ~/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82
sudo cp ~/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82 /sys/firmware/efi/efivars/
sudo chattr +i /sys/firmware/efi/efivars/SystemAudioVolume-7c436110-ab2a-4bbb-a880-fe41995c9f82
```

interestingly the apple logo disappears after it too. whatever.

guys WITH sound working with chime, can try whether these vars disable chime, while keeping the sound. Arch or otherwise.