leifliddy/macbook12-audio-driver #28: Microphone

nnylip, 2024-03-07:
Just used driver for macbook10,1 12 inch 2017 and sound output works great. Is there a solution for the microphone?

MarkDiGiovanni, 2024-04-19:
Same here; I was able to get the output working as well, but lost the mic.

alessandroForgione, 2025-02-09:
I have the same problem after installing the driver... the microphone no longer works with my 10.1 notebook with kubuntu 24.04LTS @leifliddy

aelph, 2026-10-20:
Headset microphone (and the internal mic after a headphone plug) now work on my MacBook10,1 — branch `fix/cs4208-capture-after-hp-jack` in my fork: https://github.com/aelph/macbook12-audio-driver/tree/fix/cs4208-capture-after-hp-jack

What the branch adds on top of master:

- the internal mic no longer dies when headphones are plugged in (the capture converter lost its stream tag after the headphone trace);
- headset microphone in the combo jack (TRRS, e.g. Apple EarPods) with automatic internal/headset switching — the codec's headset block behind the vendor widget, following what AppleHDA does (mechanism worked out by @IgorChicherin for the MacBookPro11,x in mbp11-3-sound, adapted here);
- headset remote buttons (play/pause, volume up/down) as media keys;
- the headphone trace is no longer replayed on every PCM open, which was popping into the microphone;
- `contrib/mic-boost-setup.sh`: WirePlumber's ACP paths reset both mic boosts to 0 on every port change, this keeps them at a usable level;
- build fixes: 6.6.y/6.1.y stable kernels (hda_quirk backport), DKMS builds for every installed kernel.

Tested on MacBook10,1 (2017) with Apple EarPods, Manjaro, PipeWire/WirePlumber 0.5, kernels 7.2 and 6.12 live, 6.6 and 6.1 build only. Not tested: MacBook9,1 (the AppleHDA layout is the same for both, so it should work), headsets with the other TRRS pinout, older kernels.

To try it:

```
git clone -b fix/cs4208-capture-after-hp-jack https://github.com/aelph/macbook12-audio-driver.git
cd macbook12-audio-driver
sudo ./install.cirrus.driver.sh -i && sudo reboot
sudo contrib/mic-boost-setup.sh        # after the reboot
```

Reports from a MacBook9,1 would be welcome before I open a pull request. If something goes wrong, `headset_mic=0` / `headset_buttons=0` module parameters turn the new parts off.