leifliddy/macbook12-audio-driver #29: Build Fails on Recent Linux Kernel

x5444, 2024-03-31:
Just a quick note, as of recently (Kernel 6.8.0 maybe?) the build fails because `strlcpy` is no longer defined.

I don't have time to fix this properly, replacing one instance of `strlcpy` with `strcpy` is enough to make it build and work again.

My best guess is some include was removed from the original source file in the kernel that the patch is injected into. I didn't investigate it and I'm not going to.

> diff --git a/patch_cirrus/patch_cirrus_a1534_pcm.h b/patch_cirrus/patch_cirrus_a1534_pcm.h
> index 9ecaaa9..d8b853e 100644
> --- a/patch_cirrus/patch_cirrus_a1534_pcm.h
> +++ b/patch_cirrus/patch_cirrus_a1534_pcm.h
> @@ -120,7 +120,7 @@ static void cs_4208_fill_pcm_stream_name(char *str, size_t len, const char *sfx,
>  
>  	if (*str)
>  		return;
> -	strlcpy(str, chip_name, len);
> +	strcpy(str, chip_name);
>  
>  	/* drop non-alnum chars after a space */
>  	for (p = strchr(str, ' '); p; p = strchr(p + 1, ' ')) {

leifliddy, 2024-04-01:
Thanks. I'll look into it this week.

leifliddy, 2024-04-11:
Issue was resolved with:
https://github.com/leifliddy/macbook12-audio-driver/commit/ef570d7a4f5c4b0c188725e73429cb5d86e0c551

closed 2024-04-11 by leifliddy