leifliddy/macbook12-audio-driver #31: MacBook 8,1 2015 Cirrus 4208 support

sebl05, 2024-07-15:
Hi @leifliddy, I've been reading up on the work you've done so far making old make hardware play nice with older macs. I see in the repo that you're not supporting 8,1 2015 macbook – but you have done some extensive research into it. Couldn't you get it to work and would you still consider releasing an updated patch for it? Would be a massive help to put some new life into my old macbook that other that the speakers and bluetooth is working fine on Ubuntu 24.04. Hope to hear from you soon.

Thanks,

Seb

sebl05, 2024-07-19:
So sorry to hear that @leifliddy. Ik hope all is well and wish you a speedy recovery.

All the best,

Seb

2bmoJake, 2024-08-29:
Yeah i hope you get well soon

2bmoJake, 2025-11-06:
Thanks for this repo, it helped me get in to linux by actually being able to live with it day-day, i now have a dell xps and. I still have my little macbook and i love it and care for it. Linux is all i ever use now and i love the tinkering aspect, this repo helped me to get into it.
I’m glad to hear you have had a speedy recovery and i can only imagine that you want to get back to tinkering and i hope you do, your a helpful genius. Wish you all the best!

-Jake

mrbubzie93, 2026-05-10:
Any update on this?