leifliddy/macbook12-audio-driver #32: Is there now a solution for headphones ?

labarbade, 2024-07-27:
Its been a few years, maybe now someone has found a solution for making headphones work without disabling this module ?
As a real beginner I dont know what this procedure in the readme file means; do you have to disable the module and reboot every time you want to use headphones ?
Has anybody found a workaround for using aux ?
Thanks !

Alth3c0w, 2024-08-13:
You can connect a Usb-C to aux jack dongle and it seems to work fine, just not through the built-in headphone port, if that helps. Haven't found a solution to get the native one to work.

labarbade, 2024-08-14:
I thought about trying this but I didnt think it would work. Well thats cheaper than buying a bluetooth headset so why not ! I dont really understand why that would work though ? Thank you !

martin-irwin, 2026-11-05:
Yes. On a MacBook10,1 I now have **internal speakers and the headphone jack both working, with automatic switching on plug/unplug**, on the **stock** `snd-hda-codec-cs420x` driver: https://github.com/martin-irwin/macbook12-cs4208-audio

How it works, in case it's useful for this driver too:

- **Stock driver stays loaded.** It handles the headphone jack and jack detection. A small root daemon talks to the codec through hwdep (`/dev/snd/hwC0D0`) and replays this project's macOS traces at the right moments:
  - `setup_a1534` + `play_a1534` (381 verbs, ~40 ms) when playback starts with nothing in the jack, or on unplug.
  - `headphones_a1534` (the disabled function in `patch_cirrus_a1534_setup.h`, including its `snd_hda_coef_item` writes and `0x24 0x7f0 0xae`) when headphones are plugged in during speaker playback. It switches instantly.
- **Catch 1: the GPIO4/5 pulses shut the speaker amp down.** `headphones_a1534` pulses GPIO4/5 (`0x20 → 0x30 → 0x10 → 0x00`). That appears to be a pulse-coded amp shutdown: after it, the speakers stayed dead until a cold boot. With those GPIO writes removed (GPIO0 alone gates the amp), switching works in both directions.
- **Catch 2: the stream tag moves.** After `headphones_a1534`, converter `0x0a` is detached, so the stream tag has to be read from `0x02`.
- **Format and volume.** The amp needs a 16-bit/44.1 kHz stream. The speaker path ignores the hardware `Master`, so I use PipeWire software volume.
- **The chime must not be muted.** As in #65, the speaker amp only gets powered when the EFI startup chime is not muted.

It's tested on one machine (MacBook10,1, kernel 7.2.15, PipeWire 1.6.9). The verb sequences come from this project, with credit in the README. Thanks for capturing the traces. None of this would have been possible without them.