leifliddy/macbook12-audio-driver #37: macbook 9,1 install fails on arch 6.12.10

popcornman209, 2025-01-19:
audio doesnt work, fails to install with dkms:

Error! Bad return status for module build on kernel: 6.12.10-arch1-1 (x86_64)
Consult /var/lib/dkms/macbook12-audio/0.1/build/make.log for more information.

the files it says have information i uploaded here. without dkms i get this outputed to the console, it doesnt work:

  CC [M]  /home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.o
In file included from /home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c:528:
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus_a1534_setup.h:71:6: warning: no previous prototype for ‘snd_hda_coef_item’ [-Wmissing-prototypes]
   71 | void snd_hda_coef_item(struct hda_codec *codec, u16 write_flag, hda_nid_t nid, u32 idx, u32 param, u32 retdata, int srcidx)
      |      ^~~~~~~~~~~~~~~~~
In file included from /home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus_a1534_setup.h:1679:
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus_a1534_pcm.h:135:6: warning: no previous prototype for ‘cs_4208_playback_pcm_hook’ [-Wmissing-prototypes]
  135 | void cs_4208_playback_pcm_hook(struct hda_pcm_stream *hinfo, struct hda_codec *codec, struct snd_pcm_substream *substream, int action)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus_a1534_pcm.h:253:5: warning: no previous prototype for ‘cs_4208_build_pcms_explicit’ [-Wmissing-prototypes]
  253 | int cs_4208_build_pcms_explicit(struct hda_codec *codec)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c: In function ‘patch_cs420x’:
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c:610:50: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Wincompatible-pointer-types]
  610 |         snd_hda_pick_fixup(codec, cs420x_models, cs420x_fixup_tbl,
      |                                                  ^~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  const struct snd_pci_quirk *
In file included from /home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c:15:
/home/leo/Downloads/macbook12-audio-driver/build/hda/hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c: In function ‘cs4208_fixup_mac’:
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c:684:41: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Wincompatible-pointer-types]
  684 |         snd_hda_pick_fixup(codec, NULL, cs4208_mac_fixup_tbl, cs4208_fixups);
      |                                         ^~~~~~~~~~~~~~~~~~~~
      |                                         |
      |                                         const struct snd_pci_quirk *
/home/leo/Downloads/macbook12-audio-driver/build/hda/hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c: In function ‘patch_cs4208’:
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c:799:50: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Wincompatible-pointer-types]
  799 |         snd_hda_pick_fixup(codec, cs4208_models, cs4208_fixup_tbl,
      |                                                  ^~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  const struct snd_pci_quirk *
/home/leo/Downloads/macbook12-audio-driver/build/hda/hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c: In function ‘patch_cs4210’:
/home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.c:1195:50: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Wincompatible-pointer-types]
 1195 |         snd_hda_pick_fixup(codec, cs421x_models, cs421x_fixup_tbl,
      |                                                  ^~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  const struct snd_pci_quirk *
/home/leo/Downloads/macbook12-audio-driver/build/hda/hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
make[3]: *** [scripts/Makefile.build:229: /home/leo/Downloads/macbook12-audio-driver/build/hda/patch_cirrus.o] Error 1
make[2]: *** [/usr/lib/modules/6.12.10-arch1-1/build/Makefile:1937: /home/leo/Downloads/macbook12-audio-driver/build/hda] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.10-arch1-1/build'
make: *** [Makefile:10: all] Error 2
cp /home/leo/Downloads/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.12.10-arch1-1/updates
cp: cannot stat '/home/leo/Downloads/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:17: install] Error 1

contents of /lib/modules/6.12.10-arch1-1/updates
total 0

[make.log](https://github.com/user-attachments/files/18470666/make.log)

leifliddy, 2025-01-19:
I just merged a PR that might help with this issue. Could you do a `git pull` and try to build it again?
Please let me know if it builds for you...

JOut of curiosity, what macbook model/(year) do you have?

closed 2025-01-20 by leifliddy

popcornman209, 2025-01-20:
> I just merged a PR that might help with this issue. Could you do a `git pull` and try to build it again?
> Please let me know if it builds for you...
> 
> JOut of curiosity, what macbook model/(year) do you have?
> 
> 

It's an early 2016 MacBook, not pro or air or anything just the super slim one. That fixed it though, Thanks!