leifliddy/macbook12-audio-driver #41: Compile Error on kernel 6.14

amirian, 2025-07-26:
It worked perfect on previous ubuntu updates, but gives errors on 6.14. Here is the compile error message:
`patch_cirrus.c: In function ‘patch_cs420x’:
patch_cirrus.c:610:50: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  610 |         snd_hda_pick_fixup(codec, cs420x_models, cs420x_fixup_tbl,
      |                                                  ^~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  const struct snd_pci_quirk *
In file included from patch_cirrus.c:15:
hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
patch_cirrus.c: In function ‘cs4208_fixup_mac’:
patch_cirrus.c:684:41: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  684 |         snd_hda_pick_fixup(codec, NULL, cs4208_mac_fixup_tbl, cs4208_fixups);
      |                                         ^~~~~~~~~~~~~~~~~~~~
      |                                         |
      |                                         const struct snd_pci_quirk *
hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
patch_cirrus.c: In function ‘patch_cs4208’:
patch_cirrus.c:799:50: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  799 |         snd_hda_pick_fixup(codec, cs4208_models, cs4208_fixup_tbl,
      |                                                  ^~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  const struct snd_pci_quirk *
hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
patch_cirrus.c: In function ‘patch_cs4210’:
patch_cirrus.c:1195:50: error: passing argument 3 of ‘snd_hda_pick_fixup’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1195 |         snd_hda_pick_fixup(codec, cs421x_models, cs421x_fixup_tbl,
      |                                                  ^~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  const struct snd_pci_quirk *
hda_local.h:380:49: note: expected ‘const struct hda_quirk *’ but argument is of type ‘const struct snd_pci_quirk *’
  380 |                         const struct hda_quirk *quirk,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: patch_cirrus.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.14.0-24-generic/Makefile:2002: .] Error 2
make[2]: *** [/usr/src/linux-headers-6.14.0-24-generic/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/me/macbook12-audio-driver/build/hda'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-24-generic'
make: *** [Makefile:10: all] Error 2
cp /home/me/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.14.0-24-generic/updates
cp: cannot stat '/home/me/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:17: install] Error 1

`

gluap, 2025-08-02:
I don't use ubuntu and I'm also not a developer of the driver, but the error message seems to indicate that the error is only an error because things were compiled with ``-Werror`` - possibly the newer ubuntu kernel changed compiling options. ``-Werror`` means: Treat this warning as an error.

Unless someone offers you a readymade solution and you are adventurous you might try to find a way to compile without ``-Werror=incompatible-pointer-types``.

amirian, 2025-08-02:
Thanks for your response. I think there is no relation to the linux distro as the installing shell downloads the kernel from the standard kernel.org source. However, I did a try as the driver programmer keeps no responding. No success in ignoring the Werror parameter as it is not found in the [Makefile](https://github.com/leifliddy/macbook12-audio-driver/blob/master/Makefile)

amirian, 2025-08-06:
I finally fixed the problem by manipulating [patch_cirrus.c](https://github.com/leifliddy/macbook12-audio-driver/blob/master/patch_cirrus/patch_cirrus.c)
There I only changed all 8 occurrences of `static const struct snd_pci_quirk ` to `static const struct hda_quirk`. Then the driver is compiled successfully. I did better put the fix here instead of uploading a fork. The author might apply the patch in the main repo.

amirian, 2025-08-07:
Ok, as the author seems not following issues, I made a [compile error free fork](https://github.com/amirian/macbook12-audio-driver-linux6.14)

amirian, 2025-09-09:
> [@leifliddy](https://github.com/leifliddy) the changes you brought in from the fork breaks compilation on kernel < 6.14 FYI (breaks 6.12 at least)

But it works for me on 6.13 and 6.14!

leifliddy, 2025-09-09:
I appreciate everyone's feedback.  This is likely bordering on TMI, but I suffered a traumatic brain injury (severe TBI) last year whilst driving home from work. 
https://www.zvw.de/stuttgart-region/unfall-in-leinfelden-echterdingen-motorradfahrer-%C3%BCber-leitplanke-abgeworfen_arid-828214
So I've been out of "commission" for a while to say the least. But I'm pretty much back now. 

What I would like to do is create 2 separate Makefile's and get rid of the `sed` statements in  `install.cirrus.driver.sh` as a result. One `Makefile` would be for older kernels (possibly 6.12 and earlier) and the other would obviously be for 6.13 and later. I'll have that pushed out a bit later tonight. Please give me your thoughts on this (if you have any). I'm hoping everyone here can help me test out the changes. I'll "report back" in a bit once I've made the changes.