leifliddy/macbook12-audio-driver #42: Install fails, even with patches unsure as to why

justanthonylee, 2025-09-03:
Hello, sorry for the trouble but thank you for the work you did @leifliddy and I have used this on the same machine in the past, but now when I install on a fresh install of the OS I get the following errors at the end. I did see the other posts but noticed you pushed a code change since that but it's not working for me after the update.

`root@anthony-MacBook:~/macbook12-audio-driver# ./install.cirrus.driver.sh -i
Creating symlink /var/lib/dkms/macbook12-audio/0.1/source -> /usr/src/macbook12-audio-0.1

Kernel preparation unnecessary for this kernel. Skipping...

Running the pre_build script:
--2025-09-02 20:05:59--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.8.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.21.176, 2a04:4e42:5::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.21.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-09-02 20:05:59 ERROR 404: Not Found.

--2025-09-02 20:05:59--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.8.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.21.176, 2a04:4e42:5::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.21.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 142502100 (136M) [application/x-xz]
Saving to: ‘build/linux-6.8.tar.xz’

linux-6.8.tar.xz                             100%[=============================================================================================>] 135.90M  23.7MB/s    in 7.1s    

2025-09-02 20:06:07 (19.2 MB/s) - ‘build/linux-6.8.tar.xz’ saved [142502100/142502100]

linux-6.8/sound/pci/hda/
linux-6.8/sound/pci/hda/Kconfig
linux-6.8/sound/pci/hda/Makefile
linux-6.8/sound/pci/hda/ca0132_regs.h
linux-6.8/sound/pci/hda/cirrus_scodec.c
linux-6.8/sound/pci/hda/cirrus_scodec.h
linux-6.8/sound/pci/hda/cirrus_scodec_test.c
linux-6.8/sound/pci/hda/cs35l41_hda.c
linux-6.8/sound/pci/hda/cs35l41_hda.h
linux-6.8/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.8/sound/pci/hda/cs35l41_hda_property.c
linux-6.8/sound/pci/hda/cs35l41_hda_property.h
linux-6.8/sound/pci/hda/cs35l41_hda_spi.c
linux-6.8/sound/pci/hda/cs35l56_hda.c
linux-6.8/sound/pci/hda/cs35l56_hda.h
linux-6.8/sound/pci/hda/cs35l56_hda_i2c.c
linux-6.8/sound/pci/hda/cs35l56_hda_spi.c
linux-6.8/sound/pci/hda/hda_auto_parser.c
linux-6.8/sound/pci/hda/hda_auto_parser.h
linux-6.8/sound/pci/hda/hda_beep.c
linux-6.8/sound/pci/hda/hda_beep.h
linux-6.8/sound/pci/hda/hda_bind.c
linux-6.8/sound/pci/hda/hda_codec.c
linux-6.8/sound/pci/hda/hda_component.h
linux-6.8/sound/pci/hda/hda_controller.c
linux-6.8/sound/pci/hda/hda_controller.h
linux-6.8/sound/pci/hda/hda_controller_trace.h
linux-6.8/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.8/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.8/sound/pci/hda/hda_eld.c
linux-6.8/sound/pci/hda/hda_generic.c
linux-6.8/sound/pci/hda/hda_generic.h
linux-6.8/sound/pci/hda/hda_hwdep.c
linux-6.8/sound/pci/hda/hda_intel.c
linux-6.8/sound/pci/hda/hda_intel.h
linux-6.8/sound/pci/hda/hda_intel_trace.h
linux-6.8/sound/pci/hda/hda_jack.c
linux-6.8/sound/pci/hda/hda_jack.h
linux-6.8/sound/pci/hda/hda_local.h
linux-6.8/sound/pci/hda/hda_proc.c
linux-6.8/sound/pci/hda/hda_sysfs.c
linux-6.8/sound/pci/hda/hda_tegra.c
linux-6.8/sound/pci/hda/hp_x360_helper.c
linux-6.8/sound/pci/hda/ideapad_s740_helper.c
linux-6.8/sound/pci/hda/patch_analog.c
linux-6.8/sound/pci/hda/patch_ca0110.c
linux-6.8/sound/pci/hda/patch_ca0132.c
linux-6.8/sound/pci/hda/patch_cirrus.c
linux-6.8/sound/pci/hda/patch_cmedia.c
linux-6.8/sound/pci/hda/patch_conexant.c
linux-6.8/sound/pci/hda/patch_cs8409-tables.c
linux-6.8/sound/pci/hda/patch_cs8409.c
linux-6.8/sound/pci/hda/patch_cs8409.h
linux-6.8/sound/pci/hda/patch_hdmi.c
linux-6.8/sound/pci/hda/patch_realtek.c
linux-6.8/sound/pci/hda/patch_si3054.c
linux-6.8/sound/pci/hda/patch_sigmatel.c
linux-6.8/sound/pci/hda/patch_via.c
linux-6.8/sound/pci/hda/tas2781_hda_i2c.c
linux-6.8/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/6.8.0-79-generic/build M=/var/lib/dkms/macbook12-audio/0.1/build/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-79-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04.2) 12.3.0
  CC [M]  /var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.o
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:399:31: error: array type has incomplete element type ‘struct hda_quirk’
  399 | static const struct hda_quirk cs420x_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:528:
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus_a1534_setup.h:71:6: warning: no previous prototype for ‘snd_hda_coef_item’ [-Wmissing-prototypes]
   71 | void snd_hda_coef_item(struct hda_codec *codec, u16 write_flag, hda_nid_t nid, u32 idx, u32 param, u32 retdata, int srcidx)
      |      ^~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus_a1534_setup.h:1679:
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus_a1534_pcm.h:135:6: warning: no previous prototype for ‘cs_4208_playback_pcm_hook’ [-Wmissing-prototypes]
  135 | void cs_4208_playback_pcm_hook(struct hda_pcm_stream *hinfo, struct hda_codec *codec, struct snd_pcm_substream *substream, int action)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus_a1534_pcm.h:253:5: warning: no previous prototype for ‘cs_4208_build_pcms_explicit’ [-Wmissing-prototypes]
  253 | int cs_4208_build_pcms_explicit(struct hda_codec *codec)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:647:31: error: array type has incomplete element type ‘struct hda_quirk’
  647 | static const struct hda_quirk cs4208_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:653:31: error: array type has incomplete element type ‘struct hda_quirk’
  653 | static const struct hda_quirk cs4208_mac_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:837:31: error: array type has incomplete element type ‘struct hda_quirk’
  837 | static const struct hda_quirk cs421x_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:837:31: warning: ‘cs421x_fixup_tbl’ defined but not used [-Wunused-variable]
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:653:31: warning: ‘cs4208_mac_fixup_tbl’ defined but not used [-Wunused-variable]
  653 | static const struct hda_quirk cs4208_mac_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:647:31: warning: ‘cs4208_fixup_tbl’ defined but not used [-Wunused-variable]
  647 | static const struct hda_quirk cs4208_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.c:399:31: warning: ‘cs420x_fixup_tbl’ defined but not used [-Wunused-variable]
  399 | static const struct hda_quirk cs420x_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/macbook12-audio/0.1/build/build/hda/patch_cirrus.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-79-generic/Makefile:1920: /var/lib/dkms/macbook12-audio/0.1/build/build/hda] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-79-generic'
make: *** [Makefile:10: all] Error 2
cp /var/lib/dkms/macbook12-audio/0.1/build/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.8.0-79-generic/updates
cp: cannot stat '/var/lib/dkms/macbook12-audio/0.1/build/build/hda/snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:17: install] Error 1

contents of /lib/modules/6.8.0-79-generic/updates
total 3552
-rw-r--r-- 1 root root 3636504 Sep  2 19:34 facetimehd.ko

Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.8.0-79-generic...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.0-79-generic (x86_64)
Consult /var/lib/dkms/macbook12-audio/0.1/build/make.log for more information.
root@anthony-MacBook:~/macbook12-audio-driver# 
`

justanthonylee, 2025-09-04:
Thanks for taking the time @leifliddy, the problem also happens with your bluetooth driver so assuming same problem. I am running Zorin 17 Pro, running on Ubuntu LTS. For patches was reading other errors that was posted by others that made workarounds in the issues on the project that I tried but also noticed you already applied to the repo source.

[make.log](https://github.com/user-attachments/files/22130117/make.log)

leifliddy, 2025-09-05:
today, I installed ubuntu LTS on a vm.  

`cat /etc/os-release`
```
PRETTY_NAME="Ubuntu 25.04"
NAME="Ubuntu"
VERSION_ID="25.04"
VERSION="25.04 (Plucky Puffin)"
VERSION_CODENAME=plucky
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=plucky
LOGO=ubuntu-logo
```
I cloned the `/macbook12-audio-driver` git repo and ran `./install.cirrus.driver.sh -i` the module built without issue. 
I'm not sure why it's not building on your system. But if I can't repro the issue -- then I can't even begin to troubleshoot it. 

[install.log](https://github.com/user-attachments/files/22162998/install.log)

justanthonylee, 2025-09-06:
@leifliddy So Zorin is running the slightly older LTS version, I am also not sure about if it's missing any packages required. There is another post about it on the project with Zoron breaking after getting a LTS update in November.

https://github.com/leifliddy/macbook12-audio-driver/issues/34

gAlleb, 2025-09-07:
Hi! Almost similar errors and warnings appear when i try installing on Void Linux:

<details> <summary> Logs </summary>

```
linux-6.12/sound/pci/hda/patch_cs8409.h
linux-6.12/sound/pci/hda/patch_hdmi.c
linux-6.12/sound/pci/hda/patch_realtek.c
linux-6.12/sound/pci/hda/patch_senarytech.c
linux-6.12/sound/pci/hda/patch_si3054.c
linux-6.12/sound/pci/hda/patch_sigmatel.c
linux-6.12/sound/pci/hda/patch_via.c
linux-6.12/sound/pci/hda/tas2781_hda_i2c.c
linux-6.12/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/6.12.43_1/build M=/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda modules
make[1]: Entering directory '/usr/src/kernel-headers-6.12.43_1'
  CC [M]  /home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.o
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:399:31: error: array type has incomplete element type 'struct hda_quirk'
  399 | static const struct hda_quirk cs420x_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:400:9: error: implicit declaration of function 'HDA_CODEC_QUIRK'; did you mean 'HDA_CODEC_ENTRY'? [-Wimplicit-function-declaration]
  400 |         HDA_CODEC_QUIRK(0x10de, 0x0ac0, "MacBookPro 5,3", CS420X_MBP53),
      |         ^~~~~~~~~~~~~~~
      |         HDA_CODEC_ENTRY
In file included from /home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:528:
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus_a1534_setup.h:71:6: warning: no previous prototype for 'snd_hda_coef_item' [-Wmissing-prototypes]
   71 | void snd_hda_coef_item(struct hda_codec *codec, u16 write_flag, hda_nid_t nid, u32 idx, u32 param, u32 retdata, int srcidx)
      |      ^~~~~~~~~~~~~~~~~
In file included from /home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus_a1534_setup.h:1679:
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus_a1534_pcm.h:135:6: warning: no previous prototype for 'cs_4208_playback_pcm_hook' [-Wmissing-prototypes]
  135 | void cs_4208_playback_pcm_hook(struct hda_pcm_stream *hinfo, struct hda_codec *codec, struct snd_pcm_substream *substream, int action)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus_a1534_pcm.h:253:5: warning: no previous prototype for 'cs_4208_build_pcms_explicit' [-Wmissing-prototypes]
  253 | int cs_4208_build_pcms_explicit(struct hda_codec *codec)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:647:31: error: array type has incomplete element type 'struct hda_quirk'
  647 | static const struct hda_quirk cs4208_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:653:31: error: array type has incomplete element type 'struct hda_quirk'
  653 | static const struct hda_quirk cs4208_mac_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:837:31: error: array type has incomplete element type 'struct hda_quirk'
  837 | static const struct hda_quirk cs421x_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:837:31: warning: 'cs421x_fixup_tbl' defined but not used [-Wunused-variable]
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:653:31: warning: 'cs4208_mac_fixup_tbl' defined but not used [-Wunused-variable]
  653 | static const struct hda_quirk cs4208_mac_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:647:31: warning: 'cs4208_fixup_tbl' defined but not used [-Wunused-variable]
  647 | static const struct hda_quirk cs4208_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.c:399:31: warning: 'cs420x_fixup_tbl' defined but not used [-Wunused-variable]
  399 | static const struct hda_quirk cs420x_fixup_tbl[] = {
      |                               ^~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:229: /home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/patch_cirrus.o] Error 1
make[2]: *** [/usr/src/kernel-headers-6.12.43_1/Makefile:1945: /home/stefan/.local/pkgs/macbook12-audio-driver/build/hda] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernel-headers-6.12.43_1'
make: *** [Makefile:10: all] Error 2
cp /home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko /lib/modules/6.12.43_1/updates
cp: cannot stat '/home/stefan/.local/pkgs/macbook12-audio-driver/build/hda/snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:17: install] Error 1

contents of /lib/modules/6.12.43_1/updates
total 0
```

</details>

dkms doesn't work either:

<details> <summary> Logs (but here probably due to kernel number) </summary>

```
Creating symlink /var/lib/dkms/macbook12-audio/0.1/source -> /usr/src/macbook12-audio-0.1

Sign command: /lib/modules/6.12.43_1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Running the pre_build script................................................ done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j4 KERNELRELEASE=6.12.43_1

Error! Bad return status for module build on kernel: 6.12.43_1 (x86_64)
Consult /var/lib/dkms/macbook12-audio/0.1/build/make.log for more information.

```

</details>

gAlleb, 2025-09-09:
Thanks @leifliddy. It worked, but not in a way it was intended.

My kernel is 6.12 (as you can see from the logs above). So i did

```bash
# Commented out this:

#if (( major_version > 6 || (major_version == 6 && minor_version >= 12) )); then
#   sed -i 's/snd_pci_quirk/hda_quirk/' $hda_dir/patch_cirrus.c
#   sed -i 's/SND_PCI_QUIRK\b/HDA_CODEC_QUIRK/' $hda_dir/patch_cirrus.c
#fi
 
# Changed <= to >=
if (( major_version == 6 && minor_version >= 11 )); then
   sed -i 's/hda_quirk/snd_pci_quirk/' $hda_dir/patch_cirrus.c
fi
```

And it works now) Thanks you!

x5444, 2025-10-16:
@gAlleb and @nathanjhill, I believe what is happening in Void Linux is that they name their Kernels something like 6.12.43_1 instead of 6.12.43-1.
This causes the cut operation at https://github.com/leifliddy/macbook12-audio-driver/blob/master/install.cirrus.driver.sh#L25 to fail and subsequently leads to the download of source code for Kernel 6.12[.0] (after 6.12.43_1 fails), which doesn't yet have the change from `snd_pci_quirk` to `hda_quirk` applied.

Fixing this would need to
- parse the version number more reliably
- parse the patch number too and apply the patches not starting from 6.12 but from 6.12.5 where the change was first introduced

@leifliddy maybe this can somehow be included in the fix for the restructuring for 6.17...